tagyoureit/nodejs-poolController #369: Last Dose Time and Volume for IntelliChem in mqtt

snagytx, 2021-11-28:
**Describe the bug**
I'm not really sure if this is a bug or a feature. I have an IntelliChem setup and I want to graph the dosing in HomeAssistant, so I enabled MQTT. I noticed that the last Dose Time and Volume are not sent to MQTT . I tried to expose the timeDosed to mqtt and it works, but this will send as timeDosed changes, 0,2,3,4,5...20 and then remains at 20 until another dose is triggered. Not sure if this is useful, but for my use case it's not useful.

**To Reproduce**
Add to mqtt.json
```
        {
          "topic": "state/chemControllers/@bind=data.id;/@bind=data.name;/orpTimeDosed",
          "message": "{\"orpTimeDosed\":@bind=data.orp.timeDosed;}"
        },
        {
          "topic": "state/chemControllers/@bind=data.id;/@bind=data.name;/pHTimeDosed",
          "message": "{\"pHTimeDosed\":@bind=data.ph.timeDosed;}"
        },
        {
          "topic": "state/chemControllers/@bind=data.id;/@bind=data.name;/orpVolumeDosed",
          "message": "{\"orpVolumeDosed\":@bind=data.orp.volumeDosed;}"
        },
        {
          "topic": "state/chemControllers/@bind=data.id;/@bind=data.name;/pHVolumeDosed",
          "message": "{\"pHVolumeDosed\":@bind=data.ph.volumeDosed;}"
        },
```
then monitor the mqtt and note that the sensors will constantly increase during dosing - this will make it hard to plot

**Expected behavior**
Get a single value at the end of the dosing, or set it to 0 when it starts then to the dosed time at the end.

**Pool Equipment**
 - Controller: [IntelliCenter]
 - Controller Model: [IntelliCenter i5PS]
 - Controller Firmware Version: [1.047]
 - Pump(s) manufacturer and model: [IntelliFlow 2 VST 011056]
 - Heater(s): [gas]
 - Chemical controller: [IntelliChem]
 - Valves: [ Intellivalve]

rstrouse, 2021-11-28:
Unfortunately, IntelliChem reports this on their status messages just as you described.   However, there is a chemicalDose event that keeps track of the current dose and its status.  Sadly, much of the data that exists for REM Chem is omitted by IntelliChem.  The output is really limited from IntelliChem.  However, I added the chemicalDose emit so that it will give you some additional data when IntelliChem is dosing (we measure this on the status edges).

Pull njsPC and you should have additional information on your mqtt emits.

```
   {
      "name": "chemicalDose",
      "description": "Event when a chemical is being dosed",
      "topics": [
        {
          "topic": "state/chemControllers/@bind=data.id;/@bind=data.chem;",
          "message": "@bind=data;",
          "enabled": true
        }
      ]
    },
```

snagytx, 2021-11-30:
The update solved my issue.

closed 2021-11-30 by snagytx