tagyoureit/nodejs-poolController #370: AutoBackup causing error

bad350trk, 2021-11-28:
I am getting an error after turning on AutoBackups.  I get the general error shown in the screen shot below, and the Backups tab is completely blank now.

[image]

Am unable to recreate the issue to get a replay capture because the Backups tab is empty.

**Pool Equipment**
 - Controller: IntelliCenter
 - Controller Model: IntelliCenter i8PS
 - Controller Firmware Version: 1.047
 - Pump(s) manufacturer and model: IntelliFlow 2 VST 011056
 - Chlorinator: IntelliChlor-40
 - Heater(s): MasterTemp 400
 - Chemical controller: N/A
 - Valves: Intellivalve
 - Any other relevant equipment:

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser Chrome
 - Version 96.0.4664.45 (Official Build) (64-bit)

Thanks,

--Jeff

rstrouse, 2021-11-28:
Upload your config.json file.  You can rename it to config.txt so that it uploads.   The servers element is probably null instead of `"servers": []`.

bad350trk, 2021-11-28:
Here is the config.json file.

[config.txt](https://github.com/tagyoureit/nodejs-poolController/files/7614217/config.txt)

Although, I did a git pull and now it all appears to be functioning.

Thanks,

--Jeff

rstrouse, 2021-11-28:
Yep I fixed the code so that it defaults the servers if they do not exist.  Your file confirms that to be the case when there are no REM servers to backup or restore.

bad350trk, 2021-11-28:
Sweet!  Thanks!

--Jeff

closed 2021-11-28 by bad350trk