tagyoureit/nodejs-poolController #371: Heater not turning on

Fumado, 2021-12-05:
I have a Pentair UltraTemp Heat Pump.  It looks like the njsPC is not reading the water temperature from the heat pump, because the heater stays off, even though I have the water temperature set to 78 and the water temp on the heat pump itself is 63.  Pool temp on the dashboard is blank.  Set temperature on the heat pump shows N/A.  I have not found any other way to turn on the heater in the dashboard.

I've tried setting the heater type to "Heat Pump" and "UltraTemp" but the result is the same.  I actually have an UltraTemp Heat Pump.

The packets in the log are:
{"id":18886,"valid":true,"dir":"out","proto":"heater","pkt":[[],[255,0,255],[165,0,112,16,114,10],[144,0,0,0,0,0,0,0,0,0],[2,49]],"ts":"2021-12-05T12:53:08.368-0800"}
{"id":18889,"valid":true,"dir":"in","proto":"heater","for":[18886],"pkt":[[],[255,0,255],[165,0,16,112,115,10],[160,1,0,1,0,0,0,0,0,0],[2,68]],"ts": "2021-12-05T12:53:08.469-0800"}

Those packets just repeat

**Pool Equipment**
 - NodeJS Pool Controller 7.5 on Raspberry Pi running Ubuntu 21.10
 - Pump(s) Pentair IntelliFlow VS
 - Chlorinator: AquaRite AQR15
 - Heater(s): Pentair Ultratemp Heat Pump

**Desktop (please complete the following information):**
 - OS: macOS Catalina 10.15.7
 - Browser Safari
 - Version 15.1

rstrouse, 2021-12-05:
Unfortunately UltraTemp does not report the water temperature over rs485.  You need a separate temp sensor.   It really is sad that it is not reported.  But I suppose the automation system would have its own anyway.

Fumado, 2021-12-07:
Thanks for the response.  That's too bad, I assumed that the temperature was there.  I'll ask in the appropriate room for a temp sensor recommendation that would work for this.

rstrouse, 2021-12-07:
No worries.  You have lots of options.  REM will manage that temperature for you.
Have a look at these.
https://github.com/tagyoureit/nodejs-poolController/wiki/DIY-Standalone-Nixie-Pool-Controller#temperature-sensors
https://github.com/rstrouse/relayEquipmentManager/wiki/Temperature-Sensors
https://github.com/rstrouse/relayEquipmentManager/wiki/1-Wire-Devices

The easiest is to pop a 10k sensor into the pipe just like you would get if you bought a Pool Automation system.  They are cheap and simple to implement.

closed 2021-12-09 by Fumado