tagyoureit/nodejs-poolController #372: Net connect should retry on disconnect

putnam, 2021-12-17:
For those using socat, it does not look like poolController attempts to recreate the connection after it is closed. If your raspberry pi (or whatever) reboots, disconnects from the network, has a service restart, etc., this currently requires the whole poolController service to be restarted on the remote machine to get the socket back up.

rstrouse, 2021-12-17:
I use socat and it will attempt to reconnect if the connection is lost every 10 seconds.  Do you see it disconnect in the log?

putnam, 2021-12-17:
Yeah, I do show that retry as well -- it looks like under specific circumstances it doesn't retry. Specifically when the socat/remote end shuts down and stops the connection vs. something transient. I'll take a look again tonight and give you a way to repro.

putnam, 2021-12-19:
If socat is stopped cleanly, I see this in the logs from poolController, and no retry is attempted thereafter:

```
[12/19/2021, 12:57:24 AM] info: Net connect (socat) end event was fired
[12/19/2021, 12:57:24 AM] info: Net connect (socat) closed : pool-pi:6420
```

socat can stop cleanly any time the service on the remote is restarted, the device is restarted, packages get updated, whatever.

Here is my socat line:
```
/usr/bin/socat /dev/ttyUSB0,b9600,cs8,cstopb=1,parenb=0,raw TCP-LISTEN:6420,fork,reuseaddr
```

rstrouse, 2021-12-19:
Hmmm... when I shut down my service I don't get a FIN from socat so it remains half-open.  Not sure why but this does explain the retry sequence since FIN would remove the inactivity timer from the socket.  This should be easy enough to fix.

putnam, 2021-12-20:
Applied the change, looks fixed to me

rstrouse, 2021-12-21:
Yeah I think we are covering the bases with this change.  I am going to close this for now.

putnam, 2021-12-23:
This may or may not be related to this change, but when I went to look at the controller again today, I noticed it had many open connections to the socat backend, and socat had like 30 forked instances running on the Pi. After killing the service on the Pi and restarting it, njsPC tried to reconnect all those dozens of times again. Something may not be garbage collecting sockets somewhere.

putnam, 2021-12-23:
After restarting njsPC, I think I see at least one of the reasons these keep accumulating:

```
Init state for Pool Controller
[12/23/2021, 1:03:56 AM] info: The current git branch output is master
[12/23/2021, 1:03:56 AM] info: The current git commit output is 2c79fb0551e2bff3e1a7f98f2e4f34a1b7b69765
[12/23/2021, 1:03:56 AM] info: Starting up SSDP server
[12/23/2021, 1:03:56 AM] info: Checking njsPC versions...
[12/23/2021, 1:03:56 AM] info: Starting Pool System intellitouch
[12/23/2021, 1:03:56 AM] info: Server is now listening on 0.0.0.0:4200
[12/23/2021, 1:03:56 AM] info: Net connect (socat) connected to: pool-pi:6420
[12/23/2021, 1:03:56 AM] info: Net connect (socat) Connection connected
[12/23/2021, 1:03:56 AM] info: Net connect (socat) ready and communicating: pool-pi:6420
[12/23/2021, 1:03:56 AM] info: Auto-backup initialized Last Backup: 1969-12-31T18:00:00.000-0600
[12/23/2021, 1:03:58 AM] info: New socket client connected L6T1UE_jde94teAeAAAB -- 127.0.0.1
[12/23/2021, 1:04:06 AM] warn: Net connect (socat) connection idle: pool-pi:6420 retrying connection.
[12/23/2021, 1:04:06 AM] info: Net connect (socat) successfully closed: pool-pi:6420
[12/23/2021, 1:04:06 AM] info: Net connect (socat) connected to: pool-pi:6420
[12/23/2021, 1:04:06 AM] info: Net connect (socat) Connection connected
[12/23/2021, 1:04:06 AM] info: Net connect (socat) ready and communicating: pool-pi:6420
[12/23/2021, 1:04:16 AM] warn: Net connect (socat) connection idle: pool-pi:6420 retrying connection.
[12/23/2021, 1:04:16 AM] info: Net connect (socat) successfully closed: pool-pi:6420
[12/23/2021, 1:04:16 AM] info: Net connect (socat) connected to: pool-pi:6420
[12/23/2021, 1:04:16 AM] info: Net connect (socat) Connection connected
[12/23/2021, 1:04:16 AM] info: Net connect (socat) ready and communicating: pool-pi:6420
```

njsPC is repeatedly noticing an idle connection, then making a new one, and I see more and more socat instances piling up.

rstrouse, 2021-12-23:
That is weird.  There should be traffic every few seconds from IntelliTouch so it should  not be getting an idle timeout.  If it doesn't hear from IntelliTouch in 10 seconds it will restart the connection.  But it does not restart the connection without first destroying the previous connection.  In the condition above is IntelliTouch actually communicating?

rstrouse, 2022-01-30:
I am closing this for now.  If you notice problems please open another issue.

closed 2022-03-05 by rstrouse