tagyoureit/nodejs-poolController #373: Raspberry PI4 dead, or something else?

rock-crusher, 2021-12-20:
I have a Raspberry PI4 with Nixie installed along with the Sequent Microsystems MEGA-BAS Building Automation Card & Sequent Microsystems 8 relay hat. I have been bench testing and getting familiar with this controller prior to putting into service. I have a keyboard, mouse & monitor attached to the PI while experimenting with different schedules, pool/spa mode scenarios, valve actuator operation, pump motor speeds, lighting & auxiliary relay control.

All has been going quite well since rstrouse walked me through the entire set-up of Nixie. I’m sure it was very frustrating for him since I have 0 experience with any of this (thanks again rstrouse).

This RPI4 I have has seemed to be a little wonky from the get-go. I think rstrouse may have suspected a problem with my PI early on in his efforts helping me get Nixie installed properly. Well, after a lot of peculiar activities like in the beginning the screen would randomly flicker on and off. This then progressed to both the mouse pointer along with the screen completely freezing up while in the middle of using it, the PI would never recover on its own until it’s unplugged/restarted. Other times while using it the screen slowly starts disappearing away into a bunch of tiny little colored pixels until it all finally turns to solid black screen and the green indicator light shuts off leaving only the red still illuminated. This scenario begins to happen more frequent and now it just simply won’t boot-up at all. When trying to restart all that happens is the red & green light will turn on leaving a blank screen, nothing else, no sign of life. I tried this several times with the same outcome just a solid red & green light on. 

After a couple days of just letting it sit I tried again, this time the green light flashed 8 times, I’ve learned this means SDRAM failure? I tried again after unplugging everything connected to the PI (mouse, keyboard, monitor) including removing the MEGA-BAS card and relay hat, but same thing, got the usual solid red & green light.

Funny thing, although short lived, it seemed to have been working quite well, I was getting schedules set-up etc. but then little by little the above started to happen.

 It is of my opinion this PI4 has bit the dust.

Anyone else had problems with the PI4 not booting, or just simply randomly shutting down?

rstrouse, 2021-12-21:
The number one suspect for the symptoms you described is insufficient power.  Unfortunately, when a RPi has insufficient power it will often corrupt the SD card because SD card writes require stable power.  After that it starts to smell like a whole host of other issues.  

However, the number 2 reason is because of failing memory.  In which case you can experience the corrupt video and halt the processor.  Just to be sure you didn't overclock it did you?  That requires futzing that neither of us are qualified to do.

rock-crusher, 2021-12-21:
>The number one suspect for the symptoms you described is insufficient power.

Well, I would think my power supply would be more than sufficient.  It is a brand new TR100VA001 as shown in the pic below.

>Just to be sure you didn't overclock it did you? That requires futzing that neither of us are qualified to do.

Lol… I don’t even know what that means, let alone know how to go about doing that.

rock-crusher, 2021-12-21:
[image]

rstrouse, 2021-12-21:
You could image a new SD card and plug it in to see if it works.  But you may be right that there is a RAM issue on the card.

tagyoureit, 2021-12-21:
I would also try to remove all Sequent hats and just power the Pi with a USB-C and see if there is any power issues between the Pi & hats.

rock-crusher, 2021-12-21:
>  You could image a new SD card and plug it in to see if it works. But you may be right that there is a RAM issue on the card.

Thank you rstrouse, I did just that, but same result.

>  I would also try to remove all Sequent hats and just power the Pi with a USB-C and see if there is any power issues between the Pi & hats.

Hi tagyoureit, yes, I did try this. I originally bought the CanaKit that came with a power supply, fan, heat sinks & some other goodies. Unfortunately that made no difference.

After doing some research, it seems this is happening quite frequently with the PI4. 

Anyway, I am going to see about an exchange, or refund, or…?

rstrouse, 2021-12-21:
In the words of the immortal Dr McCoy.  It's dead Jim.

rock-crusher, 2021-12-22:
>In the words of the immortal Dr McCoy. It's dead Jim.

Lol... May it R.I.P.

I decided to return the PI back to Best Buy where I bought it. Unfortunately, they were sold out to make an exchange, but were more than happy issuing a full refund.

Apparently these things are nowhere to be found right now and I mean nowhere!

I’ll just have to wait until they become available again, probably sometime next year. Hopefully I’ll have better luck then.

Good thing is; I wasn’t dependent on this for pool automation just yet.

Thanks again for your help & suggestions. I’ll be sure to let you know when I get a new PI up & running again.

rstrouse, 2022-01-16:
I am closing this for now.  Let us know when you get back from your epic swim and don't forget the ziploc bag for the Pi.

closed 2022-01-16 by rstrouse