tagyoureit/nodejs-poolController #374: Fatal error for unreachable code after upgrade to 7.6.0

geeto1969, 2021-12-23:
**Description**
Poolcontroller raises a fatal error for unreachable code after upgrading from 7.4.0 to 7.6.0. Error is resolved when explicitly pulling tag 7.4.0.

**To Reproduce**
Steps to reproduce the behavior:
1. Starting from working docker-based installation using v7.4.0 with no explicit image tag, i.e. `image: msmi/nodejs-poolcontroller`, or an explicit image tag recently changed to `image: msmi/nodejs-poolcontroller:7.6.0`
2. Execute docker-compose down
3. Execute docker-compose pull
4. Wait for pull to complete
5. Execute docker-compose up
6. Observe fatal error in console output (see below)

**To Resolve**
1. Modify docker-compose.yml to include explicit tag reference to v7.4.0
`    image: msmi/nodejs-poolcontroller:7.4.0`
2. Execute docker-compose down
3. Execute docker-compose pull
4. Wait for pull to complete
5. Execute docker-compose up
6. Observe error for pool-controller is resolved

Changing the image tag above to `image: msmi/nodejs-poolcontroller:7.6.0` and going through the reproduce steps will reintroduce the error.

FYI, poolcontroller-dashpanel raises an identical error after both were upgraded. This issue cannot be resolved because there is no tag to pull the build available at the time of poolcontroller 7.4.0. After restoring pool-controller to 7.4.0, it works with other clients, but pool-controller-dashpanel continues to raise an error. I understand this is a different project, but it may provide a clue about the issue.

**Expected behavior**
An upgrade should not raise fatal error.

**Console Output**
```
poolcontroller              |
poolcontroller              |
poolcontroller              | #
poolcontroller              | # Fatal error in , line 0
poolcontroller              | # unreachable code
poolcontroller              | #
poolcontroller              | #
poolcontroller              | #
poolcontroller              | #FailureMessage Object: 0xbea5f62c
poolcontroller-dashpanel    |
poolcontroller-dashpanel    |
poolcontroller-dashpanel    | #
poolcontroller-dashpanel    | # Fatal error in , line 0
poolcontroller-dashpanel    | # unreachable code
poolcontroller-dashpanel    | #
poolcontroller-dashpanel    | #
poolcontroller-dashpanel    | #
poolcontroller-dashpanel    | #FailureMessage Object: 0xbe90262c
poolcontroller exited with code 133
```

**docker-compose.yml at the time of upgrade pull**
```
version: "3.7"

########################### NETWORKS

########################### SERVICES
services:

  # Pool Controller
  # Touch (create empty files) config.json. Set config.json permissions to 600.
  # touch $DOCKERDIR/poolcontroller/config.json
  # chmod 600 $DOCKERDIR/poolcontroller/config.json
  poolcontroller:
    container_name: poolcontroller
    restart: always
    group_add:
      - dialout
    devices:
      - /dev/ttyUSB0
    ports:
      - "4200:4200"
    environment:
      - TZ=$TZ
    volumes:
      - $DOCKERDIR/poolcontroller/config.json:/app/config.json
      - $DOCKERDIR/poolcontroller/data:/app/data
      - $DOCKERDIR/poolcontroller/backups:/app/backups
    image: msmi/nodejs-poolcontroller

  # Pool Controller Dashboard
  # Touch (create empty files) config.json. Set config.json permissions to 600.
  # touch $DOCKERDIR/poolcontroller-dashpanel/config.json
  # chmod 600 $DOCKERDIR/poolcontroller-dashpanel/config.json
  poolcontroller-dashpanel:
    restart: always
    container_name: poolcontroller-dashpanel
    ports:
      - "5150:5150"
    environment:
      - TZ=$TZ
    volumes:
      - $DOCKERDIR/poolcontroller-dashpanel/config.json:/app/config.json
    image: msmi/nodejs-poolcontroller-dashpanel
    depends_on:
      - poolcontroller

```

**Pool Equipment**
 - Controller: EasyTouch
 - Controller Model: EasyTouch PS-4
 - Controller Firmware Version: [e.g. 1.047]
 - Pump(s) manufacturer and model: Pentair IntelliFlo VSF
 - Heater(s): Pentair MasterTemp 400k NG

rstrouse, 2021-12-24:
I do not use docker but there was a recent change from another user that requested the Dockerfile in the repo be changed to reference python3.  Could this be causing your issue?

geeto1969, 2021-12-24:
I don't believe that is the cause.

I did some additional investigation, and I have seen reports that newer versions of lts-alpine have this issue with some ARM processor versions. I am going to add my Raspberry Pi HW info to the original issue text.

I will test whether the fixes I saw for other projects work here and report back.

geeto1969, 2021-12-25:
I confirmed that pinning node to lts-alpine3.12 in the dockerfile resolves the issue.

commit 7c7b6f03 (2021-12-26): Merge pull request #375 from geeto1969/issue_374

Fixes #374: Fatal error for unreachable code after upgrade to 7.6.0
closed 2021-12-26 by rstrouse