tagyoureit/nodejs-poolController #376: Error on upgrade

wesgipe, 2022-01-08:
Hello!  I just tried upgrading to the latest version and am not getting the following on sudo npm start:

pi@PoolController:~/nodejs-poolController $ sudo npm start

> nodejs-poolcontroller@7.6.0 start /home/pi/nodejs-poolController
> npm run build && node dist/app.js

> nodejs-poolcontroller@7.6.0 build /home/pi/nodejs-poolController
> tsc

sh: 1: tsc: not found
npm ERR! code ELIFECYCLE
npm ERR! syscall spawn
npm ERR! file sh
npm ERR! errno ENOENT
npm ERR! nodejs-poolcontroller@7.6.0 build: `tsc`
npm ERR! spawn ENOENT
npm ERR! 
npm ERR! Failed at the nodejs-poolcontroller@7.6.0 build script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm WARN Local package.json exists, but node_modules missing, did you mean to install?

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2022-01-08T16_51_31_240Z-debug.log
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! nodejs-poolcontroller@7.6.0 start: `npm run build && node dist/app.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the nodejs-poolcontroller@7.6.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm WARN Local package.json exists, but node_modules missing, did you mean to install?

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2022-01-08T16_51_31_371Z-debug.log
pi@PoolController:~/nodejs-poolController $ 

Any ideas on what I could try?  dashPanel also will not start after the upgrade

tagyoureit, 2022-01-08:
You shouldn't/don't need sudo to run this app.  First, backup the data files "./data" as they will contain any customizations you have done.  
Then, 
1. Completely delete the nodejs-poolController directory.  `rm -rf nodejs-poolController`
2. `git clone git@github.com:tagyoureit/nodejs-poolController.git` to clone the repo
3. Copy back the poolConfig.json and poolState.json to the data directory.
4. Run `npm install` in the new njsPC folder
5. Run the app with `npm start`.

You could also follow these steps for dP (`git clone https://github.com/rstrouse/nodejs-poolController-dashPanel.git`) and backing up the data files aren't really necessary.

Subsequent upgrades should be run with `git pull` inside the njsPC and dP directories, followed by steps 4 and 5.

wesgipe, 2022-01-08:
Thanks!  I was able to back up the data files and delete the folder, though deleting the folder required sudo.  Now I get this when trying to clone:

pi@PoolController:~ $ git clone git@github.com:tagyoureit/nodejs-poolController.git
Cloning into 'nodejs-poolController'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

rstrouse, 2022-01-08:
Use the following command to clone the repo.
`git clone https://github.com/tagyoureit/nodejs-poolController.git`

wesgipe, 2022-01-08:
Working like a champ now.  Thanks!!

closed 2022-01-08 by wesgipe

tagyoureit, 2022-01-09:
Oh, ha!  The main readme was outdated.  I just updated it and included upgrade instructions.