tagyoureit/nodejs-poolController #377: LOW PRIORITY - sometimes DashPanel stays tuck at 'Loading 0%' during a service restart even though dash/nsjpc is fully loaded

johnny2678, 2022-01-11:
I frequently get this at the top right of dashpanal
[image]

doesn't happen every time njspc restarts, but sometimes.

managed to get a replay the last time it happened:
[replay (28).zip](https://github.com/tagyoureit/nodejs-poolController/files/7850127/replay.28.zip)

johnny2678, 2022-06-09:
This is still happening on the latest version of dash/njspc.  Doesn't look like any functionality is affected but still get `Loading 0%` sometimes after start up.

still not important but let me know if there's any other information I can provide.

tagyoureit, 2022-06-23:
I took a look and noticed inbound MQTT requests being processed before the system is ready.   These are:

```
[1/11/2022, 4:57:11 PM] debug: MQTT: Inbound pool/easytouch2-4/state/feature/setState: {"id":15,"isOn":"on"}
[1/11/2022, 4:57:11 PM] debug: MQTT: Inbound pool/easytouch2-4/state/circuit/setState: {"id":2,"isOn":"off"}
...
[1/11/2022, 4:57:11 PM] debug: MQTT: Inbound pool/easytouch2-4/config/chemController: {"id":50,"alkalinity":65}
...
[1/11/2022, 4:57:11 PM] debug: MQTT: Inbound pool/easytouch2-4/state/chlorinator: {"id":1,"poolSetpoint":0}
```

This has the side effect of loading up the internal pool objects before they are fully initialized.  I added code to ignore any inbound MQTT requests until the system is initialized.  Can you pull and try again?  

If this works, we should do the same for the api's.  Let me know how this works.  Thanks for your patience.

johnny2678, 2022-08-17:
Closing this... haven't noticed the percentage getting stuck in a few releases now. Thx!

closed 2022-08-17 by johnny2678