tagyoureit/nodejs-poolController #380: Solar settings for Start/Stop Temp Delta do not get sent to OCP

johnny2678, 2022-01-16:
[image]

I'll change the settings in dash (e.g. 5/2) but the settings don't get committed in the OCP (Easytouch 2 4 in my case)

rstrouse, 2022-03-05:
Can I get you to pull njsPC and dashPanel.  Then using either ScreenLogic, wireless controller, or an indoor control panel change these values while capturing a replay.  Finally, post that replay capture here so I can examine what the message handshake looks like.

johnny2678, 2022-03-05:
There doesn't seem to be any place to change this aside from the OCP - don't see any settings in screen logic and I don't have an indoor panel.  Hope changing them at the OCP is sufficient.  If so, replay attached...

I changed the start temp to 5 and stop temp to 2
[replay (29).zip](https://github.com/tagyoureit/nodejs-poolController/files/8191499/replay.29.zip)

rstrouse, 2022-03-05:
Oh wow that explains a lot.  I have recently looked at the settings for UltraTemp heaters and it appears these also can only be set on the OCP.  I think I need to find somebody who forked out the cash for the ICP or the WCP.

johnny2678, 2022-03-05:
I do have one of those white Easytouch branded pool remotes.  Is that a WCP?

Anyway, popped in some fresh batteries, went to the panel to sync the addresses and then remembered why I let it sit and collect dust.  It shows time/temp/mode but anytime you click into settings or turn on a circuit it says "communications lost"

Back to the closet it goes...

rstrouse, 2022-03-06:
Yes that is a WCP (wireless control panel) ... bummer.

tagyoureit, 2022-03-25:
Closing this as it appears we can't actually control these deltas.