tagyoureit/nodejs-poolController #383: dashPanel displays Heat Mode Off when it is on!

antamy, 2022-01-26:
This may be related to previous issues I filed 12+ months ago, which were fixed.  With the latest njspc and dashPanel, with heat mode set to Solar Only, the main UI shows Heat Mode Off. Clicking on Heat Mode shows a dialog where Solar Only is selected.

Attached log is form a running system, with the configuration reload captured.

[image]
[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/7944059/replay.zip)

rstrouse, 2022-01-27:
At some point did you disable your solar heater by either deleting it in dashPanel or changing the setting in IntelliTouch.   I don't mean simply disabling it on the body but removing the solar option from the system?  Then later put it back in while it was selected on a body.  I am trying to determine whether IntelliTouch cleans up these options as it looks like it might not on its own.

What happens when you select off then back to Solar Only.   Does the heat mode text show as Solar Only on the display?

antamy, 2022-01-27:
I don't believe I've changed anything.

If I click on Heat Mode and then change the mode, that is correctly reflected on the main dashPanel page.  Once that is done, the change persists and I see that if I reconnect to the dashPanel from another browser.

If I reload the configuration, it still displays correctly.  I also tried restarting poolController and it still displayed correctly.  It seems that going into the Heat Mode dialog fixed the issue, which seems a little odd!

I'll keep an eye on it and let you know if I see it change to Off on the main page.

rstrouse, 2022-01-27:
Actually, its not that odd but I don't know how to get around it.  If IntelliTouch reports that the solar heater is not installed and the body heater was set to one of the solar options IntelliTouch will continue to report that it is set to that heat mode even though that heater does not exist.   However, if you re-enable the solar heater by reinstalling it or enabling it in the OCP it doesn't notify that this was done.  I have seen reports that disabling the solar still moves the solar valve after until the heater is deselected off the body.  

In this instance the OCP reported that it had solar selected but the solar heater couldn't be found in dashPanel so it did the best thing that it knew to do and that is to change the label to off on the state but accept the settings from the OCP.  If it happens again we may want to add some code in njsPC to do some verification of these modes when it requests the configuration from the OCP.  Heck we could even change it if the heater isn't enabled.

rstrouse, 2022-01-30:
I am closing this for now.  Open another issue if you see this again.

closed 2022-01-30 by rstrouse

antamy, 2022-01-30:
Ok, that’s fine. It’s only cosmetic and there is an easy workaround.  Thanks, as always, for your help.