tagyoureit/nodejs-poolController #385: njsPC does not start UltraTemp Heat Pump

Fumado, 2022-01-31:
In remote=RS-485 the heater does not start heat.

I've attached a replay when the pool temp is 77 and I've set the setpoint=85°

It appears that they are communicating, because I get 'in' and 'out' packets.
[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/7972498/replay.zip)

rstrouse, 2022-01-31:
Thanks for that,  It appears that I completely misunderstood what the differential temp was supposed to do.  I assumed that it was supposed to act like a solar heater but with the air temp sensor to determine the call for heat.   I am working on switching this over to a different approach so that it operates the way it was intended.   Pentair, did not document this feature so, without one of these heaters I didn't quite have a grasp on what it was doing.

So what Nixie was doing was suspending the call for heat until the air temp got within the setpoint range.  In your case this would never happen so it would never call for heat.  I will have a fix later today.

rstrouse, 2022-02-01:
Go ahead and pull njsPC.  I believe I have the proper `call for heat` algorithms.  I now understand why you need a solar sensor in IntelliCenter to operate this heater.  This is not a requirement for Nixie.

Fumado, 2022-02-01:
OK, will do.  Now, I don't have an air sensor setup yet, but I do have a water temp sensor.  Will this work based on the water temp?

rstrouse, 2022-02-01:
Yes the reason why the Pentair automation actually checked the delta between these is so when there is a gas heater and an UltraTemp installed it would check both the air and solar sensors to see whether the UltraTemp was capable of reaching the setpoint.  If not the gas heater would be engaged.  This setting is the UltraTemp Preferred setting that only shows up when both heaters are installed.

That being said it lots of folks have had to override this on their Pentair OCPs by either installing a 10k temp probe on the solar connector or jumpering it with a resistor.   We have removed this requirement since we know whether solar is installed and/or the gas heater.  In your case since there is no gas heater you do not have the UltraTemp Preferred option and it will never try to evaluate the temperature delta.

Fumado, 2022-02-02:
I pulled njsPC and it said already up to date.

rstrouse, 2022-02-02:
Before that when did you pull njsPC?

Fumado, 2022-02-02:
January 30th, Sunday.

rstrouse, 2022-02-02:
Click on the hamburger menu then select the system tab.  What value do you have for the circled item?
[image]

Fumado, 2022-02-02:
c6d3eab

[image]<https://user-images.githubusercontent.com/47839015/152206691-4298b5ae-b014-4475-9396-22edc78ba280.png>

rstrouse, 2022-02-02:
Hmmm...  That definitely does not reflect the current repo version.

Type the following commands an post the result here.
`cd ~/nodejs-poolController`
`git branch -v`

Fumado, 2022-02-02:
~/nodejs-poolController$ git branch -v
* master 2f73c92 [ahead 72] Added floating point threshold values for pH.  Previously these were set to integer which is not valid.  Unfortunately, you cannot simply change the data type for the field and expect it to work.

rstrouse, 2022-02-02:
Did you make changes to any of the files in the repo?  That is very odd that it is so far behind.  That commit happened on Jan 5th and there were many commits after that.

Do another `git pull` from the `~/nodejs-poolController` folder and report the output from that command here.

Fumado, 2022-02-02:
No, I don't know enough to make any changes on my own ;-)
  
At one point when you said to do a 'pull' there was an error...don't remember exactly but my files were newer, they were package.json and package-lock.json.  Sometimes when the UltraTemp starts it pops the breaker and of course the Raspberry Pi loses power.  Maybe that happened and it created new files.  I searched the 'net and it seemed that "stash" was the option where I wouldn't cause any harm, so that's what I did and was able to do the 'pull.'

The output from another git pull:
~/nodejs-poolController$ git pull https://github.com/rstrouse/nodejs-poolController
From https://github.com/rstrouse/nodejs-poolController
 * branch            HEAD       -> FETCH_HEAD
Already up to date.

rstrouse, 2022-02-02:
Let's try the following commands from the `~/nodejs-poolController` folder.

`git fetch --all`
` git reset --hard origin/master`

Fumado, 2022-02-02:
OK, so I'm now at git commit fe39f89, and more importantly, the heater is now working!

Thanks for that.  Next time my files change I'll use the fetch --all and --hard origin/master.

rstrouse, 2022-02-02:
Awesome!

rstrouse, 2022-02-02:
I am going to close this for now and if you have any other issues open a new one.

closed 2022-02-02 by rstrouse