tagyoureit/nodejs-poolController #390: Controller doesn't seem to understand Ultratemp Hybrid ETI "Dual" mode.

terafin, 2022-02-25:
**Describe the bug**
Dual mode seems to show as heat mode "off" via the API and UI

**To Reproduce**

1. Set the Easytouch 8 to heat mode "Dual"
2. Wait
3. Check UI and API to see mode is "off"

**Expected behavior**
Works, as well as control!

**Packet Capture**
I ran through the heat modes on the UltraTemp ETI in this order:
Heat pump
Gas
Hybrid
Dual

Attached a packet relay for all together

**Pool Equipment**
 - Controller: EasyTouch 8
 - Controller Firmware Version: 2.180
 - Pump(s) manufacturer and model: IntelliFlow 2
 - Chlorinator: IntelliChlor-40
 - Heater(s): Ultratemp ETI Hybrid
 - Chemical controller: None
 - Valves: None
 - Any other relevant equipment: None

**Desktop (please complete the following information):**
 - OS: macOS 13 and Windows 11

Let me know how I can help!

[replay_modes.zip](https://github.com/tagyoureit/nodejs-poolController/files/8139288/replay_modes.zip)

rstrouse, 2022-02-25:
I need to know what settings are available for heat modes with the ETi heater.

Is this which ones exist:
Off
Gas Heater
Heat Pump Preferred
Heat Pump Only
Dual

terafin, 2022-02-25:
It looks like the modes are:

Off
Heat Pump
Gas
Hybrid
Dual

(there's no heat pump preferred from the EasyTouch UI, or... as far as I can see in the ETI manual)

rstrouse, 2022-02-25:
Hybrid is heat pump preferred.  What are the settings under the hybrid menu that were set at the time you created the capture.  I need to know what each of the values were set to.
[image]

terafin, 2022-02-26:
Yup, I'll take screenshots for everything, Hybrid has settings under it for when it uses Heat pump, etc.

terafin, 2022-02-26:
(and re-capture)

rstrouse, 2022-02-26:
I need to know all 4 values from both pages.

terafin, 2022-02-26:
Sounds good, give me 15-30 mins, and I'll upload :) Thanks!

terafin, 2022-02-26:
Here's the sequence:

* Heat mode off, pump on
* Reload
* Off
* Heat pump only
* Gas only
* Hybrid
* (Breaker tripped on filter, sorry - not sure what happened)
* Hybrid
* Dual
* Off

Here's the log:
[replay_full_cycle.zip](https://github.com/tagyoureit/nodejs-poolController/files/8145593/replay_full_cycle.zip)

Screenshots:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

rstrouse, 2022-02-26:
Thanks for that.  It was very enlightening.  It appears that when the original development for the firmware was created they did not have enough option space for storing all the possibilities when the UltraTemp ETi came along.  This means that the configuration slots are not quite what we expected.

So I need another replay capture that is very specific.  Start the replay then do the following steps on the EasyTouch panel.
Step 1:
Navigate to the MENU > SETTINGS > ULTRATEMP ETI > SETTINGS
Set enabled to NO then set it back to yes
Wait at least 5 seconds and go to step 2

Step 2:
Go to settings and change these to the following.
Spa Econo Time Hours = 3
Spa Boost Temperature = 7
Wait at least 5 seconds and then change them back to their original settings.  I assume this was 1 and 5.

Step 3:
Change the settings on page to to the following.
Pool Econo Time Hours = 4
Pool Boost Temperature = 8
Wait at least 5 seconds then change them back to their original settings.  I assume this was 1 and 5.

This will let us know where the bits have been stuffed.  Thanks for your help.

terafin, 2022-02-26:
No problem!

Done.

[replay_heat_settings_change.zip](https://github.com/tagyoureit/nodejs-poolController/files/8147507/replay_heat_settings_change.zip)

rstrouse, 2022-02-27:
You need to go into the hamburger menu and turn the packet logging off.  This file is too large to get a good picture of what is going on when you perform the operations.  This function will also fill up your pi with a huge amount of data.

Go into the logging section under the configuration menu and uncheck the circled options.  Then could I get a replay of the same operation from above.
[image]

rstrouse, 2022-02-27:
Ok so I read the information very carefully from the EasyTouch user manual.  Sometimes the settings are not saved from the panel until you exit so the data isn't written to the RS485 bus.

Step 1:
Navigate to the MENU > SETTINGS > ULTRATEMP ETI > SETTINGS
Set enabled to NO then set it back to yes
Wait at least 5 seconds and go to step 2

Step 2:
Go to settings and change these to the following.
Spa Econo Time Hours = 3
Spa Boost Temperature = 7
Press the Menu button
Wait at least 5 seconds and then change them back to their original settings. I assume this was 1 and 5.
Press the Menu button again

Step 3:
Change the settings on page to to the following.
Pool Econo Time Hours = 4
Pool Boost Temperature = 8
Press the Menu button
Wait at least 5 seconds then change them back to their original settings. I assume this was 1 and 5.
Press the Menu button again

terafin, 2022-02-27:
Thanks! Just got up and going here, will get this done in the next 15 minutes or so :)

rstrouse, 2022-02-27:
Thanks for the help.  The settings from the Touch controllers can be confusing.  The only way to really see what is going on is to have very specific targets and witness the changes going on.  Can you change any of the ETi configuration values with ScreenLogic or the wireless remote?

terafin, 2022-02-27:
Not a problem at all. I sadly don't have a ScreenLogic or wireless remote. Just been using MQTT [and this project] to control and remotely manage my pool!

rstrouse, 2022-02-27:
You aren't missing anything.  But sometimes it is easier than bending over the OCP to make the config changes.   I figured you were doing some MQTT work.  You do know that both the sockets and the MQTT are change constrained so that they only emit when there are changes.   I noticed you are calling /state/all every couple of seconds.   MQTT should always have the most recent state information present.  Lets get through this though before we delve into anything there.

terafin, 2022-02-27:
[heat_capture.zip](https://github.com/tagyoureit/nodejs-poolController/files/8149148/heat_capture.zip)

Uploaded.

Note: I have my own MQTT adapter that I use, before you had your own MQTT bridge: https://github.com/terafin/mqtt-pool-controller-bridge

I'll deprecate it eventually, but for now, it re-emits things in a way I handle MQTT at my home :)

rstrouse, 2022-02-28:
I am narrowing this down but I am having some issues decoding the Econo Time and Boost Temp for both bodies.  See the message display below.  This is the output from the capture provided in the previous message.   Not the messages listed with action 34.  These are telling ET that the heater configuration is changing.  All of the information for the heater setups are contained in 3 bytes.  

So when you disabled the ETi the with message id 337 it sent the proper bytes to disable it.  Then when you re-enabled it on message id 357 you can see it set the second byte to 16 which enables the Hybrid mode heater.   All good so far.  The third byte on this same action 34 message should contain the 4 values we are looking for.  However, from then on out it never changes.

The only thing I can think of is that you did not press the menu button to exit out of the settings between changes.  I do see that when the OCP (Easytouch) is communicating with the heater at some point it switched from the spa settings
```
Econo Time = 1
Boost Temp = 5
```
To the pool settings on message 440
```
Econo Time = 4
Boost Temp = 8
```
Did you do something during this capture to switch the heater between spa and pool?  The panel remained in pool mode throughout the capture.  So I never see the settings for the spa which should have been
```
Econo Time = 3
Boost Temp = 7
```
[image]

Any idea why I am not seeing all the changes?

terafin, 2022-03-02:
@rstrouse Sorry for the delay, was not feeling great.

I can confirm I did go back out to the main screen from settings every time, however, maybe not long enough. I will run a new trace tonight!

terafin, 2022-03-02:
Next shot! I waited a little longer now..

[replay_heat_latest.zip](https://github.com/tagyoureit/nodejs-poolController/files/8165698/replay_heat_latest.zip)

rstrouse, 2022-03-02:
Hope you are feeling better!   So during this output it looks like the pool was running and there was a call for heat before you disabled the heater.   Do you by chance know what the heater was using to heat the pool?  Heatpump, Gas, or Both?

terafin, 2022-03-02:
There was a call for heat before I disabled it, it was hybrid mode. I don’t know if it actually was kicking gas as well, but the heat pump was running.

I can run another trace clean if you’d like?

--  Justin

rstrouse, 2022-03-02:
I think I have the configuration down.  At this point I probably need to figure out the heat mode changes and the heat status changes.  The reason I say this is because I am seeing a very strange heat status when it was originally calling for heat.

I need a capture where you are changing the heat modes and then getting the heater to kick on.  It is important that I can figure out what order things happened it so I can see what the byte values are.

So lets focus on the pool body for now.
1. Set the pool and spa bodies to off for their heater mode.
2. Set the pool body to heat with heat pump
3. Allow the heater to start heating
4. Change the heat mode on the pool body to heat with gas only
5. Allow the heater to start heating
6. Change the heat mode on the pool body to heat with hybrid
7. Allow the heater to start heating.  I assume we will only be able to get the heat pump to kick on.
8. Change the heat mode on the pool body to heat with both
9. Allow the heaters to start heating

So I am going to follow the changes for the pool body from Off, Heat Pump Only, Gas Only, Heat Pump Pref, and Both.

terafin, 2022-03-03:
Attached

Notes:
• It took like almost 5-8 minutes to go from heat pump to gas mode. 
• I changed the mode to off at the end to cap it

[replay_all_modes.zip](https://github.com/tagyoureit/nodejs-poolController/files/8174806/replay_all_modes.zip)

rstrouse, 2022-03-03:
Ok so pull njsPC and dashPanel.  Based on the replays I think we have it.  Please test this out and let me know how it goes.

To update njsPC and dashPanel shut them both down and run the following commands.  If you are running PM2 it should restart these after the update.
`cd ~/nodejs-poolController`
`git pull`
`npm i`

`cd ~/nodejs-poolController-dashPanel`
`git pull`
`npm i`
Hard refresh your browser to make sure it picks up the new files and not what is cached.

terafin, 2022-03-03:
I updated (and confirmed the github hashes are correct), and the behavior is indeed different.

I'm seeing the following:

The heat mode dial only offers three options:
[image]

The options seem not aligned with the actual function:

Off => Off
Heater => HPOnly
Heat Pump Only => Hybrid
Heat Pump Preferred => Gas
Dual => Missing

I refreshed the browser, tried 2 computers, and 2 browsers, so I'm pretty sure no browser cache is involved?

rstrouse, 2022-03-03:
Go into the config pages and show me what heaters are installed

terafin, 2022-03-03:
[image]

rstrouse, 2022-03-03:
Hmmm.  That isn't right.  Go into the UltraTemp ETi screen on the OCP then set enabled to no and exit then go back in and set enabled to yes.   Then reload the screen above.

terafin, 2022-03-03:
Sadly same effect. I also did this:
• Disable ETI again on EasyTouch
• Reload config from dashboard
• Verified I saw only "Gas Heater" -- not sure why?
• Enabled ETI
• UltraTemp appeared again, but showed only as a Heat Pump as above.

rstrouse, 2022-03-03:
Alright, I moved the byte alignment for the detection message.  Pull njsPC again and see if it detects.

terafin, 2022-03-03:
Now it's noticed as a Hybrid. However the Address is empty, and the heat pop up doesn't show anything now other than setpoint.

[image]
[image]

I set the Address to 0, and saw this: (still doesn't show heat modes as above)

[image]

rstrouse, 2022-03-03:
Ok this is improved.  Now disable and re-enable the heater on the EasyTouch.

terafin, 2022-03-03:
No change sadly, did some reloading too :(

rstrouse, 2022-03-03:
Ok so give me a replay where you are only changing the setpoint.  This is because of the way Pentair had to shoehorn this heater into their firmware.  There are only two slots for heaters in the software and one of them was historically reserved for a gas heater.  In this instance they are using one slot (the gas slot) for a heatpump and a gas heater.

We just need to figure out which bits went where.

terafin, 2022-03-03:
[replay 2.zip](https://github.com/tagyoureit/nodejs-poolController/files/8181417/replay.2.zip)

Replay changing on the panel

rstrouse, 2022-03-03:
Pull njsPC I believe the heat modes should be showing up now.   I an not completely sure about the Heater Status though quite yet.

terafin, 2022-03-03:
Seems to work a charm! I’ll play with this a little more this evening post work if that’s ok?

rstrouse, 2022-03-04:
Yeah no worries.  Let me know if I got the heater status correct.

terafin, 2022-03-04:
I can confirm it’s working a charm, including heater status! Thank you!!

rstrouse, 2022-03-05:
I am going to close this for now.  If you have further issues or questions submit a new issue.

closed 2022-03-05 by rstrouse