tagyoureit/nodejs-poolController #393: Hayward TriStar VS900 Pump Support

nate-marshall, 2022-03-11:
**Describe the bug**
I am working on trying to add support for a Hayward TriStar VS900 pump using notes from http://www.desert-home.com/2014/07/controlling-hayward-ecostar-pump.html. I have an Aquarite SWG configured and working properly. I have wired my Hayward VS900 into the same RS485 adapter, but I am only seeing packets from the chlorinator in the logs. 

**Expected behavior**
Support for controlling Hayward VS Pumps with a Nixie Standalone Controller

**Packet Capture**
{"id":11839,"valid":true,"dir":"in","proto":"chlorinator","for":[11838],"pkt":[[128,128,0,0,0,128,128,128,0,0,128,0],[],[16,2,0,1],[0,0],[19,16,3]],"ts": "2022-03-11T14:50:00.100-0600"}
{"id":11840,"valid":true,"dir":"out","proto":"chlorinator","pkt":[[],[],[16,2,80,17],[0],[115,16,3]],"ts":"2022-03-11T14:50:03.016-0600"}
{"id":11841,"valid":true,"dir":"in","proto":"chlorinator","for":[11840],"pkt":[[128,0,0,128,0],[],[16,2,0,18],[68,0],[104,16,3]],"ts": "2022-03-11T14:50:03.085-0600"}
{"id":11842,"valid":true,"dir":"out","proto":"chlorinator","pkt":[[],[],[16,2,80,20],[0],[118,16,3]],"ts":"2022-03-11T14:50:03.421-0600"}
{"id":11843,"valid":true,"dir":"in","proto":"chlorinator","for":[11842],"pkt":[[],[],[16,2,0,3],[0,65,113,117,97,82,105,116,101,32,32,32,32,32,32,32,32],[49,16,3]],"ts": "2022-03-11T14:50:03.490-0600"}
{"id":11844,"valid":true,"dir":"out","proto":"chlorinator","pkt":[[],[],[16,2,80,0],[0],[98,16,3]],"ts":"2022-03-11T14:50:06.531-0600"}
{"id":11845,"valid":true,"dir":"in","proto":"chlorinator","for":[11844],"pkt":[[128,128,0,128,128,128,128,128,0,0,128,0],[],[16,2,0,1],[0,0],[19,16,3]],"ts": "2022-03-11T14:50:06.585-0600"}
{"id":11846,"valid":true,"dir":"out","proto":"chlorinator","pkt":[[],[],[16,2,80,17],[0],[115,16,3]],"ts":"2022-03-11T14:50:06.958-0600"}
{"id":11846,"valid":true,"dir":"out","proto":"chlorinator","pkt":[[],[],[16,2,80,17],[0],[115,16,3]],"ts":"2022-03-11T14:50:07.977-0600"}
{"id":11846,"valid":true,"dir":"out","proto":"chlorinator","pkt":[[],[],[16,2,80,17],[0],[115,16,3]],"ts":"2022-03-11T14:50:09.091-0600"}
{"id":11847,"valid":true,"dir":"in","proto":"chlorinator","for":[11846],"pkt":[[0,128,128,128,0,128,128,0,128,128,128,128,128,0,0,128,0],[],[16,2,0,18],[68,0],[104,16,3]],"ts": "2022-03-11T14:50:09.143-0600"}
{"id":11848,"valid":true,"dir":"out","proto":"chlorinator","pkt":[[],[],[16,2,80,20],[0],[118,16,3]],"ts":"2022-03-11T14:50:09.467-0600"}
{"id":11849,"valid":true,"dir":"in","proto":"chlorinator","for":[11848],"pkt":[[],[],[16,2,0,3],[0,65,113,117,97,82,105,116,101,32,32,32,32,32,32,32,32],[49,16,3]],"ts": "2022-03-11T14:50:09.543-0600"}
{"id":11850,"valid":true,"dir":"out","proto":"chlorinator","pkt":[[],[],[16,2,80,0],[0],[98,16,3]],"ts":"2022-03-11T14:50:12.573-0600"}
{"id":11851,"valid":true,"dir":"in","proto":"chlorinator","for":[11850],"pkt":[[128,0,0,128,0],[],[16,2,0,1],[0,0],[19,16,3]],"ts": "2022-03-11T14:50:12.619-0600"}
{"id":11852,"valid":true,"dir":"out","proto":"chlorinator","pkt":[[],[],[16,2,80,17],[0],[115,16,3]],"ts":"2022-03-11T14:50:12.940-0600"}
{"id":11852,"valid":true,"dir":"out","proto":"chlorinator","pkt":[[],[],[16,2,80,17],[0],[115,16,3]],"ts":"2022-03-11T14:50:13.962-0600"}
{"id":11852,"valid":true,"dir":"out","proto":"chlorinator","pkt":[[],[],[16,2,80,17],[0],[115,16,3]],"ts":"2022-03-11T14:50:15.007-0600"}
{"id":11853,"valid":true,"dir":"in","proto":"chlorinator","for":[11852],"pkt":[[128,128,128,128,0,0,128,0],[],[16,2,0,18],[68,0],[104,16,3]],"ts": "2022-03-11T14:50:15.065-0600"}
{"id":11854,"valid":true,"dir":"out","proto":"chlorinator","pkt":[[],[],[16,2,80,20],[0],[118,16,3]],"ts":"2022-03-11T14:50:15.473-0600"}
{"id":11855,"valid":true,"dir":"in","proto":"chlorinator","for":[11854],"pkt":[[],[],[16,2,0,3],[0,65,113,117,97,82,105,116,101,32,32,32,32,32,32,32,32],[49,16,3]],"ts": "2022-03-11T14:50:15.528-0600"}

**Pool Equipment**
 - Controller: 
 - Controller Model:
 - Controller Firmware Version:
 - Pump(s) manufacturer and model: Hayward TriStar VS900
 - Chlorinator: Hayward Aquarite
 - Heater(s): 
 - Chemical controller:
 - Valves:
 - Any other relevant equipment: USB to RS485 Adapter - Industrial USB to RS485 Converter Adapter Original FT232RL

rstrouse, 2022-03-11:
Ok so the Desert Home site references an Eco Start pump.  The TriStar looks like a different pump and operates on a percentage instead of RPM.  Can you confirm the TriStar functions on percentage or is it using RPM?

nate-marshall, 2022-03-11:
From what I can see from the pump controller, it is using RPM. For example, in the configuration menu, the four SPEED buttons can be set to specific RPMs. Is there another way that I should be validating this?

rstrouse, 2022-03-11:
I knew there were some that have decoded the protocol but some of them have been working at the drive level.  I am not interested in decoding that protocol since hookup is so invasive and it runs a 1200 baud.  However, it does appear that the speeds are actually percentage of 3450.  It is a little idiotic in that the underlying protocol between the motor and the drive uses RPM.  It does appear that the TriStar VS and the EcoStar VS use the same message.

I will add the message signature over the next couple of days as well as an interface to add, configure and send this message to the pump.   After that we will need to do a bit of cleanup on the code to make sure we are trapping all information related to the output.

nate-marshall, 2022-03-11:
Thanks for the update. Happy to test or assist where I can. 🙏

rstrouse, 2022-03-11:
What kind of coding experience do you have?  I am setting this up so that we will be sending messages to the pump.  I am not sure what our responses will be like.  If you have not seen it you can monitor the comms with messageManager.  Take a moment and watch the messages coming over the wire for your chlorinator.  We will be adding some for the pump.

https://github.com/rstrouse/nodejs-poolController-dashPanel/wiki/Message-Manager

rstrouse, 2022-03-12:
I coded the protocol as I know it.  I have some testing to do since the Hayward header is a bit of a strange one compared to other manufacturers.  I need to test it against my Pentair system.  As a bonus I took a moment to add the Jandy comms headers as well.  Maybe somebody with AquaLink will do some captures in the future.

rstrouse, 2022-03-12:
I checked in the changes.  Shut down njsPC and dashPanel then pull njsPC and dashPanel.  Then restart both projects.
`cd ~/nodejs-poolController`
`git pull`
`npm i`
`cd ~/nodejs-poolController-dashPanel`
`git pull`
`npm i`

Hard refresh your dashPanel browser.  Then under the pump configuration tab add your pump and select the Hayward Eco/TriStar pump type.  You will need to supply the address as this implementation supports up to 16 of these type of pumps.

Then add the circuit speeds to the pump and press `Save Pump`.  It should start sending messages to the pump based upon the speed settings and the currently running circuits on Nixie.
[image]

When you are done adding it I would appreciate another replay and if the work done by others is accurate your pump should operate.

nate-marshall, 2022-03-12:
I updated njsPC and dashPanel but I am seeing an error when trying to contact my VS900 pump. Replay: [12MAR22_2_replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/8238354/12MAR22_2_replay.zip)

The controller is wired into the same ports that I have the nsjPC wired into so I disconnected that from the pump thinking that they might be competing. That didn't seem to change anything. 

This is the [wiring diagram from Hayward](https://www.hayward-pool.com/assets/documents/pools/pdf/manuals/tristar-vs-900.pdf) and I see on section 5.4 that the first dip switch is turned off on all addresses if the pump is going to be controlled remotely. I am actually using that 12V line to power my PiZero with a buck converter to drop that down to 5V. When I flip that first dip switch off, the 12V gets turned off. I can find a different power source, but I wanted to get your thoughts on whether or not this is required. 

```
[3/12/2022, 2:22:12 PM] info: {"id":1660,"valid":true,"dir":"out","proto":"hayward","pkt":[[],[],[16,2,12,1,0],[0],[0,31,16,3]],"ts":"2022-03-12T14:22:12.007-0600"}
[3/12/2022, 2:22:12 PM] verbose: Wrote packet[16,2,12,1,0,0,0,31,16,3].Retries remaining: 1
[3/12/2022, 2:22:13 PM] silly: Retrying outbound message after 1.022 secs with 1 attempt(s) left. - 16,2,12,1,0,0,0,31,16,3
[3/12/2022, 2:22:13 PM] info: {"id":1660,"valid":true,"dir":"out","proto":"hayward","pkt":[[],[],[16,2,12,1,0],[0],[0,31,16,3]],"ts":"2022-03-12T14:22:13.033-0600"}
[3/12/2022, 2:22:13 PM] verbose: Wrote packet[16,2,12,1,0,0,0,31,16,3].Retries remaining: 0
[3/12/2022, 2:22:13 PM] debug: Begin sending chlorinator messages Aquarite
[3/12/2022, 2:22:14 PM] silly: Retrying outbound message after 1.032 secs with 0 attempt(s) left. - 16,2,12,1,0,0,0,31,16,3
[3/12/2022, 2:22:14 PM] warn: Message aborted after 2 attempt(s): 16,2,12,1,0,0,0,31,16,3
[3/12/2022, 2:22:14 PM] error: Error sending setPumpRPM for VS900: Message aborted after 2 attempt(s): 16,2,12,1,0,0,0,31,16,3
```

[image]

rstrouse, 2022-03-12:
Turn on the pool body and see if the pump actually sets the speed to 1725.   The message format that is being sent is the exact one that is identified in the Desert Home discussion.

nate-marshall, 2022-03-12:
I did try to run the pool but the pump doesn't start. I will work on rerouting the power to my Pi so that I can switch all dip switches in the off position.
[image]

rstrouse, 2022-03-12:
I do not believe this is a power issue with the pi.  Look at p 17 of the manual you linked.  This includes the dip switch settings.  I believe you need to disable the local control panel.  Your pump needs to be set up as if it is connected to an automation system.

nate-marshall, 2022-03-12:
Yeah, when I flip the #1 dip switch to the off position, the power from the `+12V` connector on the `[+12V, A, B ,COM]` drive low voltage connectors turns off. That is where I lose my power to my Pi/njsPC . I am working around that to test with all dip switches in the off position. I will let you know what I find out.

rstrouse, 2022-03-12:
Ahhh.  Now I am up to speed.

nate-marshall, 2022-03-12:
I hooked power directly up to my Pi, set up the pump on Address 1 in the dashPanel, and set all of the dip switches to OFF. However, I am seeing the same error. I tried to flip the no. 2 dip switch to ON and update the pump address to Address 2 in the dashPanel, but no change. I didn't grab a replay while doing this since the packet logs didn't seem to change at all and the pump didn't kick on.

Replay: [12MAR22_3_replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/8238519/12MAR22_3_replay.zip)  Please note that when the chlorinator was using the same `+12V` connector so you will see errors in the replay when connecting to the chlorinator. You can disregard those errors.

rstrouse, 2022-03-12:
Hmmm... we are sending the exact message that has been identified.   Perhaps try swapping the D+/D- wires on the pump.  This should be responding to our requests unless the multiple sources where folks have gotten it to communicate are wrong.  

I assume there is nothing connected to the digital input block and all dip switches are off (especially sw 1).  Also there is nothing else connected to the RS485 connector.

rstrouse, 2022-03-12:
Btw the code that sends this message starts on line 725 of `/controller/nixie/pumps.ts` the function name is setPumpRPMAysnc.

nate-marshall, 2022-03-12:
I tried swapping the D+/D- wires on the pump that didn't fix it and, correct, nothing is connected to the digital input block. I have an Aquarite SWG connected to the same usb-rs486 adapter. Though I had it disconnected when I was testing just the pump with all dip switches in the off position. 

I found a different error during the initial startup of njsPC. `error: Error sending setPumpRPM for VS900: Comms port is not open.Message aborted: 16,2,12,1,0,0,0,31,16,3`. I had to step away from my pool pad so I put everything back together the way it was before. I will have some time tomorrow to mess around with the wiring and make sure that the SWG is fully out of the picture. 

```
pi@poolcontroller0:~/nodejs-poolController $ node dist/app.js
Init state for Pool Controller
[3/12/2022, 5:37:38 PM] info: The current git branch output is master
[3/12/2022, 5:37:38 PM] info: The current git commit output is 88d18b97d85cf68f8851c7ad5dfd5920f2887a87
[3/12/2022, 5:37:39 PM] info: Starting up SSDP server
[3/12/2022, 5:37:39 PM] info: Init mqtt interface: HomeAssistant MQTT
[3/12/2022, 5:37:39 PM] info: Checking njsPC versions...
[3/12/2022, 5:37:39 PM] info: Starting Pool System nixie
[3/12/2022, 5:37:39 PM] info: Initializing Nixie Control Panel for Nixie Single Body
[3/12/2022, 5:37:39 PM] info: Initializing Intake/Return valves
[3/12/2022, 5:37:39 PM] info: Initializing Nixie Controller
[3/12/2022, 5:37:39 PM] info: Initializing Nixie body Pool
[3/12/2022, 5:37:39 PM] info: Initializing Filter Filter
[3/12/2022, 5:37:40 PM] info: Initializing Nixie circuit Pool
[3/12/2022, 5:37:40 PM] info: Initializing Nixie chlorinator AquaRite
[3/12/2022, 5:37:40 PM] info: Initializing Nixie Pump 50-VS900
[3/12/2022, 5:37:40 PM] info: Nixie Controller Initialized
[3/12/2022, 5:37:40 PM] info: Nixie Single Body control board initialized
[3/12/2022, 5:37:40 PM] info: Server is now listening on 0.0.0.0:4200
[3/12/2022, 5:37:40 PM] warn: Comms port is not open.Message aborted: 16,2,80,0,0,98,16,3
[3/12/2022, 5:37:41 PM] warn: Comms port is not open.Message aborted: 16,2,12,1,0,0,0,31,16,3
[3/12/2022, 5:37:41 PM] warn: Message aborted after 0 attempt(s): 16,2,12,1,0,0,0,31,16,3
[3/12/2022, 5:37:41 PM] error: Error sending setPumpRPM for VS900: Comms port is not open.Message aborted: 16,2,12,1,0,0,0,31,16,3
[3/12/2022, 5:37:41 PM] info: Serial port: /dev/ttyUSB0 request to open successful
[3/12/2022, 5:37:43 PM] info: {"id":3,"valid":true,"dir":"out","proto":"chlorinator","pkt":[[],[],[16,2,80,17],[0],[115,16,3]],"ts":"2022-03-12T17:37:43.881-0600"}
[3/12/2022, 5:37:44 PM] info: New socket client connected VelXdCdniv0Ami0bAAAB -- 192.168.1.165
[3/12/2022, 5:37:44 PM] info: MQTT connected to mqtt://192.168.1.109:1883
[3/12/2022, 5:37:44 PM] info: {"id":5,"valid":true,"dir":"in","proto":"chlorinator","pkt":[[128,0,0,128,0],[],[16,2,0,18],[68,0],[104,16,3]],"ts": "2022-03-12T17:37:44.464-0600"}
[3/12/2022, 5:37:44 PM] info: {"id":4,"valid":true,"dir":"out","proto":"hayward","pkt":[[],[],[16,2,12,1,0],[0],[0,31,16,3]],"ts":"2022-03-12T17:37:44.591-0600"}
[3/12/2022, 5:37:45 PM] info: [5:37:45 PM] 192.168.1.165 GET /state/all?null {}
[3/12/2022, 5:37:45 PM] info: Auto-backup initialized Last Backup: 2022-02-28T08:59:34.658-0600 Next Backup: 2022-03-14T09:59:34.658-0500
[3/12/2022, 5:37:45 PM] info: {"id":4,"valid":true,"dir":"out","proto":"hayward","pkt":[[],[],[16,2,12,1,0],[0],[0,31,16,3]],"ts":"2022-03-12T17:37:45.617-0600"}
[3/12/2022, 5:37:46 PM] info: New socket client connected z0B-YMGk5nm3xE2wAAAD -- 192.168.1.165
sendLogMessages set to true
[3/12/2022, 5:37:46 PM] warn: Message aborted after 2 attempt(s): 16,2,12,1,0,0,0,31,16,3
[3/12/2022, 5:37:46 PM] error: Error sending setPumpRPM for VS900: Message aborted after 2 attempt(s): 16,2,12,1,0,0,0,31,16,3
[3/12/2022, 5:37:46 PM] info: {"id":6,"valid":true,"dir":"out","proto":"chlorinator","pkt":[[],[],[16,2,80,20],[0],[118,16,3]],"ts":"2022-03-12T17:37:46.782-0600"}
[3/12/2022, 5:37:46 PM] info: {"id":7,"valid":true,"dir":"in","proto":"chlorinator","pkt":[[128,0,128,128,0,0,128,0],[],[16,2,0,3],[0,65,113,117,97,82,105,116,101,32,32,32,32,32,32,32,32],[49,16,3]],"ts": "2022-03-12T17:37:46.852-0600"}
[3/12/2022, 5:37:48 PM] info: {"id":8,"valid":true,"dir":"out","proto":"hayward","pkt":[[],[],[16,2,12,1,0],[0],[0,31,16,3]],"ts":"2022-03-12T17:37:48.692-0600"}
[3/12/2022, 5:37:49 PM] info: {"id":8,"valid":true,"dir":"out","proto":"hayward","pkt":[[],[],[16,2,12,1,0],[0],[0,31,16,3]],"ts":"2022-03-12T17:37:49.708-0600"}
[3/12/2022, 5:37:50 PM] warn: Message aborted after 2 attempt(s): 16,2,12,1,0,0,0,31,16,3
[3/12/2022, 5:37:50 PM] error: Error sending setPumpRPM for VS900: Message aborted after 2 attempt(s): 16,2,12,1,0,0,0,31,16,3
[3/12/2022, 5:37:50 PM] info: {"id":9,"valid":true,"dir":"out","proto":"chlorinator","pkt":[[],[],[16,2,80,0],[0],[98,16,3]],"ts":"2022-03-12T17:37:50.925-0600"}
[3/12/2022, 5:37:50 PM] info: {"id":10,"valid":true,"dir":"in","proto":"chlorinator","pkt":[[0,0,0,128,252,128,0,128,0],[],[16,2,0,1],[0,0],[19,16,3]],"ts": "2022-03-12T17:37:50.977-0600"}
[3/12/2022, 5:37:51 PM] info: {"id":11,"valid":true,"dir":"out","proto":"chlorinator","pkt":[[],[],[16,2,80,17],[0],[115,16,3]],"ts":"2022-03-12T17:37:51.295-0600"}
[3/12/2022, 5:37:51 PM] info: {"id":12,"valid":true,"dir":"in","proto":"chlorinator","pkt":[[],[],[16,2,0,18],[68,0],[104,16,3]],
```

nate-marshall, 2022-03-13:
I did a little more digging and found [this issue](https://github.com/Here-Be-Dragons/Pool-Controller/issues/4) that seems to have good information on Hayward VS pumps and also piggybacks off of the information from the [desert-home post](http://www.desert-home.com/2014/07/controlling-hayward-ecostar-pump.html). I should be able to take the hayward updates you made for this and test with the other commands in that issue.

rstrouse, 2022-03-13:
I saw that issue but bear in mind there are two protocols on the pump
one of them communicates this way
`automation <--> pump drive`
and the other this way
`pump drive <--> motor`
The later communicates at 1200 baud and it is hooked up differently yet most of the conversations regarding this protocol and hookup seem to start in the middle.  However, I believe what you are looking for is the protocol that starts with (16, 2 [0x10, 0x02])

rstrouse, 2022-03-18:
@nate-marshall so I re-read all of the writings on this topic and it seems that those using this pump have set the baud rate to 19200.   So lets see if we can get the pump to talk by doing the following.

Shut down njsPC and in the `config.json` file change the `baudRate` value under the `controller --> comms --> portSettings` section to 19200.  I'd be curious to see if the chlorinator can adjust for this speed.

rstrouse, 2022-03-20:
If you update your njsPC and dashPanel you can now edit the port speed in the configuration.

nate-marshall, 2022-03-23:
@rstrouse I tested the pump and chlorinator on 19200 baud rate, but neither worked on this setting. I appreciate the work to get this setting added to the dashPanel. That made it really easy to test the different baud rates quickly. Side note, I haven't had the time to rewire my RPi with njsPC that is connected to my SWG and pump. Hoping to get that this weekend. 

[19MAR22_0_replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/8329319/19MAR22_0_replay.zip)

rstrouse, 2022-03-24:
If you had the Aquarite hooked up during this then it clearly does not like 19200 baud rate since it is not responding.  It was responding when the port was set to 9600.  I did see something out there where a ground was required between the ground of the RS485 port and the drive controller.  The pump wouldn't respond to RS485 without this connect.  It is in the manual somewhere.

I haven't created the assignments yet for RS485 devices to use Aux ports but when I get that done you will be able to have multiple ports running at different speeds.  The advantage here is that if the Hayward pump only runs at 19200 then we can still run the aquarite at 9600.

rstrouse, 2022-03-24:
Btw if you pull njsPC and dashPanel you can now assign the pump to a port.  So if you need 2 for instance because the pump runs at a different baud rate you can assign the pump to an aux port that matches the port settings for the pump.  It is a setting in the pump setup.

rstrouse, 2022-03-28:
@nate-marshall have you tried it with 2 stop bits yet?  Unfortunately, I do not have one of these pumps but worst case scenario you can define the pump as a SuperFlo VS.  It will then use relays to control 3 speeds plus override.  Nixie will engage the relays with the appropriate circuits.  We could even set it up for relay control to allow up to 8 speeds but you will have to set the speeds on the pump and Nixie will only be able to engage whatever speed you have set up.

nate-marshall, 2022-03-29:
@rstrouse I missed your edit on the 2 stop bits. I tried testing (19200 baud, no parity, 8 data bits, 2 stop bits) with the SWG wired up on the bus and that didn't work, but I assumed that it wouldn't. I will test the pump out tomorrow without the SWG and let you know what I find.

rstrouse, 2022-03-29:
The most recent software will now manage two ports on the same controller.  You can assign the port on the pump confirmation page. The chlorinator mist still run on the primary port but I will be adding code to allow it to operate on any open port.

rstrouse, 2022-03-31:
@nate-marshall progress.  I added the ability to assign any chlorinator to any port in the configuration so you can create as many ports running at whatever speed, stop bits, or data bits that you like.  It will also pull all those ports together so that the pump and chlorinator will act like they are on the same comm port but will be managed differently.  So if you get the pump working at 19200 and the chlorinator at 9600 then you are in business.

I also added a Hayward relay pump type that will switch the relays for up to 8 assigned speeds if you want to hook up the 4 relays to a relay board.

nate-marshall, 2022-04-09:
@rstrouse I was able to separate the SWG and pump onto two separate rs485 adapters, reconfigure njspc/dashpanel, and it is working! I am able to control both the SWG and pump (19200 baud, no parity, 8 data bits, 2 stop bits), but one thing I am noticing is that I am not seeing the pump return data like the SWG does and not seeing data in the dashpanel. I can turn the pump on and adjust the speeds, but only know it is doing what I ask it by watching/listening to the pump. Thoughts on that?

I am super excited to finally see the pump accepting commands! 

[image]

[09APR22_00.zip](https://github.com/tagyoureit/nodejs-poolController/files/8457744/09APR22_00.zip)

rstrouse, 2022-04-10:
Awesome!  I'll take a look at the replays to see what the pump returns are saying.

sdpools, 2022-04-12:
Hello,

I have a Hayward vsp3400.

There are three ways to control the pump and issues with each which one IRT to pentair connect.

First the Hayward Ecostar vsp3400 is the pump used in the dessert home page. 

There are some main differences between a Hayward Ecostar and Tristar:

Ecostar:
1. Offers terminals to the motor bus via the 1200 baud rate labeled combus (pins 7&8). I think this is the speed in percent sent from the wall or motor mounted driver display.

2. Offers rs485 via a 6 pin header to the wall/ motor mount drive controller display. 
Pins 1,2,3 (TS 485) and pin 4(12 VDC) to power display ICs and LCD. Pins 5 and 6 not used.

3. Relay control. "Digital inputs"
INP1, INP2, INP3. To use this you must use PWR OUT (12 VAC) terminals.
Testing with a multimeter shows that 6 volts AC sits on one terminal and 8 volts AC on the other.

The INP is a logic high when the board sees the full 12-14VAC.

4. INP 4. This the fireman cutoff and also controller by 14VAC trigger.

5. No dip switches, so only one pump can be controlled. (Not a problem for me and I would take one or none)

Problems:

1. The motor bus needs a signal ever second which he said was crappy.

2. Rs-485 is the protocol he ended up using because it's on the only one that will work with 24/7 powered pump.

3. Relay controls. I have my pump currently wired for this and it doesn't work if your pump is hardwired to be always be on.

The relay matrix has positions for three relays corresponding to the eight speeds. Problem is all relays off still equals a speed. 

So the pump is always running. You can change speeds but it does not turn off.

The relay setup is requiring that you use the filter relay to slam power on and off to the pump. (Nope - my pump gets testy when turned off and on)

4. Relay Work around for always powered pump: I have purchased a normally closed relay. I plan to wire it to pentair filter relay contact For trigger. The contacts will be closing the fireman stop switch (INP 4). 

Theory is when the pump is not supposed to running (off schedule) the stop command will over ride the relays speed settings.

When the pentair units turns on filter relay- my fireman stop which will be open. The pump will then sense a relay speed (from matrix) and start, the new feature is that it will actually turn off again.

This is not ideal, as I wonder if the stop command is a hard stop and will damage the pump over time?

I will test and report back.

The tristar 3200:

1. Deleted drive com header from pcb. Manuals shows external com over the rs485 now.

2. Changed rs485 to 4 pin header (vs 6 on Ecostar) - removed pins 5/6 where not used.

3. Removed 14VAC over two pins as the trigger input. 

I think it is now just 12 vDC from "ICOM".

Still doesn't have way to turn pumps off via relay.

Why didn't they make off,off, off =off? They really wanted that last speed option...

sdpools, 2022-04-12:
Does controller have a way to send the the hayward rs-485 commands?

I have a second Rs-485 adapter to use for the Hayward link baud.

The only option I saw for Hayward was relay control.

Thank you.

rstrouse, 2022-04-12:
You also have an IntellTouch is that correct?  I don't know if we made that available on the IT but it is available on the IntelliCenter and Nixie to control the pump using RS485 on another port.  We could add it for you though and it must be installed on an auxiliary port because it would confuse IntelliTouch.

We did add the relay control to IntelliTouch but also use 4 relays to enable the pump so you can also use that option for up to 8 speeds + off for the pump.  The On/Off relay needs to be a 3hp relay that will disconnect power from the pump when using relay control.
[image]

sdpools, 2022-04-13:
Thank you! I will try soon. 

Today I hooked up the NC relay and activated the relay driven IPN4 fireman pump stop.

This is a super hard logic stop and not just a simple soft "stop" command one gets via 485. (Not surprised)

So, although it is possible to control any VS pump via full relay  and leave it powered on 24/7 - it is not advisable. Slamming 220vac to a vsp via the filter pump is also not advisable.

Conclusion: relay pump control is for the birds. Fun project. 

Unfortunately, while disconnecting the rest relay (it was powered), the two contacts touched and let some of the "magic out of the lamp". Haha. Oops.

The Intellitouchs i7+3 3amp circuit breaker is hard tripped. Out of curiosity I fired the up the unit and I have 485 comms & 24VDC to my actuators, so I can manually move em.

I took it apart and have smoked two of the three ULN2803A relay IC chips. Go Go Amazon: 10 chips for 7 bucks (generic), two day shipping. I will be back in business on Friday.

I plan on putting things back together before then and pulling the update you provided. 

Since I'm up on RS-485 bus the intellitouch relay ICs being down should not effect the pump RS-485 testing.

I just got distracted by some smoke :)

I'll update again soon when I have time to re-wire the Hayward pump for Rs-485 again.

I disconnected the 485 wires to use them for the full relay setup testing.

Be back soon. 

Thanks for all the help.

rstrouse, 2022-04-13:
Good thing those darlington chips are socketed.

rstrouse, 2022-04-14:
@nate-marshall I am only seeing one side of the comms with the pump.  Is it possible that the adapter is bad or your wiring has issues.  The output from njsPC is being sent but there is radio silence from the pump and there should be a message coming back as described by all the others before us.  There are just no bytes being sent back.

sdpools, 2022-04-21:
Hello Fellas,

Update. So the new LN2803A (Toshiba) chips came in. All is well and the Intellitouch is back in action. 

I update the PC and Dash, the hayward option was present and I selected it. Like @nate-marshall, I have no comms on the VS300VSP "COMBUS" port. 

The software is sending the pump code- the pump isn't talking.

Out of curiosity, I did a voltage check. Pin 7, is said to +, and Pin 8 is -.
Across the two I measure ~2.5VDC stable. I am used to seeing a variable voltage as TX is used.

I sniffed the signal at all baud rates and I get nothing...

Ohmed it out and it was high 6 Ohms (stranded CAT5e). Replaced it with solid copper core 22 AWG? wire. Ohns dropped to ~2 Ohms. Still no signal.

I swapped the RS-485 units (3 total in testing). Nothing.

I do get a signal when I tap the wall mount display RS-485 stream. 

I tried sending copying and pasting the pump 45% and 100% signals to both the "wall display" and "Combus" ports with no reaction from the pump or display unit.

My firmware is:
Drive Rev: 1.10
Display Rev: 2.70

I think they need to be updated.

I signed up for a hayward account - however - the "consumer" account does not have firmware.

So, I am at a loss. I will continue to look into updating the firmware (separate post). 

Until then, I am out of ideas.

rstrouse, 2022-04-21:
I saw somewhere that the control panel needs to be disconnected.  I don’t have a specimen to play with.

sdpools, 2022-04-21:
That is a good suggestion. I will give it a try tomorrow and report back.

rstrouse, 2022-05-03:
@nate-marshall did you ever get a response from the TriStar pump?

nate-marshall, 2022-05-03:
Unfortunately, no. Over the weekend I replaced my rs485 USB adapter as well as checked the continuity of the wires from the pump to the adapter. Everything checked out, but still nothing back from the pump.

sdpools, 2022-05-03:
@rstrouse,

I did immediately make a post on TFP. My post went to into review - but then never made it live.

I am guessing because I was asking about firmware upgrades (outside of a Hayward network / dealership) my post was sunk (just never posted).

I really appreciate your guys hard work on this project and apologize for my drop off in communication.

Our family has just completed 3 straight weeks of in home visitors that kicked off for Easter. We all got sick for a week from family member #1 (stomach bug)- then a death in the family brought 2 more unplanned family visitors - a project at work went into mandatory OT for 8 days - 3 family birthdays (in a row)- contractors at the property / house got painted - Lastly, a third group of family members came because they missed Easter due to being sick. 

Life has slowed back down as of yesterday as the last visitor flew home.

I'm still at a lose for comms on my pump and think the TTL chip on the pump board responsible is either cooked or needs a firmware upgrade.

I didn't try unplugging the drive / detachable wall controller and testing again yet. 

The Dessert home guy was using his pump with a Hayward control panel (Goldline I think)? In that setup they may move the wall controller to the gold line controller so you may be on to something.

I am working from home tomorrow and should be able to make this quick test.

Thanks again guys.

sdpools, 2022-05-03:
Ok guys, I did some reading and found this:

The thread states a few things from
a Hayward rep:

1. The drive firmware is not upgradable. The detachable display screen is also not upgradable.

2. Hayward Ecostars built before 2015 are not compatible with haywards own line of controllers.

My pumps serial numbers are located  on the back side of basket across from the 2" pump outlet pipe.

https://www.troublefreepool.com/threads/omnilogic-not-talking-to-ecostar-firmware-issue.91733/

My unit is Feb-2013.

However Dessert Homes article was posted July 2014.

Im wondering if a few of us have old pumps still kicking that don't have the proper firmware to support remote comms?

Mine actually did die (I did a drive rebuild on mine because it failed 2 yrs back)

It would be helpful for others to post there firmware builds and serial build date figure out what bucks dates will work.

rstrouse, 2022-05-05:
Sorry for your loss... and the apparent flash mob.

What the heck is going on with TFP.  I thought they were manufacturer agnostic.  Kinda kills the spirit of the site if that's the case.  Then again I had a post that got whacked because my opinion differed from the person with the reject button.

Booh to Hayward on that.  We do know that it receives data just fine it just doesn't talk back.  I have to wonder if they simply omitted the pull ups on the transmit line.  I am an armchair hardware guy so I wouldn't even know where to begin to troubleshoot that though. The RS485 chip is probably one of the few chips out there that do this function so either their firmware isn't pushing or their hardware isn't sending.  How hard can this be it only sends a single message.

I did however, implement the multi-relay functions that will allow you to assign 8 speeds + off to the pump and Nixie will switch it appropriately. This is the Hayward Relay VS pump type.

rstrouse, 2022-05-29:
@nate-marshall - Hey I think there may have been some confusion with the message headers related to this pump.  Another user pointed out that the returns from the Tri/EcoStar pumps were being confused with the chlorinator messages.  We may have been getting the returns all along but the chlorinator processes were stealing them.

rstrouse, 2022-05-29:
@sdpools - Please read the previous post.  I think we may have found where the message returns were going.

nate-marshall, 2022-05-29:
@rstrouse I updated everything and it looks like the controller is having trouble communicating with the chlorinator. `Chlorinator Comms Error` 

When I updated, I noticed that I lost my pump circuits so I restored from my latest backup. I am still able to successfully send commands to the pump, but it looks like I am still not getting the responses from the pump. Should I try and rebuild from scratch?

[image]

[image]

[29MAY22.zip](https://github.com/tagyoureit/nodejs-poolController/files/8794376/29MAY22.zip)

rstrouse, 2022-05-29:
I’ll have a look at your replay this evening.

rstrouse, 2022-05-29:
I just took a quick look at the log.  The good news I’d that the pump responses are in the file.  The reason it is not displaying is down to a simple coding error that I will fix tonight.

rstrouse, 2022-05-30:
@nate-marshall - I posted up the changes so that njsPC should be reading the data correctly for your pump.  I had a bad adapter on my test rig so it my results were inconclusive last night.  Anyhow pull njsPC and perform a replay capture for me.  This should now be working but I need to make sure I have properly checked the termination for the Hayward protocol.

nate-marshall, 2022-05-30:
All updated and here is the requested replay. I am not seeing the error with my Chlorinator anymore. Thank you for fixing that!

[30MAY22.zip](https://github.com/tagyoureit/nodejs-poolController/files/8800854/30MAY22.zip)

rstrouse, 2022-05-30:
I see all the data from the pump in the replay.  The data looks perfect.  Are you not seeing the pump data?

It went from
3450 => 0 => 3450 is that what was done.

nate-marshall, 2022-05-30:
That is awesome! Is there a way to tell that the replies are coming through besides the indication from the pump status in the UI? 

[image]

rstrouse, 2022-05-30:
Yes I have found the issue with the display.  The pump address is not where I thought it was on the input.

rstrouse, 2022-05-30:
I am confused.  You have the pump set to address 2 which dip switch configuration do you have set.
[image]

nate-marshall, 2022-05-30:
I have #2 in the ON position and the rest in the OFF position (Aux 1/Spa Filter)

rstrouse, 2022-05-31:
Perfect.  I was sooooo confused given the message returns.  The other dudes only translated the outbound message.  The inbound is not what they expected with the address.  The pump has to tell us which one it is when it sends its information back.  It appears that is starts at 0 not 1.  So your returns are coming back as pump 1 but there is a pump 0 (Pool Filter).  Pull njsPC and we should be in business.

nate-marshall, 2022-05-31:
Beautiful! It is working. I was going down the road of assuming that this wouldn't work with my type of pump, but, Oy! I really appreciate the persistence on this! 🙇 

[image]

closed 2022-05-31 by nate-marshall

avbdr, 2022-06-27:
@nate-marshall im sorry to bump an old ticket, but is it a know issue that a tristar vsp pump is not responding to the commands for the first minutes after being power cycled? Around 10-15 minutes, sorry i haven't measured precisely. Then everything is working just fine. During this time pump does nothing.

tagyoureit, 2022-06-27:
Can you upload a replay of this happening. We can look and see if there is anything in the logs. Start the replay, then turn on the pump, wait for it to finally respond (+1 minute) then end the replay and upload it.

[7 more replies]