tagyoureit/nodejs-poolController #394: Attempting to lengthen the 15 minute SWG auto-dosing cycle vi ChemController.ts

johnny2678, 2022-03-12:
This is all part of my ongoing struggle to lessen the dramatic ORP swings when dynamic dosing is enabled.

Tried two different places in the code:

1) Line 1967 - https://github.com/tagyoureit/nodejs-poolController/blob/da6af8b88d151c1123b55ed4e8dcd6a578bf018c/controller/nixie/chemistry/ChemController.ts#L1967

@tagyoureit suggested on gitter - Change the `this.chlor.chlorInterval * 60` to `this.chlor.chlorInterval * 60 * n`.  The chlorInterval is the 15 min default.  So n=1.5 would be 22.5 mins, n=2 would be 30 mins, etc.

I changed to `this.currentMix.set({ time: this.chlor.chlorInterval * 60 * 8, timeMixed: 0 });`, but mixing time stayed at 15-`lastDoseTime`

2) Line 1952 - https://github.com/tagyoureit/nodejs-poolController/blob/da6af8b88d151c1123b55ed4e8dcd6a578bf018c/controller/nixie/chemistry/ChemController.ts#L1952

@tagyoureit also suggested - Look to line ~1952 (`let mixTime = Math.min(Math.max(this.chlor.chlorInterval * 60 - lastDoseTime, 0), this.chlor.chlorInterval * 60);`. You can change this to let mixTime = n; (where n is minutes). 

I changed to `let mixTime = 45;` - going for a 45 minute interval to test.

Both code changes mentioned above were in place during the replay capture I posted below.  Net result? After 23 seconds of SWG dosing, the mixing cycle was still set for `14min 38sec` (15mins - `lastDoseTime`)

also attaching my modded `ChemController.ts` - in case you want to check my code.  Line #'s are a bit off because of comments I have added, but should be close

Only other change I made to `ChemController.ts` was to add a max Daily lbs of Cl check at Line 658: https://github.com/tagyoureit/nodejs-poolController/blob/da6af8b88d151c1123b55ed4e8dcd6a578bf018c/controller/nixie/chemistry/ChemController.ts#L658

`if ((this.chem.orp.maxDailyVolume <= schem.orp.dailyVolumeDosed && !this.chem.orp.useChlorinator) || (schem.orp.dailyVolumeDosed > 0.12)) {`

===================================

Replay attached, sorry it's so long but it took a bit for my SWG to initiate this morning and clear the Low Flow warning.  I put some timestamps below that will hopefully help you seek the main events in the replay.

~8:26:10 - restart njspc

~8:12:00 - njspc back up and running

~8:13:19 - changed ORP setpoint to trigger SWG dosing cycle (current orp level was just below setpoint +20 so only a 23 sec dosing cycle)
```
[3/12/2022, 8:13:19 AM] info: Chlor trend line is 0.057079117379405284.
[3
[replay (30).zip](https://github.com/tagyoureit/nodejs-poolController/files/8237686/replay.30.zip)
/12/2022, 8:13:19 AM] info: Chlor dosing % of time is 2.57%
[3/12/2022, 8:13:19 AM] info: Chem chlor calculated dosing at 2.57% and will dose 0.000375Lbs of chlorine over the next 23sec.
```

---- NOT RELATED TO THIS REQUEST BUT YOU WILL SEE IN LOGS, SWG reporting low flow errors, happens sometimes in the morning when pump comes on, dosing paused until `Low Flow` condition clears ----

~8:31:54 - `Low flow` condition clears, SWG dosing actually starts, SWG @ 100%
`[3/12/2022, 8:31:54 AM] info: Chem Controller orp chlorinated 0lbs of 0.000375lbs - 23sec remaining`

~8:32:18 - SWG dosing ends
`[3/12/2022, 8:32:19 AM] info: Chem Controller begin mixing orp for 14min 38sec of 14min 38sec`

[replay (30).zip](https://github.com/tagyoureit/nodejs-poolController/files/8237711/replay.30.zip)

[ChemController.txt](https://github.com/tagyoureit/nodejs-poolController/files/8237710/ChemController.txt)

tagyoureit, 2022-03-14:
I should have told you it was line https://github.com/tagyoureit/nodejs-poolController/blob/master/controller/nixie/chemistry/ChemController.ts#L1903.  Change that one.

johnny2678, 2022-03-15:
```
[3/14/2022, 8:43:51 PM] info: Chem Controller begin mixing orp for 27min 49sec of 27min 49sec
[3/14/2022, 8:44:51 PM] info: Chem Controller acid mixing Complete after 15min
```

That mostly worked, initially the mix timer is set using the multiplier added to line 1903, but somewhere else along the line it must still be expecting a 15 min timer.  AFAIK, the timer stayed on the whole time and this is only in the logs, but not sure yet.

Will watch some more and try and get a replay.

johnny2678, 2022-04-04:
quick update:
I've reverted back to the Dynamic mixing default as defined by the code.  Instead I have focused on capping the `maxDailyORPVolume`.  This keeps my SWG from dosing too much when ORP gets driven down (ironically, from dosing).

I capped `maxDailyORPVolume` my modding the following line:
https://github.com/tagyoureit/nodejs-poolController/blob/fda8f46df08117948ce2e415a777395085b71e83/controller/nixie/chemistry/ChemController.ts#L658

changed it to:
`                if (this.chem.orp.maxDailyVolume <= schem.orp.dailyVolumeDosed) {   `

and added an MQTT binding.

Then I just started playing with different max levels.  Once the max is reached, dash shows the max Daily ORP limit warning (just like pH) and stops dosing.  Works a treat so far!  But I'm still observing...

A note on my dosing strategy... prior to all this with OOTB dynamic SWG dosing, i noticed that after a large dosing day, if I turned off dynamic dosing for a day and let it settle, ORP would usually rise back to above my ORP threshold on day 3.  So I started my dosing strategy like that - a dosing day, a recovery day, and a maintenance day.  

Rinse / Repeat

So the lbs of Cl dosing cycle looks like this:
[image]

But what if the sun is blazing hot (more Fcl drain), or it's a cloudy day (less Fcl drain)?  Trying to address that too.

Each half hour while the pump is on I look at the average UV from my PWS in my front yard.  Working with the following UV modifiers right now.  Not sure if this is where I will land but works for me so far:

- `UV < 3? then max( UV/3, 1) * maxDailyORPVolume`
- `UV >=3 && UV <6 then no modifier
- `UV >=6 then (UV/6) * maxDailyORPVolume`

So it adjusts the max based on the weather.  This has produced a nice daily ORP pattern that has kept FCl in the 3-5 range (which I'm perfectly happy with).

[image]

I'm only 2 weeks in so this could obviously change with more weather (haven't had a torrential FL rain yet), or seasons (will see how this stands up to FL summers).  But it's a start.

Will be monitoring with  this grafana graph:
[image]

where period 1 is the ORP history with OOTB dyanmic dosing enabled
period 2 is me fumbling around with the code and nodered to make adjustments
period 3 represents a bit of a steady state since I haven't had much time to play lately.  Will be interesting the this produces a more steady ORP, and FCl measurement over time.

Happy to hear any critique on my approach.  Was only able to implement because you guys told me where in the code I could make changes to enable `maxDailyORPVolume`

Will keep you posted after a few more weeks.

tagyoureit, 2024-06-10:
closing due to lack of activity.

closed 2024-06-10 by tagyoureit