tagyoureit/nodejs-poolController #395: EasyTouch4P Stopped working after updating to hash 65371696b4c2707d38ba699052da304e12d0770f

terafin, 2022-03-15:
**Describe the bug**
My pool lights and relays were working wonderfully (thanks again!) until I updated (silly me) to today's build.

Since that, all my commands from the web interface (and API) fail, but the UI/API do get updates from the device.

```
MessageMessage aborted after 4 attempt(s): 165,1,16,16,134,2,3,1,1,82
```

I reverted to a hash as of here: https://github.com/tagyoureit/nodejs-poolController/commit/c563c94a51d9616ab396fa32ab853f223262c80e

Verified things work flawlessly again.

**To Reproduce**
Steps to reproduce the behavior:
1. Update to latest mimi docker
2. Try to use circuits. Fails

**Packet Capture**
Follow the instructions to complete a [packet capture](https://github.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution) and attach the resulting zip/log files.
[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/8249273/replay.zip)

terafin, 2022-03-15:
Did some manual checkouts to help bisect, this is the commit that broke things for me:
https://github.com/tagyoureit/nodejs-poolController/commit/88d18b97d85cf68f8851c7ad5dfd5920f2887a87

rstrouse, 2022-03-16:
Sorry I did not see this message right away sometimes github doesn't send me info right away.  The issue was related to the added protocols for the RS485.  It seems that our calls for the outbound to ET was talking to itself.  Source 16 to destination 16.  Please pull njsPC and let me know how it goes.

terafin, 2022-03-16:
Fixed it, thanks!

closed 2022-03-16 by terafin