tagyoureit/nodejs-poolController #397: OutboundMessageError when adjusting RPM on Intelliflo VS

mhmatttu, 2022-03-18:
**Describe the bug**
I have an ET2-8 with an intellifo vs pump. all seems to be working well on 7.6.1 but i’m unable to modify pump settings. When i attempt to modify rpms under pump circuits I receive “OutboundMessageError” each time.

**Expected behavior**
RPM adjusts to new setting

**Screenshots**
[image]

**Packet Capture**
[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/8307725/replay.zip)

**Pool Equipment**
 - Controller: EasyTouch
 - Controller Model: EasyTouch2 8
 - Controller Firmware Version: ?
 - Pump(s) manufacturer and model: IntelliFlo VS and Intelliflo VF
 - Chlorinator: IntelliChlor-40
 - Heater(s): gas
 - Chemical controller: none
 - Valves:  hayward x2
 - Any other relevant equipment:

**Desktop (please complete the following information):**
 - OS: windows 10
 - Browser: chrome
 - Version: 99.0.4844.51

**Smartphone (please complete the following information):**
 - Device: iphone 13
 - OS: iOS 15.3.1
 - Browser: safari
 - Version: ?

tagyoureit, 2022-03-19:
Few questions:
1. You have a dual-speed pump as a third pump in your config.  Do you have one of these connected?
2. If you go into the config (upper right gears) and go to pumps and try to change/save the pump speeds there does it work?
3. Do you have an indoor, screenlogic or wireless control panel?

mhmatttu, 2022-03-19:
1. no, there are only the 2 pumps.  i was not sure what the dual-speed pump was for or how it got there
2. does not work in the config either...same error message
3. the ocp did have a wireless control panel but has been disconnected as the remote quit working

mhmatttu, 2022-03-19:
Screenshot when attempting to change in the pump config:
[image]

Screenshot when attempting to change from the home screen:
[image]

rstrouse, 2022-03-19:
Ok the reason why that two-speed pump is showing up is because you have high speed circuits defined in your ET even though there is no two-speed pump installed.  These should not be set up if you do not have such a pump.

You also appear to have turned on the heat pump setting on ET but according to your equipment list above there is no heatpump.  ET is frantically trying to communicate with your heatpump but it is not responding.

You should clean up your configuration.  ET unfortunately does not check for successful message responses in a lot of cases so it doesn't report anything even when it fails.  It will just do odd things like flip the high speed bit on the panel which then affects other things.

mhmatttu, 2022-03-19:
Thank you.  Do I clean up the configuration on the OCP or in njspc?  Can I simply delete the 2 speed pump from dash pane config-pump menu? Based on some the random schedules programmed, I think the previous owner was pressing all the buttons :).  Would this config clean up also resolve my VS pump issue?

mhmatttu, 2022-03-20:
Regarding the heater, I only see the gas heater in dash panel and in the poolConfig.json file

[image]

[image]

rstrouse, 2022-03-20:
Clean it up in the OCP.  Unfortunately some of the config is not directly reported to njsPC until the device reports.

rstrouse, 2022-03-20:
The reason for the two speed pump is because there are high speed circuits defined in the ocp.

mhmatttu, 2022-03-20:
my apologies, I failed to mention that there is a polaris booster pump.  could that be the reason for the 2 speed pump?  when the cleaner circuit is on, the light for the 2 speed pump does not come on

rstrouse, 2022-03-20:
No that is not a 2 speed pump.   The booster pump is defined as a master cleaner.  This is what turns on the booster pump and makes sure it is primed.

It looks like the two high speeds that were set up are Pool and Spa.  ET is trying to flip that secondary relay but I suspect there is no 2 speed pump relay that exists.

Also, it appears that the configuration is set up for on VS and one VF pump.  Can you confirm this is the case?

mhmatttu, 2022-03-20:
Correct, the main pump is a VS and the water feature pump is a VF.  

While I am making config changes, is there something I need to look at to resolve the ongoing time discrepancy?  The config is set to use the server time.  The rpi time is correct but dashpanel frequently gets out of sync.  I can only resolve by rebooting the rpi.

rstrouse, 2022-03-20:
Did you set the timezone and localization on the rpi?  This can be found in the rpi coniguration.

mhmatttu, 2022-03-20:
I have made the various config changes on the panel (removed heat pump, removed 2 speed pump, created speed settings on the VS pump, enabled DST under time settings).  I discovered that the 2nd pump for the water feature is a Whisperflo, not an Intelliflo VF.  However, it seems to be showing up under the Intelliflo submenu on the panel.  Is this normal behavior or should it show as a different kind of pump?   I believe it also has a blown capacitor because it will not come on.

I did have timezone/localization configured on the rpi and njsPC, but resubmitted on each.  I also added the GPS coordinates for njsPC.  I did notice that DST was not selected on the OCP so I enabled that as well.   I will monitor to see if this fixes it.  

With the pre-set speeds now setup on the VS pump, I can now select those from the home screen on Dash Panel but I am still not able to make adjustments from the home screen or settings.

[image]

In the below screenshot, I attempted to increase the pool rpm from 2,000 to 2,050 and received the error message

[image]

In the below, I attempted to increase the pool rpm from 2,000 to 2,010, clicked save and received the error message

[image]

I attached a packet capture of this activity.  Thank you so much for your help!

[replay (1).zip](https://github.com/tagyoureit/nodejs-poolController/files/8311923/replay.1.zip)

mhmatttu, 2022-03-21:
Still have an issue with the time as well.  I captured a screenshot that shows the time on njsPC is 12:01am yet the time on the rpi is 10:05am cst.  Included the njsPC settings in the screenshot.

[image]

tagyoureit, 2022-03-21:
Just to keep things clean, can you open up a new issue for the time questions?  We should be updating the time zone ever 30 minutes and the time every 5 mins (when it is off by 2+ minutes).  If you could do a capture from x:28-x:34 or x:58-(x+1):04 that should get the right information in the logs for me to look at.

rstrouse, 2022-03-22:
If you do not have an IntelliFlo VF pump installed in the second pump slot you must remove it.  The OCP will keep trying to contact it and have all manner of failure trying to do so.

I assume you can change the speeds with ET or ScreenLogic.   If you have ScreenLogic I would like you to capture the changing the speeds from there.   If you do not and have a wireless remote or indoor controller do a capture while changing the speed from there.  If you have none of these things do it from the OCP.  Ultimately, I want to see if the OCP is sending a different message for this.

mhmatttu, 2022-03-23:
Thanks.  I changed the pump to a single speed pump within dash panel settings.  So far it seems to have saved without error…will see if it sticks.  

I am traveling at the moment so may be a week or so before I can do the capture while changing speeds from the OCP.  Unfortunately do not have screenlogic, remote or indoor controller so will have to be at the OCP.   Will submit as soon as I return.  Thanks again for or all the help!

rstrouse, 2022-03-23:
Unfortunately EasyTouch doesn’t really know how to deal with that when the single speed pump is not the filter pump.   In this case you need to delete it.  The relay that controls that pump is one of the aux relays you define.  I assume this is for your spa jets or a feature pump.

mhmatttu, 2022-05-14:
i've believe this to be a name resolution issue with the router.  will go ahead and close this for now.  thanks.

closed 2022-05-14 by mhmatttu