tagyoureit/nodejs-poolController #398: heater set points are behaving erratically/not working

lockeandev, 2022-03-20:
**Describe the bug**
I am unable to consistently change the heat or set point on dashPanel or via MQTT (via Home Assitant) This was all working previously just fine. I pulled updates today for the first time in 2 months and now it is behaving eratically.  I guess the other thing that changed recently is I updated the intellicenter firmware to the latest version (VERSION 1.064)

**To Reproduce**
Steps to reproduce the behavior:
Try to change the heat or cool set point in dash panel. Some times the heat set point takes, some time it doesn't. Some times when I try to set the cool set point, the set value briefly shows up for the heat set point then reverts to the previous value, and regardless the cool set point remains the same. This brief flash of the coll setpoint shown as the heat set point is also mirrored in the MQTT topics (and being picked up by Home assistant)

**Expected behavior**
should be able to adjust set points from dashPanel or MQTT as I could previously

**Screenshots**
[image]
[image]

[image]

Here's the njsPC log while I was clicking set point buttons on dash panel. The coolsetpoint and heatsetpoitn values are reflected correctly in the log, but not on dash panel and not on MQTT topics. 

On MQTT it looks like the bodies/1//pool/setPoint topic is the only thing being updated regardless of if I change the heat or cool set point in dash panel

**Packet Capture**
attached

**Pool Equipment**
 - Controller: intellicenter 
 - Controller Model: I8PS
 - Controller Firmware Version: VERSION 1.064
 - Pump(s) manufacturer and model: IntelliFlow 2 VSF
 - Chlorinator: IntelliChlor-40
 - Heater(s): ultratemp
 - Chemical controller: Relay Equipment Manager (REM)
 - Valves: Intellivalve

**Desktop (please complete the following information):**
 - OS: windows
 - Browser chrome
 - Version 99.0.4844.74 
 
[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/8310655/replay.zip)

rstrouse, 2022-03-20:
I have looked at your logs and it appears the response from IntelliCenter is slow so it piles up some of the requests when you click quickly.   So I need to make sure this isn't something on the os side so I need to know a few things about what platform njsPC is running on.

Hardware:  Is njsPC running on a Raspberry Pi if so which model
Node Version:  Run `node -v` and report the results
OS Version:  This can be obtained on the pi by running `cat /etc/os-release`
RS485 Make and Model:  What adapter are you running for RS485 comms

lockeandev, 2022-03-20:
Hardware: raspberry pi 4 (4GB RAM)

Node Version: 14.18.2

OS Version:

PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
               NAME="Raspbian GNU/Linux"
                              VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye

RS485 Make and Model:

Industrial USB to RS485 Converter Adapter Original FT232RL Fast
Communication Embedded Protection Circuits Resettable Fuse ESD Protection
TVS Diode etc Automatic Transceiving
https://www.amazon.com/dp/B081MB6PN2/ref=cm_sw_r_apan_glt_i_BH2AQA6TA1X9BF82SGWF?_encoding=UTF8&psc=1

rstrouse, 2022-03-20:
Hmmm... IntelliCenter is taking longer to respond to the request so some of the requests get backed up.  I may need to rethink this to combine the changes.  I see some of them taking over 2 seconds to respond.  When it is responding quickly the response is about 73ms.
[image]

lockeandev, 2022-03-21:
Is there anything I need to try on my end?

rstrouse, 2022-03-21:
No after the 1.064 update IC does not respond to the setpoint message as quickly and gets a bit backed up.  To be fair their controllers have a delay after pressing an increase or decrease to keep from sending the message right away.  If you are changing the value for multiple degrees you can hold it down and it will not send it off until you release it.  Or you can click inside and type it in.

By rapidly pressing the up or down button, the OCP is getting more requests than it can handle and simply starts blowing them off.  I may need to slow it down so that you cannot change the setpoints so rapidly.  It will be a matter of queuing up the requests so that if it gets another before finishing the last it will delay it.

lockeandev, 2022-03-21:
k, typing it in seems to work.

when I type in anything on the cool set point it shows in dashpanel as
being applied to the heat set point, however, the cool set point is
correctly updated on the OCP when I update the cool set point on the OCP,
it updates on dashpanel.

rstrouse, 2022-03-21:
Thanks, I'll fix that as well.

rstrouse, 2022-03-22:
Ok so pull dashPanel and njsPC.  I added a queue so that it delays sending setpoint changes so rapidly.

lockeandev, 2022-03-22:
Awesome. That appears to have worked. I did get one error where dashpanel
said it aborted after 3 attempts while clicking trying to increment the
heat point quickly. but it only happened once and I couldn't reproduce it.

rstrouse, 2022-03-22:
Yeah there is a chance that if IntelliCenter is really busy it just doesn't do anything with the messages.  This version of the firmware seems to be more susceptible to rapid changes on the setpoints than the last.  

I assume the cool setpoint worked as well.  Is that correct?

lockeandev, 2022-03-22:
Correct. The Cool setpoint seems to be working correctly now as well.

rstrouse, 2022-03-22:
Perfect closing this.  Create another issue if you see anything else.

closed 2022-03-22 by rstrouse