tagyoureit/nodejs-poolController #399: RPI and njsPC time frequently out of sync

mhmatttu, 2022-03-21:
Still have an issue with the time as well.  I captured a screenshot that shows the time on njsPC is 12:01am yet the time on the rpi is 10:05am cst.  (slight variance in time of rpi clock and terminal date command due to screen capture delay) Included the njsPC settings in the screenshot.

[image]

_Originally posted by @mhmatttu in https://github.com/tagyoureit/nodejs-poolController/issues/397#issuecomment-1074018705_

tagyoureit, 2022-03-22:
It seems that you have some write issues with updating config.json in the root directory.  I should be seeing verbose, debug, and silly level log messages in the consoleLog but I'm only seeing the info level.  Possible causes:
1. You are using docker and the permissions aren't right
2. You installed the app with sudo permissions but running the app with user level permissions

Can you verify that the user running njsPC has permissions to write to this file?

mhmatttu, 2022-03-22:
Here are the user permissions for the njspc directory 

[image]

tagyoureit, 2022-03-22:
Can you do a quick test?  At the end of your config.json should be a setting for app.level.  Right now it's info...
[image]

Can you change it in the UI to silly and see if it persists? 
[image]

You should also start to see some debug/warn/silly messages in your logs.

mhmatttu, 2022-03-22:
Yes it seems to persist. 

Before changing to silly in UI

[image]

change to silly in UI

[image]

after changing to silly

[image]

After your previous question and before these screenshots, I did change the permissions to config.json using chmod 777

mhmatttu, 2022-03-22:
When I change the setting to manual, it does not present an option to set the time. Am I missing something here?

[image]

tagyoureit, 2022-03-22:
If you do a refresh of the browser, does the time show correctly?  Any error messages in the browser console?

mhmatttu, 2022-03-23:
Browser refresh does not correct the time.  When I ran the browser console, i did see an error message: “Failed to load resource: the server responded with a status of 404 (not found).  Filtering to controller had no results.

[image]

I rebooted the server, ran the browser console, filtered to controller and it does show the correct time and no error messages.

[image]

mhmatttu, 2022-03-23:
The rpi njsPC is connected to my home network using an AP as a wireless bridge.  I’m wondering if I’m having intermittent connectivity issues or wireless interference as there have been times when the pool controller has not been accessible.  However, the AP interface has no errors in its logs and indicates uptime for the full duration.  If the rpi were to go offline intermittently, would that cause these issues?

mhmatttu, 2022-03-23:
I took a closer look at the AP interface logs.  While it shows to have continuous uptime, the logs show 3 disconnect events today.  However it reconnects immediately suggesting no real downtime. 

The rpi itself remains in sync with the correct time regardless of any network downtime so I can’t imagine this being the issue but sharing all that I can to troubleshoot.  Thanks.

tagyoureit, 2022-03-23:
I am unable to recreate this.  It looks like the correct information is getting all the way to the browser but the browser isn't rendering it properly.  Can you try a different browser or from a different machine?  Grasping at straws here but maybe that's something to try.

And two other things... are the outputs of these the same and correct?
1. http://raspberrypi.local:4200/state/time
2. http://192.168.1.50:4200/state/time
(Assuming you have njsPC and dashPanel both on this same box.)

tagyoureit, 2022-03-23:
When the time is incorrect, does it still change?  Or is it stuck on whatever time is incorrect?

mhmatttu, 2022-03-23:
I had never noticed this before but the time is stuck on the incorrect time and does not change. 

I can ping raspberry.local and get results but http://raspberrypi:4200/state/time results in “server cannot be found”

rstrouse, 2022-03-23:
Now try that same thing using the ip address of the pi.  I am guessing that the DNS for your local wifi is inconsistent with machine name resolution.  I have seen this before with Xfinity mesh setups.

The other thing that can cause havoc are some devices that are based on router os.  These are often AP devices that can be configured as a wireless bridge/access point (sound familiar).  If there is a DHCP server on these it should be disabled or it will dole out IP addresses that may conflict.  They can even respond to gateway requests.  So if you have an access point make sure it is configured to get a reserved IP address from the main router and it does not have an enabled DHCP server.

So you can change the connection in the hamburger menu to the ip address (only if you created an ip reservation in your router).  Given my experience some of the internet company supplied routers are just pisspoor and have very little configuration capabilities so the steps will vary based upon your router to get this done.

mhmatttu, 2022-03-25:
I'm using the residential gateway (RG) provided by AT&T.  Configuration options are definitely limited.  The AP bridge is a Ubiquity Picostation M2 in bridge/station mode with DHCP disabled.  I did notice that the RG/router is listing the RPI by the WLAN0 MAC address of the Picostation vs the MAC address of the RPI.  I can see how this would effect machine name resolution.  

I had the RPI configured with a static IP that was outside of the DHCP range.  I have now changed it to DHCP.  The router assigned a new IP as expected and I configured IP Allocation (reservation).  I reconfigured the dashPanel connection using "find servers" which associated by IP address.  Right now time is correct.  

If this issue continues, I may change the setup by running cat5 to the RPI for a hardwired connection (big hassle) or use a USB wifi dongle on the RPI (concerned about signal strength).

rstrouse, 2022-03-25:
The other thing that you can do is that under the hamburger menu you can turn on the proxy for njsPC.  This will route all traffic through dashPanel so there will be no CORS issues from a browser standpoint since the port comms will all be through the port for dashPanel.

mhmatttu, 2022-04-01:
Good evening gentlemen.  I'm just getting back to this after some travel.  I've had proxy turned on from the beginning.  As a temporary solution I have setup crontab to reboot the RPI every hour as that seems to be the only way to keep the time in sync.  Unfortunately it takes about 5 minutes for everything to start back up so not the best solution.  

As previously mentioned, machine name resolution is an issue with my router (AT&T residential gateway) so http://raspberrypi.local:4200/state/time nor http://raspberrypi:4200/state/time resolve.  

http://192.168.1.50:4200/state/time does show the correct time.  

[image]

Per the screenshot below, the router is resolving a machine name using the mac address of the connected AP in bridge mode.  

[image]

While I don't seem to be having network connectivity issues, if this machine name resolution issue is thought to be the problem, I will proceed with running an ethernet cable to the RPI.  

Thank you!

rstrouse, 2022-04-02:
Very strange that would be the resolved name.

tagyoureit, 2022-04-02:
https://forums.att.com/conversations/att-fiber-equipment/attlocalnet/61e064429f219f31b8f8f404

Seems to be a known issue with these ATT routers.  There are a couple solutions in that thread but basically it sucks at routing anything on your local network.

mhmatttu, 2022-05-14:
i've believe this to be a name resolution issue with the router.  will go ahead and close this for now.  thanks.

closed 2022-05-14 by mhmatttu