tagyoureit/nodejs-poolController #401: Unable to communicate with Hayward Aquarite T15 Chlorinator

Hect0rG, 2022-03-29:
**Describe the bug**
I am not seeing any communications being initiated when trying to connect the Hayward Aquarite T15 Chlorinator. 

**To Reproduce**
Steps to reproduce the behavior:
- Turn off all power.
- Connect RS485 port to Industrial USB to RS485 Converter Adapter. My cables are connected as follows the yellow 485B port on the chlorinator going to the B- port on the adapter. My ground cable is connected to the USB RS485 labeled Gnd and that goes to the GND on the chlorinator and the GND on the power adapter.
- A power adapter was installed to power the RS485 port on the chlorinator as indicated [tagyoureit/nodejs-poolController#293](https://github.com/tagyoureit/nodejs-poolController/issues/293)
- On the Nixie Controller I have added the Chlorinator and updated Name, Body, and Model as follows
[image]

**Expected behavior**
Expect to be able to control and get data from chlorinator. The bubble to the left of the chlorinator should be on.
[image]

**Packet Capture**
 attached zip/log files [replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/8367361/replay.zip)

**Pool Equipment**
 - Controller: N/A
 - Controller Model: N/A
 - Controller Firmware Version: N/A
 - Pump(s) manufacturer and model: Pentair Intelliflow VS SVRS
 - Chlorinator: Hayward AQR15 AquaRite Salt Chlorination System
 - Heater(s): N/A
 - Chemical controller: N/A
 - Valves: N/A
 - Any other relevant equipment: N/A

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04.4 LTS 
 - Browser: Chrome
 - Version Version 100.0.4896.46 (Official Build) beta (64-bit)

**Smartphone (please complete the following information):**
 - Device: Samsung Note 20 Ultra
 - OS: Android OS 12
 - Browser: Chrome
 - Version: 99.0.4844.88

**Additional context**
- The pump (Pentair VS) continues working perfectly.
- Pool is on schedule when troubleshooting. 
- Nothing shows when I run the "od -x < /dev/ttyUSB0" command after I disconnect the pump. When the pump is on I see the constant output and can control it.
- I  have the chlorinator RS485 powered with a 12 V DC power supply as instructed [here]([tagyoureit/nodejs-poolController#293](https://github.com/tagyoureit/nodejs-poolController/issues/293)) 
- Jumper on the chlorinator is left on the master setting ( not sure if this makes a difference since its the only chlorinator)
- Running pool controller on a Raspberry Pi 3B+ that is hosting the Industrial USB adapter.

rstrouse, 2022-03-29:
Ok pull njsPC and dashPanel.  Then delete the chlorinator from the configuration and re-add it.  This time choose Nixie as the controller when adding it back in.  The chlorinator is set up as an external device and should be set up as a Nixie device.
[image]

rstrouse, 2022-03-29:
The commands to pull njsPC and dashPanel are as follows from a command prompt on the pi.
`cd ~/nodejs-poolController`
`git pull`
`npm i`
`cd ~/nodejs-poolController-dashPanel`
`git pull`
`npm i`

Hect0rG, 2022-03-29:
That did the trick! Thanks!

rstrouse, 2022-03-29:
Just for my information does it need the 12v tap?

Hect0rG, 2022-03-29:
Confirmed that it is required. The connection gets disabled immediately when I unplug the adapter and comes back online within a few seconds after I unplug it. Tomorrow I am going to see if I can reduce the voltage to 10 V (to see if I can use the voltage indicated on the panel) using a lab power supply I have. I should also be able to determine the amperage.  I imagine they have a voltage regulator on the board so that it won't fry but one can never be too sure so I'd rather use a 10V DC power supply or reduce the voltage with some resistors because 10V power supplies are harder to find. 

Is there any way to retrieve the temp from the chlorinator? It shows on the display but not sure if it's accessible via RS485 connection. They also have Cell Voltage, Cell Amperage, Board version that might be useful.

rstrouse, 2022-03-29:
The thermistor on the chlorinator is not published.  However, I would suggest you implement one of these.   If you are using a BAS card you can hook it directly to one of the 10k temp probe ports or you can simply implement as described in the wiki.

Easy peasy lemon squeezy.
https://github.com/rstrouse/relayEquipmentManager/wiki/Temperature-Sensors

https://www.amazon.com/AR-PRO-Temperature-20-Feet-Replacement-Automation/dp/B07Q8RHK7D/ref=sr_1_1_sspa?keywords=pool+temperature+sensor&qid=1648523543&sprefix=pool+temp%2Caps%2C182&sr=8-1-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUEyUEJLRlc0TFlFS0lWJmVuY3J5cHRlZElkPUEwODUwMTIxMkZNRExDMUM0T0RGVCZlbmNyeXB0ZWRBZElkPUEwMDg5MTA4UlVUQjE0RTVNQks2JndpZGdldE5hbWU9c3BfYXRmJmFjdGlvbj1jbGlja1JlZGlyZWN0JmRvTm90TG9nQ2xpY2s9dHJ1ZQ==

Hect0rG, 2022-03-29:
Very cool stuff! I will take a look into that. glad I stumbled onto this. Thanks again for all the help!

rstrouse, 2022-03-29:
You should buy the 2 pack of 10k sensors and hook one up as water temp and the other as air temp.

Hect0rG, 2022-03-29:
Tested today and it seems to work anywhere between 7 V and 12 V and is pulling .001 amps so any small DC adapter will work. Sticking with the 10V should be the best option as that is what the label on the chlorinator states.

rstrouse, 2022-03-30:
Thanks for that.  You could toss an adjustable buck converter in there to get regulated 10volts.

Hect0rG, 2022-03-30:
Great idea, that makes it simple. Just ordered one. Thanks again.

rstrouse, 2022-04-05:
I am going to close this for now and I did fix your superChlorinate function.

closed 2022-04-05 by rstrouse