tagyoureit/nodejs-poolController #410: Question about my IT i7+3 Pool Config not sticking - Valve issue [BUG]

sdpools, 2022-04-10:
### nodejs-poolController Version/commit

7.6.1

### nodejs-poolController-dashPanel Version/commit

7.6.1

### Node Version

v14.19.1

### Platform

Linux raspberrypi 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l GNU/Linux

### RS485 Adapter

Waveshare (amazon) ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC

### Are you using Docker?

- [ ] Yes.

### OCP

None - Not owned

### Pump(s)

Hayward VSP3400

### Chlorinator(s)

IC-40

### What steps will reproduce the bug?

When adding a 2nd heater - Solar 

### What happens?

The gui says "saving" if I reload browser, it did not save.

### What should have happened?

heater added

### Additional information

I am also having an issue with valves. intakereturnvalve is set to false. no return valve is present in valve list.

sdpools, 2022-04-10:
I noticed the id is -1?
PUT /config/heater {"id":-1,"name":"Heater 1","type":2,"body":32,"startTempDelta":6,"stopTempDelta":3,"coolingEnabled":false}

sdpools, 2022-04-10:
PUT /config/heater {"id":-1,"name":"Heater 2","type":2,"body":32,"startTempDelta":6,"stopTempDelta":3,"coolingEnabled":false}

rstrouse, 2022-04-10:
The -1 is the correct id.  This tells njsPC that the id has not yet been assigned.  The packet not processed messages are because IntelliTouch is probing for additional power centers.  17,18, and 19 are the reserved addresses for these.  I don't see the outgoing message in the queue so maybe there is an issue with that.  In the meantime can you make a replay with changing a custom name.

rstrouse, 2022-04-13:
I am doing a little more resting but I may have a fix in the next couple of days.  It has to do with some weirdness with the IntelliTouch messages for solar enable.

rstrouse, 2022-04-22:
Pull njsPC and let me know if the Solar Enable sticks.  I don't have an IntelliTouch i7+3 so I need feedback from you on this one.

rstrouse, 2022-05-03:
I am closing this for now.  If the fix did not work please open another issue.

closed 2022-05-03 by rstrouse

sdpools, 2022-05-03:
Rstrouse,

Understood. Thank you for looking into it.