tagyoureit/nodejs-poolController #427: [BUG]

johnny2678, 2022-04-26:
### nodejs-poolController Version/commit

b2d388d

### nodejs-poolController-dashPanel Version/commit

dce00b

### relayEquipmentManager Version/commit

latest

### Node Version

v12.22.9

### Platform

Linux PiPoolv2 5.10.63-v7l+ #1496 SMP Wed Dec 1 15:58:56 GMT 2021 armv7l GNU/Linux

### RS485 Adapter

MEGA-BAS (but not currently wired to OCP atm)

### Are you using Docker?

- [ ] Yes.

### OCP

Easytouch 2 4

### Pump(s)

Intelliflo VS

### Chlorinator(s)

IC40

### What steps will reproduce the bug?

create backup on device #1 
upload backup to device #2 
initiate backup on device #2 

### What happens?

first time doing a backup/restore on a different device

first, dash froze on the restore screen
[image]

After a browser refresh, and returning to the restore screen, the backup config was listed there and I could click to to try and restore, but then I get a stacktrace error.

[image]

njspc log errors:
```
3|njsPC  | [4/26/2022, 20:48:46] error: Error extracting restore options from : ENOENT: no such file or directory, open
3|njsPC  | [4/26/2022, 20:48:46] error: Error validating restore file: Cannot read property 'poolConfig' of undefined
3|njsPC  | [4/26/2022, 20:48:46] error: Error validating restore options: Cannot read property 'poolConfig' of undefined
3|njsPC  | [4/26/2022, 20:48:46] error: Cannot read property 'poolConfig' of undefined
```

Not important - I'm still slowing getting my new enclosure mounted and wired but I have my existing (working) setup still running.

### What should have happened?

successful restore

### Additional information

replay attached....

[replay (29).zip](https://github.com/tagyoureit/nodejs-poolController/files/8566449/replay.29.zip)

johnny2678, 2022-04-26:
Just occurred to me that I should upload the backup zip
[backup (1).zip](https://github.com/tagyoureit/nodejs-poolController/files/8566666/backup.1.zip)

rstrouse, 2022-04-30:
I am having a bit of trouble replicating this.  Interestingly the console shows no path to the zip file.
`[4/26/2022, 20:51:38] info: [20:51:38] 192.168.5.146 PUT /app/restore/validate {"filename":"","options":{"version":"","name":"","backupDate":"","njsPC":true},"filePath":""}` which explains the error and why there are no options displayed for it.  Possibly because there is no name given to the backup.
```
{
  "name": "",
  "njsPC": true,
  "servers": [
    {
      "name": "Relay Equipment Manager",
      "uuid": "02d3b662-f9a6-4f1e-81cf-38352343f70c",
      "host": "",
      "backup": false,
      "errors": []
    }
  ],
  "automatic": false,
  "version": 1.1,
  "errors": [],
  "backupDate": "2022-04-26T15:33:42.683-0400"
}
```

johnny2678, 2022-05-02:
This might have to do with special chars in the filename?  It was Backup (1).zip.  When I renamed to Backupjh.zip it I got a different message:
[image]

I don't see any way to switch from nixie shared body to Easytouch so I'm guessing this is because I haven't activated my RS485 bus on the Sequent BAS yet?

rstrouse, 2022-05-03:
Yes that is why it will not allow that restore.  Part of the restore process is to sync the panel.  The best part of this is that you can now restore an EasyTouch controller but it has to be a backup from an EasyTouch.  I noticed our misspelling though.