tagyoureit/nodejs-poolController #442: [BUG] MasterTemp gas heater not working

nkm8, 2022-05-13:
### nodejs-poolController Version/commit

5264482

### nodejs-poolController-dashPanel Version/commit

0985f1a

### Node Version

14.19.2

### Platform

Linux pool-heater-pi 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l GNU/Linux

### RS485 Adapter

Industrial RS 485

### Are you using Docker?

- [ ] Yes.

### OCP

Nixie

### What steps will reproduce the bug?

- Heater set to heat mode @ 80F
- Turned off heat mode
- Turned heat mode back on @ 80F

### What happens?

- Heater always shows 'HEATER OFF' and correctly shows RS485 lockout message
- Can't view current set points to determine if changes are being applied

### What should have happened?

- Heater turns on @ 80F setpoint
- Heater displays the current water temperature (71F)

Capture:
[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/8684304/replay.zip)
[image]
[image]

 

### Additional information

Heater: Pentair MasterTemp 125 Model #: 461059

nkm8, 2022-05-13:
Forgot to mention, after unplugging the pi4 and the heater, I disconnected the RS-485 wires from the heater and turned the heater back on. After plugging the heater back in, it started heating @ 80F, which was the state that I set, but never showed, while RS-485 was connected.

This state was also different than the state of the heater before connecting the RS-485 wires - it was originally set to 68F.

rstrouse, 2022-05-13:
It doesn't look like you have a temp sensor hooked up.   You must have a temp sensor for this to work.  Unfortunately, that temp sensor in the control box for MasterTemp is not reported over RS485.  It is very easy and inexpensive to install a temp sensor though.

nkm8, 2022-05-17:
@rstrouse I was originally going to ask if I could tap the existing thermistor in the heater, but it looks like that won't work: https://www.troublefreepool.com/threads/pentair-intellichlor-temperature-sensor-hack.189425/

I am leery of drilling the PVC and using a band clamp - would something like this work? https://www.lowes.com/pd/LASCO-1-1-2-in-x-1-1-2-in-x-1-1-2-in-x-1-2-in-dia-Tee-PVC-Fitting/3573186

That should allow me to use any 1/2" threaded ds18b20 thermowell:
https://www.amazon.com/dp/B07G3ZWPLG/
https://www.amazon.com/thermowell-Stainless-Plastic-fermenter-Homebrew/dp/B07DXFWV6B/
https://www.amazon.com/Temperature-thermowell-Stainless-fermenter-Homebrew/dp/B07F32MT97/

I'm also curious - is the water temperature the only sensor required for the heater to be controlled correctly via Nixie? I thought I read something about an ambient air sensor being required - but I should be able to pull that from a weather API instead of a local sensor, right?

I appreciate all of the help. I have to figure out how to source the rest of the materials - finding a pi4 was tricky, it looks like the hat for the temperature sensor will be equally tricky to find in stock. I also need to find a good sized weatherproof enclosure to mount to a 4x4 post.

rstrouse, 2022-05-17:
Yes the water temp sensor is all that is required.  However do not overthink the temp senor.  This is what you want.  https://www.amazon.com/AR-PRO-Temperature-20-Feet-Replacement-Automation/dp/B07NWK6GJ4/ref=sr_1_2_sspa?crid=TLRX58VL34P1&keywords=pool+temp+sensor&qid=1652759312&sprefix=Pool+temp+sensor%2Caps%2C238&sr=8-2-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUFITkVBSFRKRDREMEwmZW5jcnlwdGVkSWQ9QTA5NTAwNjAxM05XTjNQTkIwSDFQJmVuY3J5cHRlZEFkSWQ9QTA0ODIzMDYxQTE4QkdFWDVYM1hSJndpZGdldE5hbWU9c3BfYXRmJmFjdGlvbj1jbGlja1JlZGlyZWN0JmRvTm90TG9nQ2xpY2s9dHJ1ZQ==

The band clamp is what all the pool systems use and it will outlast the pool.  The install takes about a minute and there are no metallic components in the water.  Your heater thermistor will be dead before the water temp sensor.

nkm8, 2022-06-23:
I'm not sure if this is still the right place to ask, but can you please confirm the wiring diagram for the temperature sensors you suggested? I finally received my automation hat: https://shop.pimoroni.com/products/automation-hat?variant=30712316554

I am following the wiring diagram here: https://github.com/rstrouse/relayEquipmentManager/wiki/Temperature-Sensors#temperature-sensor-support

Looking at the automation hat, I think I need to solder to pull 3v3 from the center connector - top left - pin 1. This will feed 2x 10k ohm resistors (1 for water temp, 1 for ambient temp)
Then, I can pull ground and ADC1/2 from screw terminals

Do I have all that right?

tagyoureit, 2022-06-23:
You should never solder anything directly to the Pi.  Because this is a "hat" it is designed to plug directly into the 40 pin header on the Pi.

Based on the [pinout page](https://pinout.xyz/pinout/automation_hat) this board supports i2c communications.  It likely won't work out of the box (with REM) and we may need to explore the i2c commands.

Which 10k temp sensor did you end up going with?  Does it support 3v3 and 5v?  The hat has the 5v screw terminals but you'll need to solder a header for the 3v3.

nkm8, 2022-06-23:
I bought the sensors you suggested: https://www.amazon.com/AR-PRO-Temperature-20-Feet-Replacement-Automation/dp/B07NWK6GJ4/ref=sr_1_2_sspa

The hat was difficult to find, but it is listed in the wiki as a suggested hat, so I am surprised to hear that it won't work out of the box.

I am not sure what the temp sensors support - they came in a clear plastic bag with no other markings. Is it ok to try 5V or is there a risk of damaging the sensors?

rstrouse, 2022-06-23:
The adc is tolerant to 24v on that card.  You can use either 3v3 or 5v but you must set the gain appropriately.  The ADC is a ADS1015 on that hat but they ran out of terminal space for ADC4 so use 1-3.  To use ADC4 you must use 3v3 on that card by soldering a header to the 3v3 section and running a 10k resistor to ADC4 then connect the sensor to the ADC4 and GND terminals.

To use 5v you will need a 10k resistor between the 5v terminal or 3v3 pin and the adc channel with 1 wire from the temp sensor in the same terminal.  Then take the other wire from the sensor and connect it to ground.

tagyoureit, 2022-06-23:
Double check the ADC when you get the board, it might be the ADS1115.

> Due to exciting global electronic parts shortages, some Automation HATs manufactured in 2021 have an [ADS1115](https://www.ti.com/lit/ds/symlink/ads1115.pdf) ADC instead of the usual [ADS1015](https://www.ti.com/lit/ds/symlink/ads1015.pdf). Our Python libraries have been updated to automatically detect and adjust for this chip change.

nkm8, 2022-06-25:
I verified that the hat I received is indeed using the ADS1115. For clarity on the hat selection, I am referencing [this section](https://github.com/rstrouse/relayEquipmentManager/wiki/Temperature-Sensors#suggested-parts-list) of the wiki - under `Some Suggested Hats`

I followed [these instructions](https://github.com/rstrouse/relayEquipmentManager/wiki/Temperature-Sensors#software) and made sure to configure the ADS1115 with a power gain of 4.096v and vcc ref of 5.0v. I am using the 5V rail from the hat, but it looks like the temperature value is always 0F:
[image]

I tried reversing the wires from the temperature sensor (white, red), but the input/resistance values were the same.

[image]
[image]

I appreciate all of the help. Both temperature sensors are next to each other - ambient temp is ~75F.

rstrouse, 2022-06-25:
It looks like you have the feeds properly configured.  First if you can take a picture of how you have connected the +5v from the hat to the ADC channel or describe how you have it connected.  There should be a 10k resistor in series from the +5v to ADC1.  In that same ADC1 one wire from the probe (it doesn't matter which one) should then be wired into the same screw terminal.   The other wire for the probe should be placed into one of the GND terminals.

If you have it wired up the way described above, first pull REM then set the Vcc Ref value on the probe device to 25.85 volts.

nkm8, 2022-06-25:
The wiring is a mess because I have both probes connected and I am testing on the bench, so I am using alligator clip jumper cables for a lot of the connections. I can confirm that everything is wired as you describe - none of the connections are touching.

I have a voltmeter, so I confirmed:
- Resistance on the resistor is 10kohms
- 5V to ground on the hat measures 5V
- terminal side of the resistor to ground is 5V
- opposite end of the resistor to ground is ~2.5V

Is there anything else I can check with the meter?

I pulled from master and applied the change you mentioned:
[image]

Not sure if it helps, but  here are the logs. I only made the vcc ref change to the water sensor, not the air sensor:
```
1|REM  | info: [11:14:48] 192.168.1.122 PUT /config/genericDevices/device {"options":{"name":"Air Temp","units":"F","inputType":"volt","vccRef":4.9999999999999964,"inputBitness":10,"inputResistanceUnits":1,"calibration":0},"values":{"adcValue":0.3246,"inputUnits":"volts","resistance":694.352,"temperature":null,"units":"F"},"id":2}
1|REM  | info: received {"options":{"name":"Air Temp","units":"F","inputType":"volt","vccRef":4.9999999999999964,"inputBitness":10,"inputResistanceUnits":1,"calibration":0},"values":{"adcValue":0.3246,"inputUnits":"volts","resistance":694.352,"temperature":null,"units":"F"},"id":2}
```

rstrouse, 2022-06-25:
Awesome, now can I get you to change that Vcc Ref value to 3.3 volts.  In looking at the code for Pimoroni it appears that they are doing some gymnastics with the vcc to tolerate 24v on a 3.3 volt pin.  I am putting together a change to allow us to see the raw adc values from the hat.  This will give us an idea as to what readings are coming from the ADC on that hat.

nkm8, 2022-06-25:
A bit warmer than reality, but we have a temp reading! 🥵 

[image]

rstrouse, 2022-06-25:
Pull REM again and then change the power gain on the ADS channel to 2.048.  I am doing some math here to back fill what they are doing mathematically to come up with the conversion.  I assume we are still in the mid to high 70s on the temp.

When you do this shoot a screenshot of the ADS device from REM it will now have the raw ADC values on the display.

nkm8, 2022-06-25:
Correct - I'll grab a thermometer to put next to the probes, but we should be at 75-76F

[image]

rstrouse, 2022-06-25:
Wow I expected the raw ADC values to be about double that value.  Set the gain to 4.096 now.  What are the values?

nkm8, 2022-06-25:
Official temp reading is 76F - note I am still only adjusting the settings on the water temp - I haven't bothered updating the air temp settings yet.

[image]

rstrouse, 2022-06-25:
Just to be sure this is how you have this wired up.

nkm8, 2022-06-25:
Confirmed - 2 pictures as a sanity check

[image]
[image]

rstrouse, 2022-06-25:
Ok pull REM again.  I went through their wiring diagram and figured out that in order to make the inputs 24v tolerant they added a bunch of resistors.  I thought they may have been doing something clever with AIN3 but as it turns out they just added enough resistors to make sure the input to the ADC was less than 3.3v.

So after you pull REM there will be additional converter types in the Converter dropdown.  Select Pimoroni ADS1115 and lets see where the voltage lands.  On the 10k sensor device set the VCC voltage to 5.0v.

tagyoureit, 2022-06-25:
*DANG*  @rstrouse beat me again... was just finishing this up.

I found this at [shopify](https://cdn.shopify.com/s/files/1/0174/1800/files/automation_hat_schematic.pdf?199375959460709850).
[image]

It looks like, in order to be 24v tolerant, they are scaling the input voltage from 24v down to the 0-3.3v range.  In order to get a true reading on voltage on the conversion back we need to take that into account.

Vin=Vadc/(120/120+820))
Vin=Vadc/.12766

from the value in the pic above, ADC readings are 2642.
Value =2642
Max=32767
PGA=2.048
`Convert Voltage ${value} / ${max} * ${pga}`
Value / max * pga = Vadc = .165125
Vin = Vadc/.12766 = .165125/.12766 = 1.29v

rstrouse, 2022-06-25:
It had me scratching my head for a while.  Apparently the previous user was using something from pimoroni called a pHat.  I thought they just renamed it like all things these days that could be twisted into something offensive.

I kept looking at that ominous resistor bank thinking they just banged them into a convenient spot.

nkm8, 2022-06-25:
Very interesting - the results are looking much better! Ambient is hovering around ~75.5, so I think the probe reading is within a reasonable margin of error:
[image]
[image]

rstrouse, 2022-06-25:
Yeah that is well within the differences you will find between devices.

nkm8, 2022-06-26:
Thank you both so much for all your help! I greatly appreciate everything - I re-connected the second temperature probe on the bench and confirmed everything is working perfectly. Tomorrow I'll try to get everything connected outside at the pool and finally get the heater working from my phone. I plan on setting up the Home Assistant integration as well.

Do either of you accept donations or at least PayPal for some beer money? I am a Software Engineer of ~10 years and both of your projects are extremely impressive. 🍻

closed 2022-06-26 by nkm8