tagyoureit/nodejs-poolController #443: Cool setpoint missing on Spa

Tegguy, 2022-05-14:
### nodejs-poolController Version/commit

7.61

### nodejs-poolController-dashPanel Version/commit

7.61

### Are you using Docker?

- [ ] Yes.

### What steps will reproduce the bug?

On the dashpanel under the Spa display the "cool setpoint" is missing for the ultratemp heater. I can see the heat setpoint but not the cool.

### What happens?

Missing setpoint display

### What should have happened?

Setpoint should display

tagyoureit, 2022-05-15:
Can you attach a packet capture replay?

Tegguy, 2022-05-15:
The setpoint is there and working and even pops up when enabling ultratemp
but the value doesn't display on the dashpanel

rstrouse, 2022-05-16:
Pull dashPanel the setpoint should appear if you are not on an EasyTouch or IntelliTouch panel.   These for some odd reason don't support the cooling setpoint on the second body.  Changes to the second body cooling setpoint end up on the first body.

Tegguy, 2022-05-16:
Got it thanks!

closed 2022-05-16 by Tegguy