tagyoureit/nodejs-poolController #451: [BUG] Changing the speed of a pump deletes the pump

terafin, 2022-05-19:
### nodejs-poolController Version/commit

9d224de

### nodejs-poolController-dashPanel Version/commit

d932f2f

### Node Version

docker

### Platform

docker

### Are you using Docker?

- [X] Yes.

### OCP

EasyTouch 2 4P

### Pump(s)

Intelliflo VS

### What steps will reproduce the bug?

Click on the pump here:
[image]
 
Change the speed (RPM) of the pump

Wait a few moments, and this shows up:
[image]

### What happens?

• Pump turns off
• If I go to the panel, the circuit is removed from the pump

### What should have happened?

RPM changed :)

### Additional information

[replay (2).zip](https://github.com/tagyoureit/nodejs-poolController/files/8734326/replay.2.zip)

rstrouse, 2022-05-19:
Please pull njsPC and make sure your dashPanel is up to date as well.  I cleaned up the code a little bit but I am surprised at our outgoing message.  It is indicating that the circuit should be deleted.

terafin, 2022-05-19:
Updating them now... thanks!

terafin, 2022-05-19:
Still seemed to fail, will upload new logs in a few minutes, as I need to add it back at the panel

terafin, 2022-05-19:
Adding logs from this happening again

terafin, 2022-05-19:
[replay 3.zip](https://github.com/tagyoureit/nodejs-poolController/files/8734787/replay.3.zip)

terafin, 2022-05-19:
[image]
Here's the full system config

vj950, 2022-05-20:
I can confirm that on my Intellitouch i93+3s system, this issue is fixed.

rstrouse, 2022-05-20:
I have set my system up to emulate yours.  This is very odd your call to change the speed from the browser looks like the following
``` 
   PUT /config/pump 
   {"id":1,
    "circuits":[
           {"speed":1900,"units":{"val":0},"id":1,"circuit":6},
           {"speed":2100,"units":{"val":0},"id":2,"circuit":6}
      ]
   }
```

And the correct call which is what I am seeing is as follows.
```
   PUT /config/pump 
   {"id":1,
    "circuits":[
         {"speed":2150,"units":{"val":0},"id":1,"circuit":6}
     ]
}
```

Your request is insisting on setting two different values for the same pool circuit (6).  That will cause a definite choke.  The only way this can be is if your dashPanel is not up to date.  Please hard refresh your browser.  Perhaps you have some cached script or the docker container did not recompile the script before running it.  I have never used docker so I don't know what it does with cached files.

When you bring up the screen to change the speeds I assume it looks like this.
[image]

terafin, 2022-05-20:
[image]

I can confirm that this is my same setup. 

Just ran this in private browsing mode, and had the same results after changing it...

[image]
 
Will try a clean docker container (with no mounts).

Note I'm using msmi's docker setup. :) 

Trying a few permutations now...

terafin, 2022-05-20:
@rstrouse I can also provide remote access for this FYI, if this is useful debugging to take out the middle man :)

terafin, 2022-05-20:
Notes:
```
Log
7:40 PM kicked off fresh docker containers with no storage
7:42 PM connected fresh dash panel to controller
7:43 PM started packet capture (with config reload)
7:43 PM config reload complete
7:44 PM toggled pump off then on from web ui
** aside note heater status reported “Heatpump” but “Heat mode” is off
7:46 PM (after priming settled) used UI to change the RPM from 2100 to 2200
7:46 PM pump shut off
7:47 PM using web UI, re-added pump on “Pool” circuit, and saved it
7:47 PM toggled pump off and on from web ui
7:48-49 PM went to easy touch panel, and selected pump, re-attached to circuit “Pool”, set RPM to 2100
Worked

```
Logs:
[replay (3).zip](https://github.com/tagyoureit/nodejs-poolController/files/8735413/replay.3.zip)

Docker setup:
```
  poolcontroller:
    logging: *default-logging
    container_name: poolcontroller
    restart: always
    user: node:dialout
    devices:
      - /dev/ttyUSB0
    environment:
      TZ: America/Los_Angeles
      PUID: 1000
      PGID: 1000
    ports:
      - "4200:4200"
    image: msmi/nodejs-poolcontroller:latest
  poolcontroller-dashpanel:
    restart: always
    logging: *default-logging
    container_name: poolcontroller-dashpanel
    ports:
      - "5150:5150"
    image: msmi/nodejs-poolcontroller-dashpanel
    environment:
      TZ: America/Los_Angeles
      PUID: 1000
      PGID: 1000
    depends_on:
      - poolcontroller
````

System con
[image]
fig:

rstrouse, 2022-05-20:
pull njsPC I added some pre-checks and logging emits for when there is an invalid attempt to set incorrect circuit data.

terafin, 2022-05-20:
@rstrouse Totally works now :) Thank you so much!