tagyoureit/nodejs-poolController #452: [BUG] UltraTemp ETI Hybrid status shows "heatpump" even when off

terafin, 2022-05-20:
### nodejs-poolController Version/commit

d915718

### nodejs-poolController-dashPanel Version/commit

d932f2f

### Are you using Docker?

- [X] Yes.

### What steps will reproduce the bug?

Seems to be pretty persistent, my heater status always shows "heatpump" in the UI, and the flame UI is showing

[image]

In the attached trace I toggled modes at the end between the various modes, and it seems to work, but "off" always shows "heatpump"
[image]

### What happens?

[image]

### What should have happened?

Status should show off

### Additional information

Attached replay!

Let me know if this is a dash-panel bug, sorry if I've filed it in the wrong spot

[replay_status_plus_mode_changes.zip](https://github.com/tagyoureit/nodejs-poolController/files/8744785/replay_status_plus_mode_changes.zip)
!

rstrouse, 2022-05-21:
I'll have a look.  We have seen this heater on the ET8 and IntelliCenter.  When it is off it should be reporting off if it isn't in the cooldown mode.   ET does a terrible job of reporting heater cooldown (meaning it doesn't).

rstrouse, 2022-05-21:
Pull njsPC.  It appears EasyTouch reports that the heat status is heatpump even though the mode is off.   However, I did notice that the information we had on the heater status directly from the ETi is not complete.  It would be great if we had a capture that showed us definitively what the data looks like when the heater is actually performing specific functions.

When it comes to heaters there are two specific pieces of information.  The first is what mode it should be running in.  In the case of the ETi these are
```
Off - No heating
Heat Pump - Use only the heat pump portion of the heater
Gas Heat - Use only the gas portion of the heater
Hybrid - Use the heat pump to maintain the heat but use the gas heater to get there
Dual Heat - Use both the heater and the heat pump together to reach the sepoint
```
Then there is the other side of this which is the status
```
Off - The heater is not doing anything at the moment
Heating w/heat pump - The heat pump is currently being used to heat
Heating w/gas - Now its heating with gas
Heating w/both - Both the heat pump and the gas heater are in use
Cooldown - The heater is in an idle state while it cools down its components
```

What makes this difficult is that the mode does not reflect the status.  For instance, you can have it set to heat pump but it could currently be off since the setpoint was reached.   Or it could be in hybrid mode and it could currently be using the gas or heat pump at the moment.

In order to get a complete understanding of the data we would need to know exactly what is going on with the heater at the time the messages come through.

terafin, 2022-05-21:
Pulled and works a charm. Do you want me to run through states and report what the actual status is with a replay? Happy to!

rstrouse, 2022-05-21:
By all means yes.  Unfortunately, we have none of this equipment, just a user base with all sorts of it.

tagyoureit, 2022-06-22:
@terafin Are you able to provide a replay to validate the states?  Thx!

terafin, 2022-07-02:
Yes sorry! Been out of town, I'll be back tomorrow. Will get to this likely Sunday!

rstrouse, 2022-11-18:
I managed to cobble all of this together from a couple of different replays.  I think all of this is good.

closed 2022-11-18 by rstrouse