tagyoureit/nodejs-poolController #457: Trouble configuring my SuperFlo VST

tagyoureit, 2022-05-23:
I'm also having trouble configuring my SuperFlo VST.  It's the new version s manufactured after 10/15/20, so I'm configuring as Intelliflo VS.

I'm attempting to use the RS-485 port on the SM megabas card.  I have only yellow/green wired to the pump on the data cable.  I made my own data cable because I wasn't able to source the digital control kit.  I believe I have the wires on the correct pins (2,3).  The pump flashes "Auto" on the display, but in the njsPC logs I get:
3|njsPC  | Processing schedule 1 - Pool : ShouldBeOn: true ManualPriorityActive: false Running: true Suspended: false Resumed: false
3|njsPC  | [5/22/2022, 8:39:45 PM] warn: Message aborted after 3 attempt(s): 165,0,96,33,5,1,6,1,50
3|njsPC  | [5/22/2022, 8:39:45 PM] error: Error sending setPumpManual for Pump: Message aborted after 3 attempt(s): 165,0,96,33,5,1,6,1,50
3|njsPC  | Processing schedule 1 - Pool : ShouldBeOn: true ManualPriorityActive: false Running: true Suspended: false Resumed: false

Any troubleshooting steps would be greatly appreciated!!!

_Originally posted by @ddyer8 in https://github.com/tagyoureit/nodejs-poolController/discussions/453#discussioncomment-2800982_

tagyoureit, 2022-05-23:
The VST seems to be a little bit of a different animal so I moved this to its own issue.

This pump looks like it can only be controlled by the 4 set speeds, similar to the digital inputs, but over RS485.  If that's the case, then this pump is not like the VS at all despite the same 2 letters in the name.

At this point, I can only suggest going through the steps in the manual (make sure you have the right address, the pump is in Start mode, etc.  We very well may need to wait until somebody with an OCP connected to the VST can do a packet capture for us.  We don't want to start spamming the pump with messages to see what will stick because you could brick the pump (it's happened with the IntelliValve).  

For now, you could fully use the VST with the 4 pre-programmed speeds by utilizing [REM](https://github.com/rstrouse/relayEquipmentManager) and using the other external control cable and having Nixie control this as a relay pump.  The Hayward Relay VS may be an equivalent and you might want to try that with 4 relay controls until some kind soul comes along where we can snoop on the messages.

[image]

ddyer8, 2022-05-23:
Yes, the pump address is 1 and the pump is in start mode. The older models could only use relays but the model I have lists RS-485. I followed the steps on page 5 in the manual:

https://www.pentair.com/content/dam/extranet/aquatics/pool-pad-pro-assets/pumps/inground-pumps/superflo-vs-variable-speed-pump/356292-superflovs-superflovst-supermaxvs-owners-manual.pdf

tagyoureit, 2022-05-23:
Have you tried swapping the yellow/green cables?  

The error in the logs `Error sending setPumpManual for Pump` means we cannot gain control of the pump.

tagyoureit, 2022-05-23:
The other thing to look for is to make sure the MegaBas RS485 is setup correctly.  Do you have a /dev/ttyS0 device?

ddyer8, 2022-05-24:
So I don’t have access to the system right now, but with raspi-config, this is done by setting serial console enable to NO and UART enable to YES under the serial interface options, correct?

Do i need to issue commands to the megabas rs485 config? I did so, but I don’t know if it’s persistent. I was also wondering if this was handled by the config in the REM menu. 

It seems like my next steps are to verify the existence of the ttyS0 port and perhaps inspect the output of the rs485 connections. Can this be done through an application on the pi? Or do I need to connect another device like an arduino and decode the packets?

rstrouse, 2022-05-24:
You should probably update your firmware on the MEGA-BAS.  It is very important that nothing that could be accessing any facet of the BAS be open at the time you update the firmware.   It it is interrupted the card can start acting strange until you get a good firmware update without any interruption.  To update the firmware download the CLI for the MEGA-BAS from here.
https://github.com/SequentMicrosystems/megabas-rpi

Then follow the instructions in the update firmware link.

After you have successfully updated the firmware issue the following commands assuming you didn't change the card stack jumpers.  If you did replace the 0 on the megabas command to the stack level for the card <0..7>.  This will ensure the MODBUS client is released for the card and sets the UART to the proper settings.

`cd ~/megabas-rpi`
`megabas 0 rs485wr 0 9600 1 0 1`

ddyer8, 2022-05-28:
Ok, I verified I can communicate with a modbus temp/humidity sensor through the megabas card now.  It'll be next week before I can test the pump and chlorinator again.

In the meantime, is there a way to feed the modbus device info to nixie pool?  Since I have it already, I'm tempted to deploy inside my electronics enclosure.

rstrouse, 2022-05-29:
You can feed any input reported by the BAS.  However, which modbus sensor are you using?

rstrouse, 2022-05-29:
Also pull njsPC and dashPanel we have added some additional message detection for the Hayward protocol.

ddyer8, 2022-05-29:
It’s sht20 temp/humidity. I think there are i2c versions out there but mine is modbus. I see REM is configured to display all dedicated inputs. Are there instructions for custom serial inputs?

Thank you,
Dustin

rstrouse, 2022-05-30:
Unfortunately, we have not written code to pull data from RS485 sensors like the sht20.  There are a ton of other options including low cost 10k sensors on ADC or SPI, 1-Wire DHT, I2c sensors from Atlas EZO-HUM, 4-20ma feeds, and quite a few others.   Unfortunately, chasing the digital protocol each of these devices would prove waaaaay to difficult as you would need an example sitting on the desk.  I have not found any to match their documentation completely.

rstrouse, 2022-05-30:
Btw please pull njsPC as we have been working with all of you to work out any issues with the Hayward VS pumps.  The SuperFlo should simply act like an IntelliFlo VS so it would be good to get a replay output.  Either it is not receiving messages from the pump or there are no comms at all on that inbound for your connection.  The changes we recently posted up better detects the second scenario.

rstrouse, 2022-06-04:
Did you get your pump communicating?

ddyer8, 2022-06-05:
I just got home from vacation, and my pump is working now. I can change speeds and verify however, I am still getting this warn and error repeatedly in the logs. Is this expected?

3|njsPC    | [6/5/2022, 11:29:02 AM] debug: Begin sending chlorinator messages AquaRite
3|njsPC  | [6/5/2022, 11:29:03 AM] verbose: Wrote packet[255,0,255,165,0,96,33,5,1,6,1,50].Retries remaining: 1
3|njsPC  | [6/5/2022, 11:29:03 AM] verbose: Suspending status check: true -- 1
3|njsPC  | [6/5/2022, 11:29:03 AM] verbose: Suspending status check: false -- 0
3|njsPC  | [6/5/2022, 11:29:04 AM] verbose: Wrote packet[255,0,255,165,0,96,33,5,1,6,1,50].Retries remaining: 0
3|njsPC  | [6/5/2022, 11:29:05 AM] warn: Message aborted after 3 attempt(s): 165,0,96,33,5,1,6,1,50
3|njsPC  | [6/5/2022, 11:29:05 AM] error: Error sending setPumpManual for Pump: Message aborted after 3 attempt(s): 165,0,96,33,5,1,6,1,50
3|njsPC  | [6/5/2022, 11:29:05 AM] verbose: Wrote packet[16,2,80,0,0,98,16,3].Retries remaining: 3
3|njsPC  | [6/5/2022, 11:29:05 AM] verbose: Wrote packet[16,2,80,17,45,160,16,3].Retries remaining: 7
3|njsPC  | [6/5/2022, 11:29:05 AM] verbose: Wrote packet[16,2,80,20,0,118,16,3].Retries remaining: 3
3|njsPC  | [6/5/2022, 11:29:06 AM] verbose: Wrote packet[255,0,255,165,0,96,33,7,0,1,45].Retries remaining: 2
3|njsPC  | [6/5/2022, 11:29:06 AM] verbose: Wrote packet[255,0,255,165,0,96,33,4,1,255,2,42].Retries remaining: 1
3|njsPC  | [6/5/2022, 11:29:06 AM] verbose: Suspending status check: true -- 1
3|njsPC  | [6/5/2022, 11:29:06 AM] verbose: Suspending status check: false -- 0
3|njsPC  | [6/5/2022, 11:29:08 AM] verbose: Wrote packet[255,0,255,165,0,96,33,6,1,10,1,55].Retries remaining: 1
3|njsPC  | [6/5/2022, 11:29:08 AM] verbose: Wrote packet[255,0,255,165,0,96,33,1,4,2,196,11,184,2,180].Retries remaining: 1
3|njsPC  | [6/5/2022, 11:29:08 AM] verbose: Wrote packet[255,0,255,165,0,96,33,5,1,6,1,50].Retries remaining: 2

tagyoureit, 2022-06-05:
Can you upload another replay capture?  We need to see the whole picture to get a better understand of what's going on.

ddyer8, 2022-06-05:
Is this what you need?  I'm not confident I'm executing the capture properly.

[replay (1).zip](https://github.com/tagyoureit/nodejs-poolController/files/8840919/replay.1.zip)

racerx9146, 2022-06-06:
FYI...I have a older intelliflow (like 12 years at least) configured as a VS. Everything works and I can ramp up/down RPM via the the dash panel on the fly. But my logs are full of same error

[6/6/2022, 10:05:52 AM] error: Error sending setPumpManual for Intelliflo: Message aborted after 3 attempt(s): 165,0,96,33,5,1,6,1,50
[6/6/2022, 10:05:59 AM] warn: Message aborted after 3 attempt(s): 165,0,96,33,5,1,6,1,50
[6/6/2022, 10:05:59 AM] error: Error sending setPumpManual for Intelliflo: Message aborted after 3 attempt(s): 165,0,96,33,5,1,6,1,50
[6/6/2022, 10:06:06 AM] warn: Message aborted after 3 attempt(s): 165,0,96,33,5,1,6,1,50
[6/6/2022, 10:06:06 AM] error: Error sending setPumpManual for Intelliflo: Message aborted after 3 attempt(s): 165,0,96,33,5,1,6,1,50
[6/6/2022, 10:06:12 AM] warn: Message aborted after 3 attempt(s): 165,0,96,33,5,1,6,1,50
[6/6/2022, 10:06:12 AM] error: Error sending setPumpManual for Intelliflo: Message aborted after 3 attempt(s): 165,0,96,33,5,1,6,1,50

rstrouse, 2022-06-06:
Pull njsPC.  I have omitted the action 5 request from njsPC for vanilla VS pumps.

ddyer8, 2022-06-06:
That eliminated the warning and error for me.

Maybe off topic, but when I do the git pulls and subsequent npm installs, I get warnings about vulnerabilities.  should I be doing audit-fixes? npm updates?

racerx9146, 2022-06-06:
I did and pull and I dont see the error anymore. Thx

rstrouse, 2022-06-06:
No.  Almost daily there is some esoteric thing that is changed in node and its dependencies.   Often these are the result of some vulnerability that does not affect this software.  Unfortunately, the audit fix is sometimes a mess so just do npm i after git pull and you will be fine.  npm update will often break the dependencies.

rstrouse, 2022-06-07:
I am going to close this so if you have any further issues please post in the discussions or open a new issue.

closed 2022-06-07 by rstrouse