tagyoureit/nodejs-poolController #481: add orpVolumeDosed to mqtt.json

johnny2678, 2022-06-09:
## 💥 Proposal

add orpVolumeDosed to mqtt.json

### What feature you'd like to see

for ORP dynamic dosing, the lbs of Cl are calculated by njspc and reported in dashpanel and the logs.  This will make those values available to mqtt.

```
        {
          "topic": "state/chemControllers/@bind=data.id;/@bind=data.name;/orpVolumeDosed",
          "message": "{\"orpVolumeDosed\":@bind=data.orp.volumeDosed;}"
        },
```

## Motivation

Tracking total lbs of Cl dosed allows for limiting mechanisms to be enforced.  ex. if lbs of Cl > 0.12 then disable dosing for the day

## Pitch

Will attempt this pull request myself as it's a purely selfish request so I can get back on the latest code branch without having to move mqtt.json out before doing a `git pull` and then replace.

allows for graphs like this:
[image]

johnny2678, 2022-06-09:
hmm... are you referring to MQTT alt?  Not clear where to add my own bindings in the interfaces tab.

[image]

rstrouse, 2022-06-09:
Nope if you update your njsPC and dashPanel (yeah you have to save off your json this time) then you can click the upload bindings dialog to add it to the list of installed bindings.  Custom bindings will be located in `web/bindings/custom`  directory and not interfere with git.  From there you can edit the file all you like and it will be all good.  
[image]

Now that I know you are using a custom MQTT binding, I amend my prior statement about code cleanup.  Previously, njsPC was not releasing the subscriptions properly and it is likely that every time you saved the custom bindings file it would not wait until it tore down the old subscriptions before creating new ones.  This probably made skeeter cut us off.

rstrouse, 2022-06-09:
Did you know there was a chemicalDose event that is fired at the start, end and  of every chemical dose?  It contains the name of the chemical it dosed, the volume it dosed, the time it spent dosing, what the demand was when it was emitted, and the setpoint.  So the data structure looks like this,

```
{
   id: number - id of the chem controller,
   chem: string - The chemical being dosed [acid/orp/base]
   method: string - auto/manual,
   start: datetime - The date and time the dose started,  A combination of the id, chem, and start fields uniquely identify a dose
   end: datetime - The end time for the dose or null if the dose is currently underway,
   demand: number - The current chemical demand,
   level: number -  The current ph or orp level
   volume: number - The calculated dose volume
   time: number - The calculated dose time in seconds 
   status: string [cancelled/completed/dosing]
   volumeDosed: number - The total volume dosed so far
   timeDosed: number - The total number of seconds that have been dosed
 }
```

johnny2678, 2022-06-09:
Is there a format for the custom binding file?

Will it just take the code snippet?
```
        {
          "topic": "state/chemControllers/@bind=data.id;/@bind=data.name;/orpVolumeDosed",
          "message": "{\"orpVolumeDosed\":@bind=data.orp.volumeDosed;}"
        }
```

 or do I need to wrap it with the same json that's in `mqtt.json`
```
{
  "context": {
    "name": "MQTT",
    "options": {
      "formatter": [
        {
...
```

Actually, to idiot proof this for me, could you post what file format dash is looking for so I can add `orpVolumeDosed` into MQTT? 😄

rstrouse, 2022-06-09:
It will be your own interface and you can copy the existing file that is MQTT.json then make your edits.

johnny2678, 2022-06-09:
ok, so no worries about dupes if fields are in mqtt.json and also in my own file?

johnny2678, 2022-06-09:
or do I turn off MQTT and just use my custom binding instead?

rstrouse, 2022-06-09:
Turn off MQTT and use your custom bindings.

johnny2678, 2022-06-09:
done!

So periodically I'll just do a diff/merge with the git version to keep my json up to date.

thanks for the help.

johnny2678, 2022-06-09:
Guess I can cancel my pull request ;)

johnny2678, 2022-06-09:
Ok, my MQTT custom binding works.

Tried to do the same thing with Influx (turn off stock binding and add my custom one), and I'm getting an unhandled exception

```
1|poolController  | [6/9/2022, 6:17:35 PM] info: [6:17:35 PM] 192.168.5.146 PUT /app/interfaces/add {"name":"Influx-JH","type":"influx","fileName":"custom/influxjh.json"}
1|poolController  | unhandled rejection
1|poolController  |   == Node detected an unhandled rejection! ==
1|poolController  |   An invalid type was specified influx
1|poolController  |   ApiError: An invalid type was specified influx
1|poolController  |       at /home/pi/nodejs-poolController/web/services/utilities/Utilities.ts:83:47
1|poolController  |       at Layer.handle [as handle_request] (/home/pi/nodejs-poolController/node_modules/express/lib/router/layer.js:95:5)
1|poolController  |       at next (/home/pi/nodejs-poolController/node_modules/express/lib/router/route.js:144:13)
1|poolController  |       at Route.dispatch (/home/pi/nodejs-poolController/node_modules/express/lib/router/route.js:114:3)
1|poolController  |       at Layer.handle [as handle_request] (/home/pi/nodejs-poolController/node_modules/express/lib/router/layer.js:95:5)
1|poolController  |       at /home/pi/nodejs-poolController/node_modules/express/lib/router/index.js:284:15
1|poolController  |       at Function.process_params (/home/pi/nodejs-poolController/node_modules/express/lib/router/index.js:346:12)
1|poolController  |       at next (/home/pi/nodejs-poolController/node_modules/express/lib/router/index.js:280:10)
1|poolController  |       at /home/pi/nodejs-poolController/web/Server.ts:705:25
1|poolController  |       at Layer.handle [as handle_request] (/home/pi/nodejs-poolController/node_modules/express/lib/router/layer.js:95:5)
1|poolController  |       at trim_prefix (/home/pi/nodejs-poolController/node_modules/express/lib/router/index.js:328:13)
1|poolController  |       at /home/pi/nodejs-poolController/node_modules/express/lib/router/index.js:286:9
1|poolController  |       at Function.process_params (/home/pi/nodejs-poolController/node_modules/express/lib/router/index.js:346:12)
1|poolController  |       at next (/home/pi/nodejs-poolController/node_modules/express/lib/router/index.js:280:10)
1|poolController  |       at /home/pi/nodejs-poolController/node_modules/body-parser/lib/read.js:137:5
1|poolController  |       at AsyncResource.runInAsyncScope (async_hooks.js:190:9)
1|poolController  | [6/9/2022, 6:17:37 PM] info: [6:17:37 PM] 192.168.5.140 GET /state/all/ {}

```

Double checked the json format was valid.  Any ideas?

attaching my json if it helps...
[influx-jh.txt](https://github.com/tagyoureit/nodejs-poolController/files/8874272/influx-jh.txt)
?

rstrouse, 2022-06-09:
That is an undocumented and unwanted feature.  Somebody did not include the default options for the type.  Not going to name names but I shower with him all the time.  Pull njsPC.

closed 2022-06-22 by tagyoureit