tagyoureit/nodejs-poolController #486: [BUG] Not recognizing Sta-Rite Intellipro VSF

guru-florida, 2022-06-15:
### nodejs-poolController Version/commit

0a6e754  [7.7.0]

### nodejs-poolController-dashPanel Version/commit

3328fdf  [7.7.0]

### Node Version

12.2

### Platform

Linux raspberrypi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux

### RS485 Adapter

JBTek (I think)

### Are you using Docker?

- [ ] Yes.

### OCP

Easytouch 8

### Pump(s)

Intellipro VSF

### Chlorinator(s)

IC-20

### What steps will reproduce the bug?

Buy an intellipro VSF   ;)

### What happens?

The Intellipro VSF is by Sta-Rite but was bought by Pentair years ago. It wires into my Easytouch controller directly and the ET recognizes and controls it. It's apparently the same as Intelliflo VSF...but IDK. I am wondering if it's supported and what I can do to diagnose the RS485 messaging on it. I turned on only broadcast and pump logging and I do get these protocol messages:
```
[6/15/2022, 3:12:46 PM] info: {"port":0,"id":54389,"valid":true,"dir":"in","proto":"pump","pkt":[[],[255,0,255],[165,0,16,96,4,1],[255],[2,25]],"ts": "2022-06-15T15:12:46.085+0100"}
[6/15/2022, 3:12:46 PM] info: {"port":0,"id":54390,"valid":true,"dir":"in","proto":"pump","pkt":[[],[255,0,255],[165,0,96,16,6,1],[10],[1,38]],"ts": "2022-06-15T15:12:46.102+0100"}
[6/15/2022, 3:12:46 PM] info: {"port":0,"id":54391,"valid":true,"dir":"in","proto":"pump","pkt":[[],[255,0,255],[165,0,16,96,6,1],[10],[1,38]],"ts": "2022-06-15T15:12:46.126+0100"}
[6/15/2022, 3:12:46 PM] info: {"port":0,"id":54392,"valid":true,"dir":"in","proto":"pump","pkt":[[],[255,0,255],[165,0,96,16,1,4],[3,33,0,8],[1,70]],"ts": "2022-06-15T15:12:46.146+0100"}
[6/15/2022, 3:12:46 PM] info: {"port":0,"id":54393,"valid":true,"dir":"in","proto":"pump","pkt":[[],[255,0,255],[165,0,16,96,1,2],[0,8],[1,32]],"ts": "2022-06-15T15:12:46.167+0100"}
[6/15/2022, 3:12:47 PM] info: {"port":0,"id":54394,"valid":true,"dir":"in","proto":"broadcast","pkt":[[255,255,255,255,255,255,255,255],[255,0,255],[165,32,15,16,2,29],[10,16,32,0,0,0,0,0,0,32,0,0,0,4,91,91,0,0,101,0,0,0,0,0,0,72,80,0,14],[3,34]],"ts": "2022-06-15T15:12:47.910+0100"}
[6/15/2022, 3:12:49 PM] info: {"port":0,"id":54395,"valid":true,"dir":"in","proto":"broadcast","pkt":[[255,255,255,255,255,255,255,255,255,255,255],[255,0,255],[165,32,15,16,2,29],[10,16,32,0,0,0,0,0,0,32,0,0,0,4,91,91,0,0,101,0,0,0,0,0,0,72,80,0,14],[3,34]],"ts": "2022-06-15T15:12:49.924+0100"}
[6/15/2022, 3:12:51 PM] info: {"port":0,"id":54396,"valid":true,"dir":"in","proto":"broadcast","pkt":[[255,255,255,255,255,255,255,255,255,255,255,255,255,255],[255,0,255],[165,32,15,16,2,29],[10,16,32,0,0,0,0,0,0,32,0,0,0,4,91,91,0,0,101,0,0,0,0,0,0,72,80,0,14],[3,34]],"ts": "2022-06-15T15:12:51.931+0100"}
[6/15/2022, 3:12:53 PM] info: {"port":0,"id":54397,"valid":true,"dir":"in","proto":"broadcast","pkt":[[255,255,255,255,255,255,255,255,255,255,255,255,255,255],[255,0,255],[165,32,15,16,2,29],[10,16,32,0,0,0,0,0,0,32,0,0,0,4,91,91,0,0,101,0,0,0,0,0,0,72,80,0,14],[3,34]],"ts": "2022-06-15T15:12:53.938+0100"}
[6/15/2022, 3:12:55 PM] info: {"port":0,"id":54398,"valid":true,"dir":"in","proto":"broadcast","pkt":[[255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255],[255,0,255],[165,32,15,16,2,29],[10,16,32,0,0,0,0,0,0,32,0,0,0,4,91,91,0,0,101,0,0,0,0,0,0,72,80,0,14],[3,34]],"ts": "2022-06-15T15:12:55.945+0100"}
[6/15/2022, 3:12:57 PM] info: {"port":0,"id":54399,"valid":true,"dir":"in","proto":"broadcast","pkt":[[255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255],[255,0,255],[165,32,15,16,2,29],[10,16,32,0,0,0,0,0,0,32,0,0,0,4,91,91,0,0,101,0,0,0,0,0,0,72,80,0,14],[3,34]],"ts": "2022-06-15T15:12:57.960+0100"}
[6/15/2022, 3:13:00 PM] info: {"port":0,"id":54400,"valid":true,"dir":"in","proto":"pump","pkt":[[255,255,255,255,255,255,255,255],[255,0,255],[165,0,96,16,4,1],[255],[2,25]],"ts": "2022-06-15T15:13:00.138+0100"}
[6/15/2022, 3:13:00 PM] info: {"port":0,"id":54401,"valid":true,"dir":"in","proto":"pump","pkt":[[],[255,0,255],[165,0,16,96,4,1],[255],[2,25]],"ts": "2022-06-15T15:13:00.169+0100"}
[6/15/2022, 3:13:00 PM] info: {"port":0,"id":54402,"valid":true,"dir":"in","proto":"pump","pkt":[[],[255,0,255],[165,0,96,16,6,1],[10],[1,38]],"ts": "2022-06-15T15:13:00.186+0100"}
[6/15/2022, 3:13:00 PM] info: {"port":0,"id":54403,"valid":true,"dir":"in","proto":"pump","pkt":[[],[255,0,255],[165,0,16,96,6,1],[10],[1,38]],"ts": "2022-06-15T15:13:00.209+0100"}
[6/15/2022, 3:13:00 PM] info: {"port":0,"id":54404,"valid":true,"dir":"in","proto":"pump","pkt":[[],[255,0,255],[165,0,96,16,1,4],[3,33,0,8],[1,70]],"ts": "2022-06-15T15:13:00.228+0100"}
[6/15/2022, 3:13:00 PM] info: {"port":0,"id":54405,"valid":true,"dir":"in","proto":"pump","pkt":[[],[255,0,255],[165,0,16,96,1,2],[0,8],[1,32]],"ts": "2022-06-15T15:13:00.249+0100"}
```

### What should have happened?

 Expecting the pump to be recognized in njsPC or once selected in config, and report the pump RPM.

### Additional information

I have experience in RS45, javascript/typescript and this sort of thing but nothing much in particular to proprietary Pentair protocols. So with some guiding help here I may be able to add the necessary support for this pump.

rstrouse, 2022-06-15:
Interesting.  I thought we had users with IntelliFlo / pro VSF already.  If you want real time display of the messages you can use messageManager.  https://github.com/rstrouse/nodejs-poolController-dashPanel/wiki/Message-Manager

It is possible that the pump output from the Sta-Rite version is sending a different action byte for its status but you should be able to see the output in the messageManager.  Likely rpm values will be a series of 2 bytes where byte0 * 256 + byte1 = rpm and the watts will be the same.  So if the pump is emitting a status this will be a message where the source is the pump and at least a 4 byte payload.

guru-florida, 2022-06-15:
I like the message manager!  So there are messages exchanging between OCP and Pump.
* OCP sets runstate to 0xFF, Pump replies 0xFF
* OCP sets drivestate to 0x0A, Pump replies 0x0A
* OCP sets speed/flow to 0x03 21 00 08, Pump replies 0x00 08

My pump is set to 2200RPM right now. I dont seen any flow of numbers in that last report 0x03210008 that could be 2200 no matter little or big endian. Also, pump responding with only 0,8.

Any suggestions?

closed 2024-06-10 by tagyoureit