tagyoureit/nodejs-poolController #510: Timezone & Locality 'Units' option not available in 7.7.0

theRealMCQwerty, 2022-06-28:
Didn't want to log this as a bug as it appears to be working for others. Not sure if this is a Nixie only feature or if it can be done for my Intellicenter i8PS?

I saw this issue as I would like to set my temp units to deg C: https://github.com/tagyoureit/nodejs-poolController/issues/341

I updated and am currently running v 7.7.0 (see screenshot)
[image]

But I haven't got the 'Units' option in the Timezone and Locality settings (see screenshot)
[image]

Thanks

theRealMCQwerty, 2022-06-28:
@rstrouse ah bummer. Could the temp conversion be done on the fly in the UI only?

rstrouse, 2022-06-28:
Have you checked the OCP?  Maybe that option is an OCP only option.

closed 2022-06-29 by theRealMCQwerty