tagyoureit/nodejs-poolController #631: [BUG] Intellichem Tank Level reset to 0

snagytx, 2022-11-12:
### nodejs-poolController Version/commit

7.7.0/79f330b

### nodejs-poolController-dashPanel Version/commit

7.7.0/0193678

### relayEquipmentManager Version/commit

none

### Node Version

docker

### Platform

docker

### RS485 Adapter

Wlfin

### Are you using Docker?

- [X] Yes.

### OCP

Intellicenter i5PS

### Pump(s)

Intelliflo VS & WisperFlow

### Chlorinator(s)

Intellichem

### What steps will reproduce the bug?

Reset Alkalinity Level

### What happens?

The Tank Level for ORP and Acid are reset to 0

### What should have happened?

The Tank Level for ORP and Acid should remain untouched

### Additional information

When I change the alkalinity this message is sent:

PUT http://XXXX:4200/state/chemController
`{"id":1,"ph":{"setpoint":7.5,"dailyVolumeDosed":1000},"orp":{"setpoint":700,"dailyVolumeDosed":10},"alkalinity":95,"calciumHardness":350,"cyanuricAcid":40,"borates":0,"saturationIndex":"-0.1"}`

In messageManager I see:

[image]
[image]

rstrouse, 2022-11-13:
I posted up a change that I think fixes the issue.  However, I never did get a replay from somebody with IntelliChem while the tank levels were adjusted from IntelliChem.  At the time I think that the IntelliCenter web app was not working for this function.   
Can I get a replay where you set the tank levels from IntelliCenter?  

The message will look something like below where byte indexes 11 and 12 define the tank level + 1.  So byte 11 set to 7 means the tank level is 6 in IntelliChem for pH (byte[11]) and 0 for chlorine (byte[12]).
[image]

snagytx, 2022-11-14:
You can't set the tank level from IntelliCenter. But you can adjust the Alkalinity, Calcium Hardness and Cyanuric Acid level.

I ran a test to increase Alkalinity from 80 to 81:
[image]
response:
[image]

Calcium Hardness from 350 to 351:
[image]
response:
[image]

Cyanuric Acid from 40 to 41:
[image]
response:
[image]

The actual Chlorine Tank level is 4 while the Acid tank level is 2. Since they can't be tweaked from the web UI, I cant capture that. But I went outside and adjusted the Tank levels from IntelliCenter directly and I got:

For Chrorine - changed from 2 to 3:
[image]
Response:
[image]

Acid - changed from 4 to 5:
[image]
Response:
[image]

tagyoureit, 2022-11-14:
The screenshots are helpful, but @rstrouse is referring to a [packet capture](https://github.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution#-dashpanel). Can you turn on the capture and run the same exact scenario and then upload the resulting files?  Often we can see what's going on with the packets and pick up nuances that are otherwise not available to us just by screen shots. But it does sound like you are on the right track.

snagytx, 2022-11-14:
I have I have dropped all numbers by 1 unit. See attached the logs.
[replay (1).zip](https://github.com/tagyoureit/nodejs-poolController/files/10003950/replay.1.zip)

rstrouse, 2022-11-14:
Did you pull the changes from the repo before running the capture?  I believe you should be able to even set the tank levels from dashPanel if you click/tap on the tank.   The output from njsPC is identical to the setting changes from IntelliCenter but only if you pull the latest code.

snagytx, 2022-11-14:
my bad, I bad, I didn't pull the latest docker image, from the screenshots it seemed that the OCP sends to Chem to set bytes 4 & 5, not 11 & 12, I though that the change would work, I should have still pulled and tested.

I just did that and it seems to work to asjust Alkalinity, Calcium Hardness, Cyanuric Acid and Tank levels.

closed 2022-11-14 by snagytx

rstrouse, 2022-11-14:
Awesome.  It looks like you can actually set tank levels from njsPC but not IntelliCenter... funny.

Keep in mind that there are more than one conversation going on with the messages related to IntelliChem.

1. There is the conversation between any external controller to the OCP.  This communication is done through action 168 messages with the first two bytes being [8, 0] and the responses from action 30 and 222 with the first two bytes being [8, 0] and [8, 1].  This is the conversation we are concerned with.
2. Then the OCP communicates with IntelliChem using action 210 with a response of action 18 to get the status.  When the OCP sets a value to IntelliChem it sends this on action 146 and simply gets an ACK response when IntelliChem receives it.

The conversation from your screenshots above is the one between IntelliChem and the OCP.  The messageManager filter should allow you to identify the proper conversations in the filter.