tagyoureit/nodejs-poolController #637: MEGA-BAS errors

spifftek70, 2022-11-15:
### nodejs-poolController Version/commit

7.7.0 / 18396ae

### nodejs-poolController-dashPanel Version/commit

7.7.0 / 78c01be

### relayEquipmentManager Version/commit

7.5.0

### Node Version

v12.18.3

### Platform

Raspberry pi 4

### RS485 Adapter

ttyUSB0  Building Automation  8-Layer Stackable HAT

### Are you using Docker?

- [ ] Yes.

### Pump(s)

 Intelliflo VSF and two Single Speed 2.5HP

### What steps will reproduce the bug?

Can someone tell me what is causing THESE error messages from REM?

```
|││ REM > error: Error reading Sequent MEGA-BAS register 3: , Remote I/O error                           │
│││ REM > info: [16:41:23] 192.168.111.147 GET /devices/state?null {}                                    │
│││ REM > error: Sequent MEGA-BAS error getting source voltage: , Remote I/O error                       │
│││ REM > error: Sequent MEGA-BAS error getting Raspberry Pi voltage: , Remote I/O error                 │
│││ REM > error: Sequent MEGA-BAS error getting cpu temp: , Remote I/O error                             │
│││ REM > error: Error sending on internal connection generic:5:1: Maximum call stack size exceeded      │
│││ REM > error: Error sending on internal connection generic:5:2: Maximum call stack size exceeded      │
│││ REM > error: Error sending on internal connection generic:5:2: Maximum call stack size exceeded      │
│││ REM > error: Error sending on internal connection generic:5:2: Maximum call stack size exceeded      │
│││ REM > info: [16:41:30] 192.168.111.147 GET /config/options/i2c/1?null {}                             │
│││ REM > info: [16:41:30] 192.168.111.147 GET /config/options/genericDevices?null {}                    │
│││ REM > error: Error reading Sequent MEGA-BAS register 3: , Remote I/O error                           │
│││ REM > error: Sequent MEGA-BAS error getting source voltage: , Remote I/O error                       │
│││ REM > error: Sequent MEGA-BAS error getting Raspberry Pi voltage: , Remote I/O error                 │
│││ REM > error: Sequent MEGA-BAS error getting cpu temp: , Remote I/O error                             │
│││ REM > error: Error sending on internal connection generic:5:1: Maximum call stack size exceeded      │
│││ REM > error: Error sending on internal connection generic:5:2: Maximum call stack size exceeded      │
│││ REM > error: Error sending on internal connection generic:5:2: Maximum call stack size exceeded      │
│││ REM > error: Error sending on internal connection generic:5:2: Maximum call stack size exceeded ```

### What happens?

No connection to anything connected to the Building Automation 
8-Layer Stackable HAT.  But the Eight Relays 8-Layer Stackable HAT connections work fine and it connects into the BA stack.

No temps and no RS485 connectivity. 

### What should have happened?

Temps and comms

rstrouse, 2022-11-16:
Interesting.  Shut down REM and follow the instructions to update the firmware on the BAS from the repo.

https://github.com/SequentMicrosystems/megabas-rpi/blob/master/update/README.md

tagyoureit, 2022-11-16:
I've seen this a number of times on my own system but have not been able to mail down why. Check disk space to make sure it isn't full. However, more often this feels like a hardware error where only a hard power cycle will reset the i2c bus.

spifftek70, 2022-11-16:
Tried those both already...  Is Node v12.18.3 the best version to use?

rstrouse, 2022-11-16:
Give me a little background.  Was this working before?

Some things to check
1. Ensure the J3 jumpers are installed.  This is RXD and TXD
2. Run the following commands and verify the results look like the following 
```
cd ~/megabas-rpi
megabas 0 rs485rd
```
The results should look like this
  `<mode> <baudrate> <stopbits> <parity> <add> 0 9600 1 0 1`

rstrouse, 2022-11-16:
Oh wait!  You are using a MEGA-BAS.  The port is not on USB it should be using `/dev/ttyS0`.  I should have read your settings much more carefully.

spifftek70, 2022-11-16:
damn!  that's right.  do I need to install something?

rstrouse, 2022-11-16:
Nope all you should need to do is type in `/dev/ttyS0` for the port.  For that matter if you have the jumpers installed and the RS485 pass thru is enabled on the BAS it will probably show up in the dropdown.
[image]

spifftek70, 2022-11-16:
YAY.  that did it.

closed 2022-11-16 by spifftek70