tagyoureit/nodejs-poolController #682: [BUG] Salt Cell Sensor Errors

FlaMike, 2023-01-17:
### nodejs-poolController Version/commit

7.7.0

### nodejs-poolController-dashPanel Version/commit

7.7.0

### relayEquipmentManager Version/commit

N/A

### Platform

Ubuntu 22.04 LTS server

### RS485 Adapter

Moyina 

### Are you using Docker?

- [X] Yes.

### OCP

EasyTouch 2 4

### Pump(s)

Intelliflo VSF

### Chlorinator(s)

Intellichlor 20

### What steps will reproduce the bug?

There appear to be two sensor reporting errors that relate to the salt cell on my system.
   -  current output is reported as equal to the setpoint (8% in my case) even though the water in the pool is too cold for the cell to be running (water is 53 degrees currently and the red "cold water" LED is illuminated on the cell itself)
   - the salt required to reach the setpoint is reported as zero even though the most recent salt concentration is reported as 2850 and the target level is 3400

The additional information section below shows the current dump of the chlorinator state, showing the details.

### What happens?

As above

### What should have happened?

As above

### Additional information

Output from http://<ipaddress>:4200/state/chlorinator/1

{"id":1,"disabled":false,"isActive":true,"spaSetpoint":10,"poolSetpoint":8,"body":{"val":32,"name":"poolspa","desc":"Pool/Spa"},"name":"Intellichlor--20","type":{"val":0,"name":"pentair","desc":"Pentair"},"saltLevel":2850,"saltRequired":0,"status":{"val":0,"name":"ok","desc":"Ok"},"superChlorHours":8,"superChlor":false,"superChlorRemaining":0,"targetOutput":8,"model":{"val":2,"name":"intellichlor--20","desc":"IntelliChlor IC20","capacity":20000,"chlorinePerDay":0.7,"chlorinePerSec":0.000008101851851851852},"lastComm":1673984950152,"currentOutput":8,"equipmentType":"chlorinator","saltTarget":3400}

tagyoureit, 2023-01-18:
Sometimes the salt cell doesn't report when it is below the freeze temp.  Thus, we won't get the proper updates.  But the way we would know is to upload a [packet capture](https://github.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution#-dashpanel).  Then we can look at the actual packets and see what is really being reported.

FlaMike, 2023-01-18:
Here is the packet capture from the pool

[pollcontrollerreplay1.18.2023.zip](https://github.com/tagyoureit/nodejs-poolController/files/10450262/pollcontrollerreplay1.18.2023.zip)

closed 2024-06-10 by tagyoureit