tagyoureit/nodejs-poolController #688: [BUG] Intelliflo speeds not working correctly

antamy, 2023-02-15:
### nodejs-poolController Version/commit

7.7.0 / master / 1a4b2d7

### Are you using Docker?

- [ ] Yes.

### What steps will reproduce the bug?

Steps:
- turn on logging
- wait for config reload
- access Pump detail page
- 4 pump speeds are displayed - Pool(2500), Spa(1830), Blank(2500), Air Blower(2500)
- change Spa(1830) to Spa(1800)
- Save Pump

### What happens?

- initial display of pump speeds shows a Blank, when it should be Solar.  This is displayed correctly on the Pentair panel
- after changing pump speed and saving, the 3rd and 4th pump speeds disappear
- on the Pentair panel, the 3rd and 4th pump speeds are now shown as None-450, None-450

If I now change the values on the Pentair panel back to Pool(2500), Spa(1800), Solar(2500), Air Blower(2500), njspc picks these up but displays  Pool(2500), Spa(1800), Blank(2500), Air Blower(2500)

### What should have happened?

Pump speed should have been changed and other values remain unaffected.

Note, that I also see variations of this problem where changing a speed adds a new pump speed to the list.  Even if the pump speeds are ordered in ascending numerical sequence, I get the same results.

This isn't a recent thing, it's been happening for months, but I haven't got around to reporting it.

### Additional information

[replay(4).zip](https://github.com/tagyoureit/nodejs-poolController/files/10744856/replay.4.zip)

rstrouse, 2023-02-16:
Thanks I'll have a look.  It looks like IntelliTouch is reporting a different virtual circuit id for solar than what we have.  I also noticed a SOLARVALVE feature in your setup.  I presume you are not setting a pump speed for that.  Am I correct?

antamy, 2023-02-16:
That's a hangover from our discussion this past weekend about running solar and heat simultaneously. Still haven't got that working.

If it helps, the solar circuit id is 128.  It's just odd that speed entries disappear or get added.

rstrouse, 2023-02-16:
It does that because IntelliTouch is confused that we sent it 128.  It expects a different number.  I'll have a looksee as to which number it expects for the solar virtual circuit.  This is what I presume is used for EasyTouch but these have been known to be different.

tagyoureit, 2023-02-16:
Digging into this, it looks like the values should be the following:

(Reporting values)
128: Solar Active
129: Pool or Spa Heater Active
130: Pool Heater Active
131: Spa Heater Active
132: Freeze Mode Active
133: Pool Remote Heat Boost
134: Pool Remote Heat Enable
135: Pump Increase Speed
136: Pump Decrease Speed
(Circuit values)
155: Pool Heater
156: Spa Heater
157: Either Heater
158: Solar
159: Freeze

Give 158 a try.

antamy, 2023-02-16:
This morning while everything was idle, I removed all of my custom speeds on the panel and started a packet capture.  I added the speeds one by one, saving and exiting out of the screens between each one.  I created Pool, Spa, Solar, Air Blower.

After this, njspc displayed Pool, Spa, blank, Air Blower and the values in data\poolConfig.json are:

          "id": 1,
          "master": 0,
          "circuit": 6,
          "speed": 1850,
          "units": 0
        },
        {
          "id": 2,
          "master": 0,
          "circuit": 1,
          "speed": 1850,
          "units": 0
        },
        {
          "id": 3,
          "master": 0,
          "circuit": 158,
          "speed": 2550,
          "units": 0
        },
        {
          "id": 4,
          "master": 0,
          "circuit": 2,
          "speed": 2550,
          "units": 0
        }

If I stop njspc, change circuit: 158 to circuit: 128, then restart, I now see Pool, Spa, Solar, Air Blower, but as soon as the reload finished, the blank re-appears.  When I change the speeds in njspc and Save, then only Pool and Spa appear, on both the panel and njspc.

[replay(5).zip](https://github.com/tagyoureit/nodejs-poolController/files/10756477/replay.5.zip)

rstrouse, 2023-02-19:
@antamy I need to understand what the replay above represents.  I looks like you added circuits 4 times to pump 1 from the indoor control panel.  If that is the case then 158 is the right value but we don't have it mapped correctly.

antamy, 2023-02-19:
I reset the pump so it had no custom speeds.  Then I added, one by one, 4 pump speeds. After adding one, I saved the configuration, then went back in and added the next.

Thanks for investigating this!

rstrouse, 2023-02-19:
Was this from the Indoor Control Panel?

rstrouse, 2023-02-19:
Alright I thing I remember our work to use the rule processor.  However, if you pull njsPC it should have the correct identifier for Solar now.  I think the IntelliTouch owner that we were working with originally didn't have solar set up.

Anyway I also debugged the rule for the rule processor that I originally gave you and it had some issues.  If you want one that works here is the rule.

```
{
  "context": {
    "name": "myPoolRules",
    "options": {},
    "vars": {
      "solarFeatureId": 41
    }
  },
  "events": [
    {
      "name": "temps",
      "description": "Process alt solar valve when temps change",
      "processor": [
        "try {",
        "let solarOn = false;",
        "for(let i = 0; i < data[0].bodies.length;i++) {",
        "let body = data[0].bodies[i];",
        "if(typeof body.heatStatus !== 'undefined' && (body.heatStatus.name === 'heater' || body.heatStatus.name === 'solar')) solarOn = true;",
        "}",
        "await sys.board.features.setFeatureStateAsync(vars.solarFeatureId, solarOn);",
        "} catch(err) { logger.error(`Error triggering solar feature ${err.message}`); }"
      ]
    }
  ]
}
```

antamy, 2023-02-19:
I did the git pull and restarted, but it's the same behavior.  I am going to have to leave this on hold for now since I am leaving the house today and won't be back for a few weeks.  I'll dive back into it when I return.  Thanks for the updated rule.

rstrouse, 2023-02-19:
Are you running 8.0.0?

rstrouse, 2023-02-20:
Do not run 8.0.0 until we post it to the master repo.  We have a few folks that are running this to help us test it out.

tagyoureit, 2023-03-27:
@antamy 8.0 is published.  Can you try this now?

antamy, 2023-03-27:
Thanks for letting me know. I'll be at that house in 3 weeks and will give it a try then.

antamy, 2023-04-21:
I looked at this again with v8.0.1.  It seems to work - I can change the active pump speed and that is reflected on dashPanel and also on the Pentair panel.  

There is an odd behavior though - change the speed and dismiss the Pump Settings dialog; speed changes and Pump info updates; click on Pump to bring up Pump Settings - this displays the prior speed.

Botched1, 2023-05-05:
Just a side note that there still seems to be some serious issues with Intelliflo pumps and the rpm/gpm settings. I can get both the circuits and speeds all scrambled just by going into the speed settings,. changing a speed, and getting out. If I do it multiple times, sometimes it comes back with everything scrambled/offset weird.

If I go into settings and change the pump setup gpm/rpm settings and try to save it, it just says saving pump forever and never clears.

johnny2678, 2023-05-05:
Not at home at the moment but can confirm I saw this last also on my ET-4 setup 

One of my speed circuits was like 34000 😂 not sure how it could have be set higher than the pump maximum of 3450 🤷‍♂️

 and my spa circuit was only 480 in the screens above in settings but when the Spa schedule actually runs it's at 2800, which is what it should be

Botched1, 2023-05-05:
What I find is that when this happens, it is not actually set to the values shown on the controller. It is cosmetic...

Well, cosmetic UNLESS you try to change a value thinking it is 'real' then it I've seen it try to write some odd things back down. That's why I do Reload Config when I see it like that.

tagyoureit, 2024-06-10:
Closing due to lack of activity.

closed 2024-06-10 by tagyoureit