tagyoureit/nodejs-poolController #692: Errors controlling Hayward Super Pump VS 700 [BUG]

camaro4life18, 2023-02-21:
### nodejs-poolController Version/commit

7.7.0

### nodejs-poolController-dashPanel Version/commit

7.7.0

### Node Version

v12.22.12

### Platform

Linux poolcontroller 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux

### RS485 Adapter

Industrial USB to RS485

### Are you using Docker?

- [ ] Yes.

### OCP

None

### Pump(s)

Hayward Super Pump VS 700

### Chlorinator(s)

Ezo Pump

### What steps will reproduce the bug?

@rstrouse Pointed me to using this app, and after following another issues thread, I was able to get my pump configured and working. It controls the RPM perfectly, gets the actual RPM response and watt usage. However my console log is full of errors about setting drive state and requestPumpStatus.

Also it seems odd that there is an info message to turn the pump on through relays. At one point I did have the config set to use Hayward VS Relays, but that is not the current config in dashpanel.

### What happens?

[21/02/2023, 14:00:47] info: NCP: Setting Circuit Pool to true
[21/02/2023, 14:00:48] info: NCP: Setting Pump Filter Pump relays to Relay 1: on, Relay 2: off, Relay 3: off, and Relay 4: off.
[21/02/2023, 14:00:48] warn: Message aborted after 3 attempt(s): 165,0,96,33,7,0,1,45
[21/02/2023, 14:00:48] error: Error sending requestPumpStatus for Filter Pump: Message aborted after 3 attempt(s): 165,0,96,33,7,0,1,45
[21/02/2023, 14:00:51] warn: Message aborted after 3 attempt(s): 165,0,96,33,7,0,1,45
[21/02/2023, 14:00:51] error: Error sending requestPumpStatus for Filter Pump: Message aborted after 3 attempt(s): 165,0,96,33,7,0,1,45
[21/02/2023, 14:00:53] warn: Message aborted after 2 attempt(s): 165,0,96,33,4,1,255,2,42
[21/02/2023, 14:00:53] error: Error sending setPumpToRemoteControl for Filter Pump: Message aborted after 2 attempt(s): 165,0,96,33,4,1,255,2,42
[21/02/2023, 14:00:55] warn: Message aborted after 2 attempt(s): 165,0,96,33,4,1,255,2,42
[21/02/2023, 14:00:55] error: Error sending setPumpToRemoteControl for Filter Pump: Message aborted after 2 attempt(s): 165,0,96,33,4,1,255,2,42

### What should have happened?

No errors in the log
Not trying to control the relays.

### Additional information

I did notice in the Hayward Display's menu, it only has a config Stand Alone or Relay control. Perhaps the setRemoteControl command doesn't work on this particular pump???

rstrouse, 2023-02-21:
Are you using relay control or RS485 control for the pump?  The message above probably is not required for this pump I just need to know which pump type you have selected.

camaro4life18, 2023-02-21:
RS485
[image]
[image]

rstrouse, 2023-02-21:
Pull njsPC and restart it.  I think I found all the spurious calls to the pump.

camaro4life18, 2023-02-22:
Looks good. No more errors and still controlling the pump.

closed 2023-02-22 by camaro4life18