tagyoureit/nodejs-poolController #695: Minimum OAT for Heater

camaro4life18, 2023-02-28:
## 💥 Proposal

### What feature you'd like to see

In the heater controls, I would like the ability to set a minimum OAT before the heater turns on.

## Motivation

I live in the upper midwest, where we get days with a wide temperature range. We can see 75 during the day, and 40 at night. The colder the OAT, the more heat lost from the pool. So I don't want my gas heater running when its too cold out. I would rather run it when its 75+.

## Pitch

Help users save money by maximizing energy efficiency.

rstrouse, 2023-02-28:
Are you using Nixie or do you have a Pentair OCP?

camaro4life18, 2023-02-28:
I am using Nixie.

camaro4life18, 2023-03-01:
Heh, is that a $2,000 annual cost? Because most homes here would easily burn through that depending on how early and late they open their pool. I opened my pool today, but really just so I could get the new pump hooked up and get everything setup in njsPC. My wife won't be very happy in a couple months if I'm still messing with automation.

You mentioned a rule processor, is this an existing feature, or under development? I started checking out all the code for njsPC, DashPanel, and REM. I've never done njs, mostly core-java development or stuff with spring-boot, but it all looks pretty similar. I got a local version of the 8.0.0 branch running, but haven't dug into any of the changes. I'll have to spend some time digging into your architecture, but it looks similar where I was planning to take my original app. Backend, front end, device comm. with REST calls between everything.

If you need any testing done just let me know. I may submit a pull request or 2 once I get familiar with everything. I have idea's for VS pump schedules, like run at 600RPM for parts of the day, just to keep water moving through the sensors. Then run at 3000 rpms for filtration. I still have to see what features are available already, but I would rather build onto an existing project than keep going down the path of building my own.

rstrouse, 2023-03-01:
That my friend is a month.

rstrouse, 2023-03-01:
The rule processor is an integration type driven by events in the system.  It captures events and provides hooks for executing code within its own sandbox.

The ability to manage schedules as you describe are part of the core function of the software.  Set up a test system and play around with a Nixie configuration.

tagyoureit, 2023-03-11:
> I have idea's for VS pump schedules, like run at 600RPM for parts of the day, just to keep water moving through the sensors. Then run at 3000 rpms for filtration. I still have to see what features [...]

The built-in schedules would already take care of this.  The rule processor is any Javascript code that we can inject into the process to add additional functionality (like OAT).  This is an example of a [rule](https://github.com/tagyoureit/nodejs-poolController/discussions/687#discussioncomment-4950270) you can check out.

tagyoureit, 2024-04-16:
Closing due to inactivity.

closed 2024-04-16 by tagyoureit