tagyoureit/nodejs-poolController #696: Additional Temp Display

camaro4life18, 2023-02-28:
## 💥 Proposal

### What feature you'd like to see

I would like the ability to show generic temperatures on the dash panel as well as pipe it into influxdb.

## Motivation

I wrote my own custom pool controller last year, just moved into a house with a pool. I noticed my pool temperature was not going up, even though it was calling for heat. I have a gas heater, and it was definitely working as I could see the flames. So I installed another temp sensor after the heater to capture the "Supply" temp. I quickly noticed the water was the same temp as the "Return" temp. This helped me diagnose the heater, which turned out to be a stuck bypass inside the heater. I cleaned it up and it started working and I was getting a +4-5 degree temperature difference.

In REM I can add multiple 1WIRE probes, and I have it feeding data as waterSensor2, but nowhere in DashPanel do I see a place to display an additional temp. I'm assuming it is designed to be used in the multiple pool display, which means the backend probably needs a better place to store the data as well as DashPanel having the ability to display it.

## Pitch

Users will be able to capture heated water temperatures and determine if their heater is working or not immediately, rather than waiting days to see their pool water is not getting heated. Written generically, like adding it similar to a circuit, users can add any number of temps for whatever they want.

johnny2678, 2023-02-28:
I've had a similar setup for a while now using ds18b20 sensors over 1wire.  I have grafana dashboards showing all of the extra temp data, so haven't really missed it not being on dashPanel, but wouldn't complain if it could be added.

for influx, I use an REM feed to send to MQTT and then have a nodered flow that subscribes the temp topic and pipes it into influx.  Clunky, but I've been manually pumping pool temp data to influx since before njsPC had influx support, so I'm used to workarounds  😂

rstrouse, 2023-02-28:
Actually, njsPC should send that waterSensor2 reading directly to influx without needing to go through a broker regardless of whether the temp sensor is attached to a body.  As an aside though, this temperature sensor is typically for dual body systems where the equipment separation exists.

Typically the best place for a water sensor probe is before the heater and often before the filter assembly.  This is the most representative temperature of the water before external factors have a chance to tweak it.

As for the display, we have been kicking around the ability to add panels dynamically to the dashboard.  I'll have a look this weekend as the groundwork for such an addition has already been laid.

Crewski, 2023-03-07:
I did the same thing as you described....one sensor before the heater and one sensor after the heater.  Mine are ds18b20 sensors.  The after-heater sensor is waterSensor2.  I use Home Assistant for my control/dashboard so I'm able to easily see and track it there.  It helped me find out I was having an issue with my pump not working probably (wasn't moving fast enough for the flow switch on the heater).