tagyoureit/nodejs-poolController #716: [BUG] Can't run v8.0.0 or 8.0.1 docker image

Botched1, 2023-04-12:
### nodejs-poolController Version/commit

8.0.1

### Are you using Docker?

- [X] Yes.

### What steps will reproduce the bug?

Update 7.6.1 docker image with 8.0.1, run it.

### What happens?

Docker comtainer continually stops/restarts.

### What should have happened?

Docker container should run.

### Additional information

I have been running 7.6.1 for a long time with no issues on a rpi with USB RS485 dongle.

Tried 8.0.1 today, and I can not get the docker image to run at all. All I see in the docker logs is:

```
Init state for Pool Controller
[4/12/2023, 1:40:39 PM] info: The current git branch output is 
[4/12/2023, 1:40:39 PM] info: The current git commit output is 35be048
[4/12/2023, 1:40:39 PM] info: Starting up SSDP server
Init state for Pool Controller
[4/12/2023, 1:40:43 PM] info: The current git branch output is 
[4/12/2023, 1:40:43 PM] info: The current git commit output is 35be048
[4/12/2023, 1:40:43 PM] info: Starting up SSDP server
Init state for Pool Controller
[4/12/2023, 1:40:47 PM] info: The current git branch output is 
[4/12/2023, 1:40:47 PM] info: The current git commit output is 35be048
[4/12/2023, 1:40:47 PM] info: Starting up SSDP server
Init state for Pool Controller
[4/12/2023, 1:40:51 PM] info: The current git branch output is 
[4/12/2023, 1:40:51 PM] info: The current git commit output is 35be048
[4/12/2023, 1:40:51 PM] info: Starting up SSDP server
Init state for Pool Controller
[4/12/2023, 1:40:55 PM] info: The current git branch output is 
[4/12/2023, 1:40:55 PM] info: The current git commit output is 35be048
[4/12/2023, 1:40:55 PM] info: Starting up SSDP server
```

And the container continually restarts. Not sure what other logs/info to look at to see what is happening?

Going back to the 7.6.1 image worked fine, so it isn't a huge issue or anything for me.

rstrouse, 2023-04-12:
You appear to not have the proper version of nodejs in your container.  It needs to be rebuilt with node 16+ for v8+ of nodejs-PoolController.

Botched1, 2023-04-13:
Very well could be... But shouldn't that be a dependency this is already defined in the docker container? I did not build a custom docker container, rather just used the docker hub one.

I guess I could console into the v8 container and try to see what node js version it is using.

Botched1, 2023-04-13:
Removed the containers, removed all images, and did this - same issue. Container dies.
```
jason@pool:~ $ docker-compose pull && docker-compose up -d
Pulling poolcontroller           ... done
Pulling poolcontroller-dashpanel ... done
Creating poolcontroller ... done
Creating poolcontroller-dashpanel ... done
```

Log:
```
Init state for Pool Controller
[4/13/2023, 1:40:15 PM] info: The current git branch output is master
[4/13/2023, 1:40:15 PM] info: The current git commit output is 5aa1455
[4/13/2023, 1:40:16 PM] info: Starting up SSDP server
Init state for Pool Controller
[4/13/2023, 1:40:20 PM] info: The current git branch output is master
[4/13/2023, 1:40:20 PM] info: The current git commit output is 5aa1455
[4/13/2023, 1:40:20 PM] info: Starting up SSDP server
Init state for Pool Controller
[4/13/2023, 1:40:24 PM] info: The current git branch output is master
[4/13/2023, 1:40:24 PM] info: The current git commit output is 5aa1455
[4/13/2023, 1:40:24 PM] info: Starting up SSDP server
```

rstrouse, 2023-04-13:
> Very well could be... But shouldn't that be a dependency that is already defined in the docker container? I did not build a custom docker container, rather just used the docker hub one.

Very well could be but my limited knowledge of Dockerfile syntax has me puzzled.  I believe if you look at the the Dockerfile it has the following lines in it `FROM node:lts-alpine3.12`

I believe this determines which node version is included but for the world of me I cannot figure out which version is in this and how to ensure node v16+ is included.  Is it as simple as changing the lines to `FROM node:lts-alpin3.18` to get it to use node 18?

Botched1, 2023-04-14:
Don't know. The docker hub container image the wiki points to is "msmi/nodejs-poolcontroller".

That one does not seem to use the docker file that is on taguyourit's github page. But I'm quickly getting out of my area of expertise!

rstrouse, 2023-04-14:
No errors in the logs it just simply looks like it is restarted over and over again.

Botched1, 2023-04-14:
This is all I see in the docker container logs when I built it with 16-alpine:

```
jason@pool:~ $ docker logs 55
Init state for Pool Controller
[4/14/2023, 8:51:45 AM] info: The current git branch output is master
[4/14/2023, 8:51:45 AM] info: The current git commit output is 5aa14551c3505d009e14cf36551710a2f67b0641
[4/14/2023, 8:51:45 AM] info: Starting up SSDP server
Init state for Pool Controller
[4/14/2023, 8:51:49 AM] info: The current git branch output is master
[4/14/2023, 8:51:49 AM] info: The current git commit output is 5aa14551c3505d009e14cf36551710a2f67b0641
[4/14/2023, 8:51:49 AM] info: Starting up SSDP server
Init state for Pool Controller
[4/14/2023, 8:51:54 AM] info: The current git branch output is master
[4/14/2023, 8:51:54 AM] info: The current git commit output is 5aa14551c3505d009e14cf36551710a2f67b0641
[4/14/2023, 8:51:54 AM] info: Starting up SSDP server
```

johnny2678, 2023-04-14:
if the question is, which version of node is created by the Dockerfile in the root of the repo, it's v16
```
# docker exec -it testpool node -v
v16.13.1
```

To reproduce:

1. `docker build -t this/njspc .`
2. `docker run -d -n testnjspc this/njspc`
3. `docker exec -it testpool node -v`

I also tried this with the pre-built image in the wiki and it returns v16 also:
1. `docker run -d --name testpool msmi/nodejs-poolcontroller`
2. `docker exec -it testpool node -v`

The `FROM` line in the dockerfile refers to the base image.  Something that already exists on the docker hub that subsequent dockerfile lines can be run on.

Botched1, 2023-04-14:
Thanks for confirming. 

OK, then I'm stumped. I have no idea why the docker containers that were built on v8 and later won't run for me for more than a few seconds before stopping/restarting.

Not a big problem, I can just stick with 7.6.1. My whole goal in reporting it was really in case there is an actual bug/something that will affect others.

johnny2678, 2023-04-14:
I don't run njspc from docker, but I spun up an image on my production pi out at the pad and it seems to be working - or at least not getting stuck on the log loop you posted above.  Output below:

```
pi@PiPoolv2:~ $ docker -v
Docker version 18.09.1, build 4c52b90

pi@PiPoolv2:~ $ docker run -d --name testpool msmi/nodejs-poolcontroller
Unable to find image 'msmi/nodejs-poolcontroller:latest' locally
latest: Pulling from msmi/nodejs-poolcontroller
80ca8924263d: Pull complete 
04d9ef8631da: Pull complete 
3f3387defed0: Pull complete 
d8cda29d9826: Pull complete 
89d10ed30b1a: Pull complete 
4f4fb700ef54: Pull complete 
2cae267837ae: Pull complete 
Digest: sha256:20c677dd873cb458a99d8d90165a9bc6ffcddbd15301d12ef435c28b8fb863e7
Status: Downloaded newer image for msmi/nodejs-poolcontroller:latest
a70c2787ec8e36689370f0db39fd9c0455a2acdf6dd63f265e1db7e76f5bc592
pi@PiPoolv2:~ $ 
pi@PiPoolv2:~ $ 
pi@PiPoolv2:~ $ docker logs testpool
Init state for Pool Controller
[4/14/2023, 5:41:55 PM] info: The current git branch output is master
[4/14/2023, 5:41:55 PM] info: The current git commit output is 5aa1455
[4/14/2023, 5:41:55 PM] info: Starting up SSDP server
[4/14/2023, 5:41:55 PM] info: Server is now listening on 0.0.0.0:4200 - 172.17.0.2:4200
[4/14/2023, 5:41:55 PM] error: Error opening port 0: Error: No such file or directory, cannot open /dev/ttyUSB0. Retry in 10 seconds
[4/14/2023, 5:41:55 PM] info: Checking njsPC versions...
[4/14/2023, 5:41:55 PM] info: Starting Pool System undefined
[4/14/2023, 5:41:55 PM] info: Listening for any installed OCPs
[4/14/2023, 5:41:55 PM] info: Auto-backup initialized Last Backup: 1970-01-01T00:00:00.000+0000
RESETTING DATA -- Controller type changed from undefined to nixie
[4/14/2023, 5:42:02 PM] info: Initializing Nixie Control Panel for Nixie Single Body
[4/14/2023, 5:42:02 PM] info: Initializing Intake/Return valves
[4/14/2023, 5:42:02 PM] info: Initializing Nixie Controller
[4/14/2023, 5:42:02 PM] info: Initializing Nixie body Pool
[4/14/2023, 5:42:02 PM] info: Initializing Filter Filter
[4/14/2023, 5:42:02 PM] info: Initializing Nixie circuit Pool
[4/14/2023, 5:42:02 PM] info: Nixie Controller Initialized
[4/14/2023, 5:42:02 PM] info: Nixie Single Body control board initialized
pi@PiPoolv2:~ $ docker logs testpool -f
Init state for Pool Controller
[4/14/2023, 5:41:55 PM] info: The current git branch output is master
[4/14/2023, 5:41:55 PM] info: The current git commit output is 5aa1455
[4/14/2023, 5:41:55 PM] info: Starting up SSDP server
[4/14/2023, 5:41:55 PM] info: Server is now listening on 0.0.0.0:4200 - 172.17.0.2:4200
[4/14/2023, 5:41:55 PM] error: Error opening port 0: Error: No such file or directory, cannot open /dev/ttyUSB0. Retry in 10 seconds
[4/14/2023, 5:41:55 PM] info: Checking njsPC versions...
[4/14/2023, 5:41:55 PM] info: Starting Pool System undefined
[4/14/2023, 5:41:55 PM] info: Listening for any installed OCPs
[4/14/2023, 5:41:55 PM] info: Auto-backup initialized Last Backup: 1970-01-01T00:00:00.000+0000
RESETTING DATA -- Controller type changed from undefined to nixie
[4/14/2023, 5:42:02 PM] info: Initializing Nixie Control Panel for Nixie Single Body
[4/14/2023, 5:42:02 PM] info: Initializing Intake/Return valves
[4/14/2023, 5:42:02 PM] info: Initializing Nixie Controller
[4/14/2023, 5:42:02 PM] info: Initializing Nixie body Pool
[4/14/2023, 5:42:02 PM] info: Initializing Filter Filter
[4/14/2023, 5:42:02 PM] info: Initializing Nixie circuit Pool
[4/14/2023, 5:42:02 PM] info: Nixie Controller Initialized
[4/14/2023, 5:42:02 PM] info: Nixie Single Body control board initialized
[4/14/2023, 5:42:05 PM] warn: Inactivity timeout for 0 serial port /dev/ttyUSB0 after 10 seconds
[4/14/2023, 5:42:05 PM] error: Error opening port 0: Error: No such file or directory, cannot open /dev/ttyUSB0. Retry in 10 seconds
[4/14/2023, 5:42:07 PM] warn: dt:Fri Apr 14 2023 17:42:07 GMT+0000 (Coordinated Universal Time) lat:undefined lon:undefined Not enough information to calculate Heliotrope.  See https://github.com/tagyoureit/nodejs-poolController/issues/245
[4/14/2023, 5:42:07 PM] warn: dt:Fri Apr 14 2023 17:42:07 GMT+0000 (Coordinated Universal Time) lat:undefined lon:undefined Not enough information to calculate Heliotrope.  See https://github.com/tagyoureit/nodejs-poolController/issues/245
[4/14/2023, 5:42:08 PM] info: Checking njsPC versions...
[4/14/2023, 5:42:15 PM] warn: Inactivity timeout for 0 serial port /dev/ttyUSB0 after 10 seconds
[4/14/2023, 5:42:15 PM] error: Error opening port 0: Error: No such file or directory, cannot open /dev/ttyUSB0. Retry in 10 seconds
```

The errors are just because my rs485 runs on a different port than what njsPC uses by default.  Not sure how to help but maybe try and spin one up with a different name than your current container and see if you have similar output as above?  If that works, you can copy over config files or restore from backup.

treyrich, 2023-05-02:
@johnny2678 you didn't grant device access to the docker container, so it's unable to connect to the /dev/ttyUSB0 device. If you were to run this command `docker run --rm -it -u node:dialout -p 4200:4200 --device /dev/ttyUSB0 msmi/nodejs-poolcontroller` you would see the issue reproduced.

johnny2678, 2023-05-02:
Yep @treyrich - I wasn't trying to actually run njspC.  Just confirming the version of node used in the docker container.  My production setup runs on a Mega-BAS/pi

emes, 2023-05-29:
> I think you are right. Do we know who "msmi" is that maintains the docker containers linked to in the wiki? Is it @emes ?
> 
> EDIT: Might be more to it than that... I tried building a custom container with 16-alpine, and lts-alpine (which should be 18?) and got the same results with both builds.

Sorry to be late to this.  I've done a bunch of testing and remain stumped.  As has been said, works pre v8.  I've built with above and ubuntu.  Same results on all.

It works fine with no mounted volumes (thereby giving a default config).

I cannot make it work with /app/config.sys mounted.

I'm getting this...

'''Error reading configuration information.  Aborting startup: SyntaxError: Unexpected token  in JSON at position 0

/app/config/Config.ts:39
            this._cfg = fs.existsSync(this.cfgPath) ? JSON.parse(fs.readFileSync(this.cfgPath, "utf8")) : {};
                                                           ^
SyntaxError: Unexpected token  in JSON at position 0
    at JSON.parse (<anonymous>)
    at new Config (/app/config/Config.ts:39:60)
    at Object.<anonymous> (/app/config/Config.ts:195:31)
    at Module._compile (node:internal/modules/cjs/loader:1101:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1153:10)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:1005:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/app/logger/Logger.ts:24:1)
Error reading configuration information.  Aborting startup: SyntaxError: Unexpected token  in JSON at position 0'''

I've gone through the v8 commits and don't see anything obvious.

If I mount the config.json as another file name in /app it's perfectly readable.

It feels like maybe an async issue, but don't really even have a theory.

tagyoureit, 2023-05-29:
'Unexpected token' means it can't read a JSON file properly. If you created an empty file you will get that.  Try to create a file with '{}' and see if that resolves the error.

emes, 2023-05-29:
Had already done that. It was not an empty file but a valid config.  

Works fine with .trim() added.

Anyone else feel free to test by pulling msmi/nodejs-poolcontroller:latest

tagyoureit, 2023-05-31:
I accepted the PR and pushed v8.0.2 release.

treyrich, 2023-05-31:
> Had already done that. It was not an empty file but a valid config.
> 
> Works fine with .trim() added.
> 
> Anyone else feel free to test by pulling msmi/nodejs-poolcontroller:latest

@emes latest has the same issue for me. It actually doesn't work for me with no mounted volumes either though. The following command fails for me on latest:
`docker run --rm -u node:dialout -p 4200:4200 --device /dev/ttyUSB0 msmi/nodejs-poolcontroller:latest bash`
with the following logs:

```
Init state for Pool Controller
[5/31/2023, 9:12:39 PM] info: The current git branch output is master
[5/31/2023, 9:12:39 PM] info: The current git commit output is 9eb0f5d
[5/31/2023, 9:12:40 PM] info: Starting up SSDP server
```

emes, 2023-06-04:
@treyrich hmm.  It's working for me with or without mounted volumes/config now.

```docker run --rm -u node:dialout -p 4200:4200 --device /dev/ttyUSB0 msmi/nodejs-poolcontroller:latest```
 gets me:

```Init state for Pool Controller
[6/4/2023, 3:18:35 AM] info: The current git branch output is master
[6/4/2023, 3:18:35 AM] info: The current git commit output is 59f6b20
[6/4/2023, 3:18:35 AM] info: Starting up SSDP server
[6/4/2023, 3:18:35 AM] info: Server is now listening on 0.0.0.0:4200 - 172.17.0.2:4200
[6/4/2023, 3:18:35 AM] info: Serial port: /dev/ttyUSB0 request to open successful 9600b 8-none-1
[6/4/2023, 3:18:35 AM] info: Checking njsPC versions...
```

Can you post output of:

```% uname -a``` and ```% docker version```

treyrich, 2023-06-04:
```
uname -a
Linux pihole 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
```

```
docker version
Client: Docker Engine - Community
 Version:           24.0.2
 API version:       1.43
 Go version:        go1.20.4
 Git commit:        cb74dfc
 Built:             Thu May 25 21:50:59 2023
 OS/Arch:           linux/arm64
 Context:           default

Server: Docker Engine - Community
 Engine:
  Version:          24.0.2
  API version:      1.43 (minimum version 1.12)
  Go version:       go1.20.4
  Git commit:       659604f
  Built:            Thu May 25 21:50:59 2023
  OS/Arch:          linux/arm64
  Experimental:     false
 containerd:
  Version:          1.6.21
  GitCommit:        3dce8eb055cbb6872793272b4f20ed16117344f8
 runc:
  Version:          1.1.7
  GitCommit:        v1.1.7-0-g860f061
 docker-init:
  Version:          0.19.0
  GitCommit:        de40ad0
```

treyrich, 2023-06-05:
Would also like to confirm that I'm running successfully on version 7.6.1 on docker with no problems at all. So definitely some difference between 7.6.1 and 8.x.x images. I also tried 8.0.2 today just in case something had changed, but same issue persists.

emes, 2023-06-07:
I upgraded to be on same versions as you and am still good.

You're definitely right that something changed between v7 and v8 of the app, but it wasn't how the docker images are built.  I've been going through the commits to try and find something.  

Have you tried to build the docker image locally?

treyrich, 2023-06-08:
Yes, I have, building the image doesn't change anything. From what I can tell the app is crashing with a segmentation fault, I may have some time tomorrow to try to dig deeper into the crash. I can confirm the following though:
- It runs natively as a node app without issue (v. 8.x.x)
- 7.6.1 runs without issue in a docker container
- Building the docker image locally makes no difference
- Using default config or config via volume doesn't make a difference

Like I said, tomorrow I'll see if I can dig a little deeper into this, so far I've only had a minute here and there to look at it.

treyrich, 2023-06-09:
Ok. I've found the issue, it seems it's an upstream issue here https://github.com/serialport/node-serialport/issues/2438

As a fix I've submitted PR #785 which I've tested and is working for me. Not sure if there's anyone else who cares to test this, but I believe that this should close this issue out once merged.

tagyoureit, 2023-06-10:
Released v8.0.3 with the serialport updates and other bug changes.

PR #785 (2023-06-10): Update serialport
Resolves #716 which was caused by upstream issue https://github.com/serialport/node-serialport/issues/2438