tagyoureit/nodejs-poolController #721: [BUG]

camaro4life18, 2023-04-17:
### nodejs-poolController Version/commit

8.0.1

### nodejs-poolController-dashPanel Version/commit

8.0.0

### relayEquipmentManager Version/commit

Latest as of 4/17/2023

### Node Version

v16.19.1

### Platform

Linux poolcontroller 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

### RS485 Adapter

Industrial USB to RS485

### Are you using Docker?

- [ ] Yes.

### OCP

Nixie

### Pump(s)

Hayward Super Pump VS 700

### Chlorinator(s)

Ezo Pump

### What steps will reproduce the bug?

ORP Tank is at empty. 
1) Click on Chemistry
2) Click on ORP Tank
3) Set to 4 Gallons (Max is 5)
4) Click Save
5) Wait a few seconds
6) Click on Dose Chlorine
7) Dose any amount

### What happens?

An exception is thrown saying ORP tank is empty. Sometimes just waiting for a time period the ORP tank just shows empty. It does show the 4 gallons after you save the new level, but eventually goes back to empty.

`ApiError: The ORP tank is empty
    at NixieChemicalORP.manualDoseVolumeAsync (/appl/nodejs-poolController/controller/nixie/chemistry/ChemController.ts:2014:61)
    at NixieChemController.manualDoseAsync (/appl/nodejs-poolController/controller/nixie/chemistry/ChemController.ts:489:32)
    at NixieChemControllerCollection.manualDoseAsync (/appl/nodejs-poolController/controller/nixie/chemistry/ChemController.ts:29:21)
    at NixieChemControllerCommands.manualDoseAsync (/appl/nodejs-poolController/controller/boards/SystemBoard.ts:4918:39)
    at /appl/nodejs-poolController/web/services/state/State.ts:100:61
    at Layer.handle [as handle_request] (/appl/nodejs-poolController/node_modules/express/lib/router/layer.js:95:5)
    at next (/appl/nodejs-poolController/node_modules/express/lib/router/route.js:144:13)
    at Route.dispatch (/appl/nodejs-poolController/node_modules/express/lib/router/route.js:114:3)
    at Layer.handle [as handle_request] (/appl/nodejs-poolController/node_modules/express/lib/router/layer.js:95:5)
    at /appl/nodejs-poolController/node_modules/express/lib/router/index.js:284:15
    at Function.process_params (/appl/nodejs-poolController/node_modules/express/lib/router/index.js:346:12)
    at next (/appl/nodejs-poolController/node_modules/express/lib/router/index.js:280:10)
    at /appl/nodejs-poolController/web/Server.ts:749:25
    at Layer.handle [as handle_request] (/appl/nodejs-poolController/node_modules/express/lib/router/layer.js:95:5)
    at trim_prefix (/appl/nodejs-poolController/node_modules/express/lib/router/index.js:328:13)
    at /appl/nodejs-poolController/node_modules/express/lib/router/index.js:286:9
`

### What should have happened?

The tank should stay at entered level until dosing starts.

### Additional information

[replay (6).zip](https://github.com/tagyoureit/nodejs-poolController/files/11253412/replay.6.zip)

camaro4life18, 2023-04-17:
I found a work around by going through REM and adding volume there.

rstrouse, 2023-04-17:
Are you using the Atlas EZO-PRS?

camaro4life18, 2023-04-17:
Yes

rstrouse, 2023-04-18:
Ahah.  That is why I was baffled.  I will double check the code to make sure we are forwarding that to REM when it involves a PRS.

camaro4life18, 2023-04-18:
I never would have guessed that would interfere. I'm working on keeping Atlas in business.
[image]

rstrouse, 2023-04-18:
I mistyped I had the PRS in my head.  I am referring to the EZO-PMP.

camaro4life18, 2024-03-06:
Closing this as I'm no longer seeing the issue.

closed 2024-03-06 by camaro4life18