tagyoureit/nodejs-poolController #725: [BUG] v8.0.1 - reverts to Nixie controller when a log capture started

antamy, 2023-04-21:
### nodejs-poolController Version/commit

8.0.1 / master 67b0641

### nodejs-poolController-dashPanel Version/commit

8.0.0 / master cb05840

### Platform

raspberry pi / debian

### Are you using Docker?

- [ ] Yes.

### OCP

Intellitouch i5+3

### Chlorinator(s)

IC40

### What steps will reproduce the bug?

I upgraded to v8.0.1 using git pull / npm i / npm start

I run it from the cli using npm run start:cached

dashPanel connects and things look fine, except the Chemistry section is empty.  I try to initiate a packet capture with the reload option set.  The reload starts and the system reverts to a Nixie system!

### What happens?

Here is the output 

$ npm run start:cached

> nodejs-poolcontroller@8.0.1 start:cached
> node dist/app.js

Init state for Pool Controller
sendRS485PortStats set to false
RESETTING DATA -- Controller type changed from nixie to unknown
RESETTING DATA -- Controller type changed from unknown to intellitouch
Pentair IntelliTouch System Detected!
[21/04/2023, 12:16:16] warn: dt:Fri Apr 21 2023 12:16:01 GMT-0400 (Eastern Daylight Time) lat:undefined lon:undefined Not enough information to calculate Heliotrope.  See https://github.com/tagyoureit/nodejs-poolController/issues/245
[21/04/2023, 12:16:16] warn: dt:Fri Apr 21 2023 12:16:01 GMT-0400 (Eastern Daylight Time) lat:undefined lon:undefined Not enough information to calculate Heliotrope.  See https://github.com/tagyoureit/nodejs-poolController/issues/245
RS485 Stats:{ "bytesReceived": 1009 "success": 36, "failed": 1, "bytesSent": 385, "collisions": 0, "failureRate": 2.70% }
RS485 Stats:{ "bytesReceived": 1968 "success": 74, "failed": 2, "bytesSent": 817, "collisions": 0, "failureRate": 2.63% }
RS485 Stats:{ "bytesReceived": 2666 "success": 100, "failed": 3, "bytesSent": 1141, "collisions": 0, "failureRate": 2.91% }
RS485 Stats:{ "bytesReceived": 2917 "success": 109, "failed": 4, "bytesSent": 1249, "collisions": 0, "failureRate": 3.54% }
RS485 Stats:{ "bytesReceived": 3316 "success": 125, "failed": 5, "bytesSent": 1441, "collisions": 0, "failureRate": 3.85% }
RS485 Stats:{ "bytesReceived": 3409 "success": 131, "failed": 6, "bytesSent": 1453, "collisions": 1, "failureRate": 4.38% }
[21/04/2023, 12:17:01] warn: dt:Fri Apr 21 2023 12:16:01 GMT-0400 (Eastern Daylight Time) lat:undefined lon:undefined Not enough information to calculate Heliotrope.  See https://github.com/tagyoureit/nodejs-poolController/issues/245
[21/04/2023, 12:17:01] warn: dt:Fri Apr 21 2023 12:16:01 GMT-0400 (Eastern Daylight Time) lat:undefined lon:undefined Not enough information to calculate Heliotrope.  See https://github.com/tagyoureit/nodejs-poolController/issues/245

+++ at this point I have just initiated the packet capture +++

RESETTING DATA -- Controller type changed from intellitouch to unknown
[21/04/2023, 12:17:22] info: New socket client connected F7_svDDxP25yttkvAAAE -- 10.10.20.50
[21/04/2023, 12:17:22] info: [12:17:22] 10.10.20.50 GET /state/all?null {}
sendRS485PortStats set to false
RESETTING DATA -- Controller type changed from unknown to nixie
[21/04/2023, 12:17:26] info: Initializing Nixie Control Panel for Nixie Single Body
[21/04/2023, 12:17:26] info: Initializing Intake/Return valves
[21/04/2023, 12:17:26] info: Initializing Nixie Controller
[21/04/2023, 12:17:26] info: Initializing Nixie body Pool
[21/04/2023, 12:17:26] info: Initializing Filter Filter
[21/04/2023, 12:17:26] info: Nixie Controller Initialized
[21/04/2023, 12:17:26] info: Nixie Single Body control board initialized
[21/04/2023, 12:17:26] silly: MQTT send:
topic: nixie-singlebody/state/temps/air
message: {"temp":null}
opts:{"retain":true,"qos":0}
[21/04/2023, 12:17:26] silly: MQTT send:
topic: nixie-singlebody/state/temps/units
message: {"units":{"val":0,"name":"F","desc":"Fahrenheit"}}
opts:{"retain":true,"qos":0}
[21/04/2023, 12:17:26] silly: MQTT send:
topic: nixie-singlebody/state/temps/bodies/1/pool
message: {"id":1,"isOn":"off"}
opts:{"retain":true,"qos":0}
[21/04/2023, 12:17:26] silly: MQTT send:
topic: nixie-singlebody/state/temps/bodies/1/pool/heatMode
message: {"heatMode":{"val":0,"name":"off","desc":"Off"}}
opts:{"retain":true,"qos":0}
[21/04/2023, 12:17:26] silly: MQTT send:

### What should have happened?

Should have reloaded and detected the correct panel!

If I run od -x < /dev/ttyUSB1, I see data flowing

rstrouse, 2023-04-21:
Can I get an actual replay capture not just an output of the logs.  The issue appears to be failure data coming from the status output of your IntelliTouch but I cannot be sure.

antamy, 2023-04-21:
Here you go.  I started njspc from the command line, it loaded correctly (albeit not detecting the chlorinator).  Once it had loaded I initiated the capture.  dashPanel showed Initializing, then Intellitouch i5+3 changed to Nixie Single Body and status became Ready.  I stopped the capture.

[replay(4).zip](https://github.com/tagyoureit/nodejs-poolController/files/11297923/replay.4.zip)

rstrouse, 2023-04-21:
Your port is not receiving any data.  Did you go into the config portion of dashPanel then select the Controller->Comms.  Open the primary port and check the port status.

antamy, 2023-04-21:
Yes, that is all working fine.  I can control equipment through the dashPanel.  But as soon as I try to capture a log, it stops working!  I can capture a replay without the config reload and that works fine.  With the config reload, it causes this error.

I tried System > Reload Config and that works fine.  The attached file is from that.

[replay(6).zip](https://github.com/tagyoureit/nodejs-poolController/files/11298376/replay.6.zip)

[image]

tagyoureit, 2024-06-10:
Closing due to lack of activity.  If this is still an issue, please reopen.

closed 2024-06-10 by tagyoureit