tagyoureit/nodejs-poolController #740: [BUG] Chemistry panel is empty, even though a chlorinator is present

antamy, 2023-04-25:
### nodejs-poolController Version/commit

8.0.1 / master 67b0641

### nodejs-poolController-dashPanel Version/commit

8.0.0 / master cb05840

### Platform

raspberry pi / debian

### Are you using Docker?

- [ ] Yes.

### Chlorinator(s)

IC40

### What steps will reproduce the bug?

Run njspc.  dashPanel dashboard has no information under Chemistry

### What happens?

I captured a replay by turning on Logging, going to System and chose Reload Config.  
[replay(1).zip](https://github.com/tagyoureit/nodejs-poolController/files/11324432/replay.1.zip)

### What should have happened?

Chlorinator info should be displayed.   If I run 7.6.1 on the same system (having shutdown 8.0.1), it displays chlorinator info on dashPanel 8.0

rstrouse, 2023-04-25:
For IntelliTouch models the system was not requesting the IntelliChlor status on reload.  Pull njsPC and this problem is fixed.

antamy, 2023-04-25:
Fixed, thanks

closed 2023-04-25 by antamy