tagyoureit/nodejs-poolController #746: Chemistry delay on Circuit start or PSI instead of Pump

camaro4life18, 2023-04-28:
## 💥 Proposal

### What feature you'd like to see

Currently there is a feature to delay ORP or pH from dosing for a period of time after the pump starts. I would like to be able to select a circuit, or just delay off the PSI, that the pump uses to delay the start instead of just the overall pump start.

## Motivation

I like to run my pump at a low RPM all the time to keep the temperature sensors more accurate. However the ORP and pH probes need more water flow to be accurate, to which I have circuits setup to run at higher RPMs to actually filter the pool and accurately gauge the Sand filter life. 

## Pitch

This feature will allow users to keep the pump running all the time at lower RPMs, but still be able to accurately control the chemistry by  delaying the dosing when a circuit with a higher RPM starts to allow the probes to accurately gauge the water. This feature could also be done by delaying off the selected pressure minimum. If the pressure is 10psi, then you could set the delay to 10minutes after the pressure has been at least 10 psi.

rstrouse, 2023-04-29:
Do you have a flow cell and how did you connect the diversion?  The pH and ORP should require very little flow to be accurate.  That being said, I think I have your configuration as this is tied to the pool circuit.  The primary reason is that there is more than one trigger for these delays depending on whether there is a cleaner, freeze protection, ...etc.

camaro4life18, 2023-05-24:
I had my probes inline with the main water flow before my heater. I decided to stop being cheap and just bought a flow cell. I'm having more luck with that and not as much fluctuation with different pump speeds.

closed 2023-05-24 by camaro4life18