tagyoureit/nodejs-poolController #752: [BUG]  restoring backup appears to not complete

samsiam, 2023-05-03:
### nodejs-poolController Version/commit

8.0.1

### nodejs-poolController-dashPanel Version/commit

8.0.0

### RS485 Adapter

Industrial

### Are you using Docker?

- [ ] Yes.

### OCP

Easytouch 2p

### Pump(s)

Intelliflow VS, ExtremePowerUS

### What steps will reproduce the bug?

Doing some troubleshooting on my Easytouch.  Backed my board up with poolcontroller and tried to restore to a different one.  Restore took a long time and the restoring box never went away. I did see date up top incrementing like it was bytes loaded.  That eventually stopped (restore window never went away).  After about 20 minutes, I closed the browser and opened it back up.  got a broken screen and now restarting NPM it errors out.

### What happens?

[image]

[image]

### What should have happened?

hoped to get a restore to confirm if my board has an issue

### Additional information

Also trying to see if I can get any info from VSP - lightening in the area took out comms chip on Easytouch and Screenlogic.  Fixed board (we think), replaced Screenlogic, and even keypad of Intelliflo.  Intelliflo still not communicating.  used different cable as well.  rstrouse said he can help me here with Message Manager (i think).

samsiam, 2023-05-03:
meant to include this - after that, restarting controller fails

[image]

samsiam, 2023-05-03:
will do.  it was correct on old board.

samsiam, 2023-05-03:
[2023-05-02_15-34-34.zip](https://github.com/tagyoureit/nodejs-poolController/files/11391233/2023-05-02_15-34-34.zip)

rstrouse, 2023-05-03:
What is the date/time set for the computer that is running the njsPC server?

samsiam, 2023-05-03:
It is current.  
[image]

rstrouse, 2023-05-03:
That is so bizzare. I just restored that backup to a blank mock panel and it restored fine.  Is the RS485 working for the target panel?  Did it load on its own as soon as it started?

[image]

samsiam, 2023-05-03:
hey, that is odd.  i installed the new board and it showed up blank.  then tried the restore as noted.  after that it all went south.  not sure what you mean by "load on its own"  i dont have a startup script set yet.  i did try a host reboot as well.  should I try killing the config files and starting over?

rstrouse, 2023-05-03:
If the RS485 port is connected and communicating it should read the current configuration from the panel.  If communication is not working then that would explain why the restore failed.

First hook up the panel and let the initialization finish.  It should detect and EasyTouch2 8 and begin interrogating any information that is currently on the panel.  If you are not getting any broadcast messages from the panel then double check the  RS485 connection.

This software will give you all current communication messages from the panel and if you have dashPanel connected and installed it is already on your system.

https://github.com/rstrouse/nodejs-poolController-dashPanel/wiki/Message-Manager

rstrouse, 2023-05-03:
Oh and kill the files in the data directory to see what is currently configured on the panel.

samsiam, 2023-05-03:
I renamed the files and it found the board.  It is showing as it was when I got it from Tom.  It read it in.  Restoring mine seems to be an issue.  I picked an earlier backup than the last one to see if it had an issue but same thing - it is loading now (date at top is incrementing).  Loading Message Manager in different window shows nothing in messages as restore is going

samsiam, 2023-05-03:
I may have missed this the first time.  I just noticed it gives an error finally.  and checking console, same crash.  Going to remove my suspect VSP from the bus and start over
[image]

samsiam, 2023-05-03:
well this is new.  almost as soon as i clicked restore this popped up.  time still shows incrementing thought
[image]

samsiam, 2023-05-03:
also, is this OK?  i have been getting this from day 1

[image]

rstrouse, 2023-05-03:
Go into the general configuration and set your latitude and longitude.  This will allow you to run schedules based upon sunrise and sunset.

samsiam, 2023-05-03:
just realized message manager has a start/stop.  started it and i am seeing info now. restore still going

rstrouse, 2023-05-03:
You will see constant traffic on the bus.  This is normal as EasyTouch will send action 2 status messages and communicate with your pump constantly.

samsiam, 2023-05-03:
thanks, yea i figured that would be the case.  just got lucky - the icon was not obvious

rstrouse, 2023-05-04:
MessageManager in many cases is the only way we can troubleshoot equipment we do not have.  When users give us a replay we can play that back to a mock install of the panel to watch what is going on with it.  Without it decoding of the protocol would be near impossible.

So now that you have the system running I would like you to click on the hamburger menu in dashPanel, then navigate to the logging tab and press `Capture Replay`.  Do not check the `Reload Configuration` checkbox at this point I want to see what EasyTouch is sending us on the action 5 messages.  Let it run for about a minute then press the `End Capture` button.  This should download a zip file to your browser.  Upload that file here.

samsiam, 2023-05-04:
is that what you are looking for?

rstrouse, 2023-05-04:
Yes and I am not seeing anything that reflects the strange date being set from your panel.  In every instance I am seeing May the fourth be with you.  The capture data looks good and the results look good in the configuration.

samsiam, 2023-05-04:
this capture is before i did a restore so it has whatever configuration on it that was sent with the board.  do you want me to do the capture at a different point?

rstrouse, 2023-05-04:
Yes I would like to see it when it has the strange 2733 date.

samsiam, 2023-05-04:
ok let me try that.  start capture then do restore.

samsiam, 2023-05-04:
[replay (1).zip](https://github.com/tagyoureit/nodejs-poolController/files/11401650/replay.1.zip)
so not sure if this gets what you are looking for.  i started capture, started restore.  took forever and never completed - got the connection error on web interface.  had to hit ctrl-c on console.  once i did this the panel came back on web interface.  ended capture and here it is.  very strange.  restore did not happen.  back to what i started with but at least its not the broken interface i showed before.

rstrouse, 2023-05-04:
I see the error message that we are not capturing.  I'll have a look.
```
[5/4/2023, 3:43:34 PM] info: [3:43:34 PM] 192.168.86.25 PUT /app/restore/validate {"filename":"2023-05-01_15-34-59.zip","options":{"version":"1.1","name":"pool-backup","backupDate":"05/01/2023 03:34:59pm","njsPC":true},"filePath":"C:\\Users\\sjs\\Documents\\GitHub\\nodejs-poolController\\backups\\2023-05-01_15-34-59.zip"}
[5/4/2023, 3:43:34 PM] error: Error extracting restore options from C:\Users\sjs\Documents\GitHub\nodejs-poolController\backups\2023-05-01_15-34-59.zip: Cannot read properties of undefined (reading 'length')
```

rstrouse, 2023-05-04:
Can I get a copy of the 2023-05-01_15-32-59.zip.  The one you posted earlier is different.

samsiam, 2023-05-04:
[2023-05-01_15-34-59.zip](https://github.com/tagyoureit/nodejs-poolController/files/11402287/2023-05-01_15-34-59.zip)
[2023-05-01_15-34-34.zip](https://github.com/tagyoureit/nodejs-poolController/files/11402288/2023-05-01_15-34-34.zip)

possible the -32 is something else?  i only have 2 from that day.
34?
[image]
i did try a different one just to be sure it was not the file

rstrouse, 2023-05-05:
I meant to type -34 so I got the file I needed.  Pull njsPC and dashPanel.  There was an error thrown early on that did not allow it to complete.

samsiam, 2023-05-05:
still get this error.  looking at the backup file it is set to 10.

[image]

 {
      "id": 10,
      "type": 65,
      "name": "Two Speed",
      "portId": 0,
      "circuits": [
        {
          "id": 1,
          "master": 0,
          "circuit": 3
        },
        {
          "id": 2,
          "master": 0,
          "circuit": 12
        },
        {
          "id": 3,
          "master": 0,
          "circuit": 6
        },
        {
          "id": 4,
          "master": 0,
          "circuit": 11
        }
      ],
      "isActive": true

rstrouse, 2023-05-05:
Yes that is because of the way EasyTouch handles two speed pumps.  It was shoehorned into the ET firmware at some point and requires a two step process where it is first disabled then re-enabled.  We probably should not display the message as an error when it is doing this though.

samsiam, 2023-05-05:
should i edit the backup file?  it won't restore - just stops immediately now after that error.  i believe i was messing around when i added that - not sure.

rstrouse, 2023-05-05:
Yes change it to id 9 in poolState.json and poolConfig.json and it will restore properly.

samsiam, 2023-05-06:
well it acted like it was uploading.  stopped here.  restarted panel and it is a mix of mine and what was there.  not complete though
[image]

samsiam, 2023-05-13:
finally got the right part for my pump so it is back online.  pulled Tom's board to retest mine and aux6/7 are bad.  i will put his back in and try the restore again.  i pulled the fixes - thanks

samsiam, 2023-05-15:
so i got a chance to swap boards back.  his board had a partial restore from past attemps but not complete.  i got what i believe is good backups now and so i went to restore.  when i do i see a console message immediately but after that nothing happens.  i have been waiting almost 10 minutes for it to show some kind of completion.  should i expect it to complete and backup menu to go away or show complete?
[image]

samsiam, 2023-05-15:
ok, swapped board and restore worked.  thanks for the help!

closed 2023-05-15 by samsiam