tagyoureit/nodejs-poolController #764: Enable/Disable Schedules

Crewski, 2023-05-19:
## 💥 Proposal

### What feature you'd like to see

The ability to enable/disable a schedule via the dashboard and http api.

## Motivation

Last night I went to super chlorinate my pool, so I set my SWG to super chlorinate.  I have a schedule that turns the pool off at night.  When the pool off time happened, it turned off my super chlorinate (I had the pump running on a named circuit to keep it going).  If I was able to enable/disable a schedule, I could disable the pool off when the super chlor is on.  I used Home Assistant and could do the scheduling in there, but I am of the mindset that the "local" device should do as much automation/control on its own and HA should just interact with it, so if my HA goes down things are still working like they should.  Side note, it would be even better if there was an option you could select that would allow the pool/spa to continue to run until a superChlor is completed.  I only have experience with Nixie so I don't know how that works with other controllers.

Right now I think the only way is to delete the schedule to "disable it".  I poked around the code and know isActive and disable are in the interface and have getters/setters.  

## Pitch

The ability to enable/disable a schedule would be helpful for several different reasons.  My personal use case is to allow the pool circuit to continue to run during a superchlor event.  I can also see it useful if my water was cloudy so I wanted to run the filter all night, or maybe not turn the pool light on when I'm out of town.

tagyoureit, 2023-05-19:
Nixie, and (I believe) Intellicenter, have the ability to disable schedules.  *Touch controllers do not.

[image]

Here is an example of disabling the schedule with the API:
```
{
    "method": "put",
    "url": "/config/schedule",
    "data": "{\"id\":-1,\"circuit\":6,\"scheduleType\":128,\"display\":0,\"heatSource\":2,\"heatSetpoint\":78,\"coolSetpoint\":100,\"disabled\":true,\"startTimeType\":0,\"startTime\":480,\"endTimeType\":0,\"endTime\":1020,\"scheduleDays\":127,\"startDate\":\"05/19/2023\"}"
}
```

tagyoureit, 2023-05-19:
Separate note... Super chlorinate on the Pentair controllers will override any schedules and keep the pump running.  That would be a separate issue if we are treating that differently than the OCP's.

subcritical71, 2023-05-23:
I was thinking schedules should also have the ability to enable/disable, I am running standalone nixie. I think the disabled button shown is for the heater as it only shows up on circuits that have heaters installed (for me).

Crewski, 2023-05-23:
Same for me.  The pool circuit has the disable function, the max speed (general) circuit does not.  

I can confirm that the functionality to disable via the API is working for all circuits.  I have included it in my Home Assistant integration.

[image]

subcritical71, 2023-05-24:
I have a dumb question I can't seem to find the answer to... how do you pull from a branch (ie. not master)?

Crewski, 2023-05-24:
Looks like the commit was pushed to the 8.0.0 branch and not the master.

tagyoureit, 2023-05-26:
That's what happens when I work on two different computers.  Now pushed to master.

closed 2023-06-06 by Crewski