tagyoureit/nodejs-poolController #775: after upgrade new version chlorinator quit display

omo22, 2023-06-05:
### nodejs-poolController Version/commit

8.01

### Node Version

8

### Are you using Docker?

- [ ] Yes.

### What steps will reproduce the bug?

upgraded from 6 or 7 not sure

### What happens?

since upgrade no longer able to see salt level nor chontrol chlorinator
before and after update  mqtt nerver worked to pump=(interiflow VF)
please see log i changed flow and watts as well gpm changed but nerver receiving mqtt raw or through homeassistant 

### What should have happened?

[replayIntIflow.zip](https://github.com/tagyoureit/nodejs-poolController/files/11647654/replayIntIflow.zip)

rstrouse, 2023-06-05:
Are you using MQTT to integrate with Home Assistant?  I cannot recommend strongly enough that you instead use the socket connector and the https://github.com/Crewski/njsPC-HA integration for HA.

The reason why your MQTT is not working is because the broker is returning that it is not authorized to connect.  So your username and password must not be correct.  This connection process is eating up all available cycles to during the connect.

Also, the chlorinator in your config is set to be controlled by an external application not Nixie so it will not even attempt t communicate with it. Do you have some other integration that is supposed to be controlling it?  If not you must delete and re-add but make sure you choose Nixie as the master.

omo22, 2023-06-05:
chloronator worked
i had to choose nixie and then picked T15 aqua
MQTT still puzzle !! may be need to add usr psw for mqtt to connect probably ?

omo22, 2023-06-05:
homeassistant integration worked well
there seem little bug showing PRIMING pump status all times.
THANKS a lot 
i will post other issue for this pump reg adding reset button case pump quit while i am on vacation !!

closed 2023-06-06 by omo22