tagyoureit/nodejs-poolController #786: [BUG] Docker image not running

blackax1, 2023-06-10:
### nodejs-poolController Version/commit

 Latest 8197e90

### Are you using Docker?

- [X] Yes.

### What steps will reproduce the bug?

` /usr/bin/docker run --name poolcontroller --group-add dialout --device /dev/ttyUSB0 -v /mnt/data/vm/docker/config/poolcontroller/config.json:/app/config.json -v /mnt/data/vm/docker/config/poolcontroller/data:/app/data -v /mnt/data/vm/docker/config/poolcontroller/logs:/app/logs -p 4200:4200 msmi/nodejs-poolcontroller`

### What happens?

```
Init state for Pool Controller
[6/10/2023, 12:28:13 AM] info: The current git branch output is master
[6/10/2023, 12:28:13 AM] info: The current git commit output is 8197e90
[6/10/2023, 12:28:13 AM] info: Starting up SSDP server

```
Then the process dies and no container is left running

### What should have happened?

The container should have started or at least made some logs with the error message.

### Additional information

If I removed the permissions for ttyUSB0 I can get error that is past after the "Starting up SSDP server" message but as soon as I make ttyUSB0 available the container exits

I get the same error when I use the default docker run command.

`docker run --rm -it -u node:dialout -p 4200:4200 --device /dev/ttyUSB0 msmi/nodejs-poolcontroller
Init state for Pool Controller
[6/10/2023, 12:31:40 AM] info: The current git branch output is master
[6/10/2023, 12:31:40 AM] info: The current git commit output is 8197e90
[6/10/2023, 12:31:40 AM] info: Starting up SSDP server
root@pool-pi:/#
`

tagyoureit, 2023-06-10:
Have you followed the setup instructions in the [wiki](https://github.com/tagyoureit/nodejs-poolController/wiki/Docker)?

blackax1, 2023-06-10:
Yes, that it why I also posted the output from the default docker run command, I also have nodejs-poolcontroller-dashpanel running without any issues.

Im able to run the node app and that is how I got the base config for config.json and data directory

blackax1, 2023-06-10:
I have tried older versions and all of the 8.x.x release are affected. I was able to get 7.6.1 to start

treyrich, 2023-06-10:
Duplicate of #716 as soon as a new docker image is published you should be solid, or you can clone v8.0.3 and build the image yourself to test.

treyrich, 2023-06-14:
@emes are you responsible for publishing new docker images? Any interest in getting that automated? I'd be happy to help.

emes, 2023-06-29:
hey @treyrich the docker build is automated, but with a delay as it checks every two hours.  right way would be to do this via GitHub action or similar.  very happy if someone else wants to take over.

treyrich, 2023-07-06:
@tagyoureit it looks like this is possibly related to the fact that 8.0.2 is tagged as the latest release rather than 8.0.3. I'm not certain what would have caused that, but I believe it's simply based on what date is on the git tag itself. You may need to tag a new release for it to be flagged as the latest

alexdelprete, 2023-07-20:
The wiki instructions weren't working for me.

The only way to make the docker images work was to create a config.json with `{}` both in poolcontroller and poolcontroller-dashboard data folders.

But that was not enough, on start it gave me access denied on `poolState.json` and `poolConfig.json`:

```
[7/20/2023, 10:16:33 PM] error: Error writing pool config Error: EACCES: permission denied, open '/app/data/poolConfig.json'
    at Object.openSync (node:fs:601:3)
    at Object.writeFileSync (node:fs:2249:35)
    at /app/controller/Equipment.ts:376:30 {
  errno: -13,
  syscall: 'open',
  code: 'EACCES',
  path: '/app/data/poolConfig.json'
} /app/data/poolConfig.json
[7/20/2023, 10:16:38 PM] error: Error writing pool config Error: EACCES: permission denied, open '/app/data/poolConfig.json'
    at Object.openSync (node:fs:601:3)
    at Object.writeFileSync (node:fs:2249:35)
    at /app/controller/Equipment.ts:376:30
    at runNextTicks (node:internal/process/task_queues:60:5)
    at listOnTimeout (node:internal/timers:538:9)
    at processTimers (node:internal/timers:512:7) {
  errno: -13,
  syscall: 'open',
  code: 'EACCES',
  path: '/app/data/poolConfig.json'
} /app/data/poolConfig.json
[7/20/2023, 10:16:38 PM] error: Error writing pool state Error: EACCES: permission denied, open '/app/data/poolState.json'
    at Object.openSync (node:fs:601:3)
    at Object.writeFileSync (node:fs:2249:35)
    at /app/controller/State.ts:87:20 {
  errno: -13,
  syscall: 'open',
  code: 'EACCES',
  path: '/app/data/poolState.json'
} /app/data/poolState.json

```

I solved that by introducing PID/GID in the docker-compose file, and chowned the two data folders (and subfolders) to that id (I used 1000).

After a restart it finally created everything as expected:

```
Init state for Pool Controller
[7/20/2023, 10:17:21 PM] info: The current git branch output is master
[7/20/2023, 10:17:21 PM] info: The current git commit output is 8197e90
[7/20/2023, 10:17:21 PM] info: Starting up SSDP server
[7/20/2023, 10:17:21 PM] info: Server is now listening on 0.0.0.0:4200 - 172.19.0.23:4200
[7/20/2023, 10:17:21 PM] error: Error opening port 0: Error: No such file or directory, cannot open /dev/ttyUSB0. Retry in 10 seconds
[7/20/2023, 10:17:21 PM] info: Checking njsPC versions...
[7/20/2023, 10:17:21 PM] info: Starting Pool System undefined
[7/20/2023, 10:17:21 PM] info: Listening for any installed OCPs
[7/20/2023, 10:17:21 PM] info: Auto-backup initialized Last Backup: 1970-01-01T01:00:00.000+0100
```

This is my final docker-compose.yml file, in case someone has my same issues and stumbles here:

```
version: "3.8"
services:
  poolcontroller:
    image: msmi/nodejs-poolcontroller:latest
    container_name: poolcontroller
    restart: unless-stopped
    ports:
      - "4200:4200"
    environment:
      - TZ=Europe/Rome
      - PUID=1000
      - PGID=1000
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - $PWD/poolcontroller/config.json:/app/config.json
      - $PWD/poolcontroller/data:/app/data
      - $PWD/poolcontroller/logs:/app/logs

  poolcontroller-dashpanel:
    image: msmi/nodejs-poolcontroller-dashpanel:latest
    container_name: poolcontroller-dashpanel
    restart: unless-stopped
    ports:
      - "5150:5150"
    environment:
      - TZ=Europe/Rome
      - PUID=1000
      - PGID=1000
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - $PWD/poolcontroller-dashpanel/config.json:/app/config.json
    depends_on:
      - poolcontroller

```

treyrich, 2023-08-23:
I believe that this issue is now resolved that v8.0.2 is available as a docker image. I've been up and running for a couple weeks on it and everything is working as expected now.

jonathanq, 2023-10-12:
I'm having this issue as well - even when using the 8.0.2 or 8.0.3 tags on the docker images.  Log below was from trying 8.0.2.  But 8.0.3 tag does the same thing with just a different git commit output hash.

```
Init state for Pool Controller
[10/12/2023, 6:37:04 PM] info: The current git branch output is 
[10/12/2023, 6:37:04 PM] info: The current git commit output is 64743d6
[10/12/2023, 6:37:04 PM] info: Starting up SSDP server
```

Is there a way to get more details on what the error is that is causing it to restart?  There is nothing in the logs.

emes, 2023-10-12:
@alexdelprete thanks.  helpful for new users.

@treyrich yes, it builds a tagged version (and latest) whenever a git tag is pushed and sometime package.json version gets updated and git tag is not always pushed at the same time.  new commits are tagged latest.

@jonathanq can you pull/try msmi/nodejs-poolcontroller (without the version tag) and let me know results?

jonathanq, 2023-10-13:
When I run without the version tag it gets git commit 8197e90 (which is from June 4, the fix for https://github.com/tagyoureit/nodejs-poolController/issues/716 was merged on June 9).  Not sure why - but the docker latest tag is still pointing to the older version.   That's why I was trying with the specific 8.0.3 tag.

```
$   docker run --rm -it --group-add dialout -p 4200:4200 --device /dev/ttyUSB0 msmi/nodejs-poolcontroller
Init state for Pool Controller
[10/13/2023, 3:20:50 PM] info: The current git branch output is master
[10/13/2023, 3:20:50 PM] info: The current git commit output is 8197e90
[10/13/2023, 3:20:51 PM] info: Starting up SSDP server
```

For now - I've opted to run the app directly on my Pi4 rather than in docker.  But happy to help test a fix if its available so i can switch over to docker later.

emes, 2023-12-20:
Please pull the latest and let me know how it goes:

`docker pull msmi/nodejs-poolcontroller`

tagyoureit, 2024-06-10:
I set up Github Actions to automatically create/update Docker builds.  https://github.com/tagyoureit/nodejs-poolController/discussions/967

closed 2024-06-10 by tagyoureit