tagyoureit/nodejs-poolController #809: [BUG] Manually setting end time in a schedule to midnight is saved as noon

CodeSlinger69, 2023-07-05:
### nodejs-poolController Version/commit

HEAD

### nodejs-poolController-dashPanel Version/commit

HEAD

### Node Version

18

### Platform

Linux, raspian

### Are you using Docker?

- [ ] Yes.

### What steps will reproduce the bug?

With clock mode set to "12 Hour", click on "Add Schedule" in the Schedules Tab. Select a circuit, then manually type"12:00pm" in the End Time box. You can just replace the default "5" with "12" as well. Click "Save Schedule".

[image]

### What happens?

The file data/poolConfig.json contains the following:

```
"schedules": [
    {
      "id": 1,
      "startTimeType": 0,
      "endTimeType": 0,
      "display": 0,
      "circuit": 2,
      "scheduleDays": 127,
      "scheduleType": 128,
      "heatSetpoint": 78,
      "coolSetpoint": 100,
      "heatSource": 0,
      "startTime": 480,
      "endTime": 720,
      "startDate": "2023-07-04T00:00:00.000-0700",
      "isActive": true,
      "disabled": false,
      "master": 1
    }
```

Note that endTime is saved as 720 (minutes from the start of day).

### What should have happened?

endTime should be recorded as 1440.

### Additional information

There are two workarounds:
1. Use a 24 hour clock. In this case, manually entering 24:00 is recorded correctly.
2. Use the spinners to set the end time.

I only mention this relatively small bug as I did get caught by it, and spent a good deal of time figuring out why my schedules were not activating. I am not an expert on the timeSpinner object, so I'm not sure if this an inherent problem with that object, or if overriding something like the parser function would help solve for this.

For now, I'm just using the 24 hour clock to avoid problems.

rstrouse, 2023-07-05:
12PM is actually noon and 12AM is midnight.  I'll still take a look to see if there is a bug here though.

CodeSlinger69, 2023-07-05:
Yes, I agree, noon should be represented as 12pm and midnight should be represented as 12am.  My brain got all twisted up on this as I was trying to figure out what was going on.

However, use the spinner buttons next to the time field:

- Moving from 11:30am to 12:00pm is correct.
- Moving from 11:30pm also moves to 12:00pm which is not correct. 

So try this instead:
- Create a new schedule
- Manually type in 12:00am as the end time
- Save the schedule.
- In poolConfig.json, the endTime is reported as 0, not 1440 as expected.

But once again, in 24 hour mode:
- Create a new schedule
- Manually type in 24:00 as the end time
- Save the schedule.
- In poolConfig.json, the endTime is reported as 1440 as expected.

I hate boundary cases :(  However, I just realized that timeSpinner is a custom widget in widgets.js. I'll poke through the code a bit and see if I can see something obvious.

CodeSlinger69, 2023-07-06:
I think I figured out what is going on.

What is happening is that the parseTime function in the timeSpinner widget does not understand the meaning of "midnight" (i.e. 12:00am) in the context of the time field that it is parsing.

The meaning for 12:00am in the Start Time field is **start of day**, or 0 minutes from midnight as recorded in poolConfig.json.
The meaning for 12:00am in the End Time field should be **end of day** however, or 1440 minutes from midnight. 

parseTime does not understand this meaning and treats 12:00am as a start of day thing regardless of which timeSpinner element that its attached to.

You could probably add some branching logic to treat the meaning of 12:00am differently depending on the element. o.binding contains "starTime" or "endTime" after all, but I'm not sure its worth the effort. I'm also not sure where you guys pulled this widget.js code from (or if you wrote it from scratch for this project), but it was clearly written to be unaware of the logic of the underlying application that it supports, which is usually a good design principle.

For me, I'm just going to stick with the workaround of using the 24hour clock setting. 00:00 and 24:00 are the exact same time on the clock, but the meaning is unambiguous, as least in how those times are translated into "minutes from midnight" for this njsPC application.

rstrouse, 2023-07-10:
With the new changes to the njsPC schedule calculations this should no longer be a problem.  12:00am is actually midnight on the following day or 0 offset on the following day.  It is not 1440 on the current day because the end times are inclusive.  11:59pm is the last minute of the current day but it really doesn't matter anymore because of the aforementioned changes.

CodeSlinger69, 2023-07-11:
I'll test the schedule calculations, but in terms of the dashPanel interface, my original observations remain. From above:

> However, use the spinner buttons next to the time field:
> * Moving from 11:30am to 12:00pm is correct.
> * Moving from 11:30pm also moves to 12:00pm which is not correct.

The display is incorrect, but config.json does have the correct value:

[image]

[image]

> * Manually type in 12:00am as the end time
> * In poolConfig.json, the endTime is reported as 0, not 1440 as expected.

This is still wrong in config.json:

[image]

[image]

But, like I said, using 24 hour mode = no problems. So maybe this is just pedantic at this point. I did get caught by this at one point and had to debug through code to figure it out though.

rstrouse, 2023-07-11:
I just tested this and got 0/720 for the offset and it went from 11:30am to 12:00pm.  Hard refresh your browser or clear your cache.  This has to be picking up an old dashPanel script.

CodeSlinger69, 2023-07-11:
> The bottom one is correct 12am (midnight) for the end time is actually the next day because of inclusivity.

I'm not meaning to nitpick on this one, honest. In 24 hour mode, if you click from 23:30 to 24:00, the end time is recorded as 1440. In 12 hour mode if you click from 11:30pm to midnight (currently shown in dashpanel as 12:00pm) then the end time is recorded as zero. I think the 24 hour mode handling/interpretation is correct. Midnight should be "the end of the day" when used in the end time field, regardless if the clock is set to 12 or 24 hour mode. 

> I just tested this and got 0/720 for the offset and it went from 11:30am to 12:00pm. Hard refresh your browser or clear your cache. This has to be picking up an old dashPanel script.

I clear cache before testing any changes. But I just cleared and tested again. However I'm talking about moving from 11:30 PM to midnight. Not 11:30 AM to noon. In this test, the time moves from 11:30 pm to 12:00 pm - it should move from 11:30 pm to 12:00 am.

Man do I hate this AM/PM stuff. 24 hour clock is so much less problematic.

rstrouse, 2023-07-13:
Dates in general are a nightmare.  The end time is meant to be inclusive so 24:00 is the first minute on the next day.  So 0 is the correct output.  There are 0-1339 minutes in a day.  I have been reworking the Nixie schedules a bit to accommodate the nocturnal cooling setpoints and multiple overlapping schedules for the same start time.  Although those do not really make much sense.  Normally you would create for instance a pool schedule that is inclusive of all the other overlapping schedules needed for operation.  Then the schedules would include circuits within that schedule that are needed for other things like cleaning and spillway... etc.

Either way the schedule algorithms are getting much smarter in that they are managing the window for the next scheduled run time.

I'll let you know when it is complete.

rstrouse, 2023-07-15:
I have posted some updates to njsPC and dashPanel that reworks the schedule calculations.  It also adds offsets for sunrise and sunset.

CodeSlinger69, 2023-07-16:
Nice work. The issue where a schedule is set to midnight in 12hour mode seems to be fixed - the endTime is recorded as 1440 as expected.

I'll close this thread.

closed 2023-07-16 by CodeSlinger69