tagyoureit/nodejs-poolController #810: [BUG] Nixie - ticking "Auto Adjust DST" is not saved.

CodeSlinger69, 2023-07-05:
### nodejs-poolController Version/commit

Head

### Are you using Docker?

- [ ] Yes.

### What steps will reproduce the bug?

Go to Settings/General, tick "Auto Adjust DST"

### What happens?

Box shows as ticked, message is displayed on console:

`7/4/2023, 5:06:07 PM] info: [5:06:07 PM] 192.168.1.87 PUT /config/general {"options":{"units":0,"adjustDST":true,"clockMode":24},"location":{"timeZone":132}}
`

The setting is not stored in data/poolConfig.json:

`  "pool": {
    "master": 0,
    "options": {
      "units": 0,
      "clockMode": 24,
      "adjustDST": false,
      "freezeThreshold": 35,
      "pumpDelay": false,`

Reloading the page reverts "Auto Adjust DST" back to unticked.

### What should have happened?

"Auto Adjust DST" stays ticked on page refresh, and is recorded in data/poolConfig.json correctly.

### Additional information

Do we care? Since this is Nixie, we are getting the local time including DST adjustments from the underlying operating system. We probably don't need this DST adjustment when setup as a Nixie. We probably don't need the timezone setting either, for that matter. But if it is needed, maybe grab it from the OS instead? I can't imagine a case where the Nixie timezone should be different than the OS timezone.

rstrouse, 2023-07-05:
Yeah this should probably be removed from the Nixie config,  This is a Pentair *Touch panel thing.

rstrouse, 2023-07-10:
I have disabled these controls when the clockMode is server.  It is good to see what the server is reporting though.

CodeSlinger69, 2023-07-11:
> I have disabled these controls when the clockMode is server. It is good to see what the server is reporting though.

I see that those controls are disabled now, but where do you set clockMode? And what server reporting are you looking at? This is what I see:

[image]

rstrouse, 2023-07-11:
With nixie the clock is always server and whichever configuration is set for the time sync on it.  This configuration is done on the host OS level and can hit whichever time server you like.  In fact you can set this up for your router to do the sync but if I remember correctly the pi uses pool.npt.org by default.

On *Touch it is always manual but the dallas clock on the board is so awful that it is almost never right so we constantly sync it when the user selects server.  On IntelliCenter it reports from an ntp server when it is connected to the internet but falls back to its internal clock.  We allow all 3 methods for that.

CodeSlinger69, 2023-07-11:
Awesome, thanks. I know how to configure a NTP client, so no worries. I was just confused by your statement that "It is good to see what the server is reporting though."  When in Nixie mode, the only way to see this is in the host OS, not dashPanel.

Thanks for the help, and I'll close this issue.

closed 2023-07-11 by CodeSlinger69