tagyoureit/nodejs-poolController #812: [BUG] (minor visual) dashPanel Schedules panel does not respect 12/24 hour setting

CodeSlinger69, 2023-07-05:
### nodejs-poolController Version/commit

head

### Are you using Docker?

- [ ] Yes.

### What steps will reproduce the bug?

Change 12 or 24 hour clock setting.

### What happens?

 The schedules panel on the dashPanel main screen will continue to display 12hour clock values.

### What should have happened?

24 hour clock values (i.e. HH:mm) should be displayed when the clock is set to the 24 hour option.

### Additional information

This is a low priority, minor visual issue.

I attempted to fix this in nodejs-poolController-dashPanel/scripts/schedules.js, [lines 109 & 110](https://github.com/rstrouse/nodejs-poolController-dashPanel/blob/c8bf32c06a910807b37027a76d5f97f91d490185/scripts/schedules.js#L109) as so:

```
el.find('.picStartTime').text(startTimeType.name !== 'manual' ? data.startTimeType.desc : startTime.formatTime((o.clockMode == 12 ? 'h:mmtt' : 'HH:mm'), '--:--'));
el.find('.picEndTime').text(endTimeType.name !== 'manual' ? data.endTimeType.desc : endTime.formatTime((o.clockMode == 12 ? 'h:mmtt' : 'HH:mm'), '--:--'));
```

But it does not work correctly. I'll readily admit to not being an expert in HTML coding, so maybe o.clockMode is not valid in this scope?

rstrouse, 2023-07-09:
I’ll have a look

rstrouse, 2023-07-10:
I fixed this in dashPanel.  Pull njsPC and dashPanel.

closed 2023-07-10 by rstrouse