tagyoureit/nodejs-poolController #820: [BUG] Heliotrope and schedule flexibility logic

CodeSlinger69, 2023-07-08:
### nodejs-poolController Version/commit

head

### Are you using Docker?

- [ ] Yes.

### What steps will reproduce the bug?

I wanted to leverage Nixie to automatically cool my pool at night using my solar system. Nixie checks to see "isNight" as a condition for doing so, but the logic in this method was incorrect.

Additionally, I wanted schedules that cross the midnight boundary, for example sunset to sunrise, or just 22:00 - 06:00. The logic to determine if a schedule should be "on" does not handle this situation.

I've corrected the logic, it works in my testing so far, and I created a number of pull requests ([1](https://github.com/tagyoureit/nodejs-poolController/pull/816), [2](https://github.com/tagyoureit/nodejs-poolController/pull/818), [3](https://github.com/tagyoureit/nodejs-poolController/pull/819)) for inclusion in the official code base if wanted. I'll be honest, I am not a git expert, so forgive me if I messed up the branches, pulls, pushes, commits, and so on.

I've also created a few other pull requests ([1](https://github.com/tagyoureit/nodejs-poolController/pull/807), [2](https://github.com/tagyoureit/nodejs-poolController/pull/813)) for other bugs I discovered along the way.

I'm pleased that Nixie is working the way I want now, and please feel free to use my code if its helpful.

### What happens?

See above.

### What should have happened?

See above.

rstrouse, 2023-07-09:
I’ll check them out tomorrow and integrate them.

CodeSlinger69, 2023-07-09:
Closing this due to learning that njsPC had a design decision to NOT allow schedules to cross the midnight boundary.

closed 2023-07-09 by CodeSlinger69