tagyoureit/nodejs-poolController #822: Compool (pre-Easytouch)

jimshew, 2023-07-10:
## 💥 Proposal

Compool isn't explicitly listed as a supported system, but should just a subset of the same stuff we see in an Easytouch. 

### What feature you'd like to see

I believe the probing is failing as it goes from unidentified to nixie with no features.

## Motivation

Should just be a simple update of version test or probing to correctly identify the system.

## Pitch

It appears to be the same traffic, for something like this temperature heartbeat:

```
00000000: 5aff aa0f 1b00 01ff 02d3 5afa 5aff aa0f  Z.........Z.Z...
00000010: 1b00 01ff 02d3 5afa 5aff aa0f 1b00 01ff  ......Z.Z.......
00000020: 02d3 5afa 5aff aa0f 1b00 01ff 02d3 5afa  ..Z.Z.........Z.
00000030: 5aff aa0f 1b00 01ff 02d3 5afa 5aff aa0f  Z.........Z.Z...
00000040: 1b00 01ff 02d3 5afa 5aff aa0f 1b00 01ff  ......Z.Z.......
00000050: 02d3 5afa 5aff aa0f 1b00 01ff 02d3 5afa  ..Z.Z.........Z.
```

or here with spa and heater turned on:

```
00000000: 005a ffaa 0f1b 0210 330a 0202 106f 0000  .Z......3....o..
00000010: fe5a ff7b b428 e802 d35a da5a ffaa 0f1b  .Z.{.(...Z.Z....
00000020: 0210 340a 0202 106f 0000 c05a ffaa 0f1b  ..4....o...Z....
00000030: 0001 ff02 d35a fa5a ffaa 0f1b 0001 ff82  .....Z.Z........
00000040: 5aff aa0f 1b00 01ff 02d3 5ada aa0f 1b00  Z.........Z.....
00000050: 01ff 02d3 5afa 5aff fa23 4040 02d3 5ada  ....Z.Z..#@@..Z.
00000060: 5aff 6824 0844 6a24 2408 f40b 0810 906c  Z.h$.Dj$$......l
00000070: d902 0401 a275 ff5a ffaa 0f1b 0081 5aff  .....u.Z......Z.
00000080: aa0f 1be0 ffaa 0001 0001 ff00 01ff 02ff  ................
00000090: 5aff aa0f 1b00 01ff 02d3 5aff aa0f 1b00  Z.........Z.....
000000a0: 01ff 02d3 5afa 5aff aa0f 1b00 01ff f801  ....Z.Z.........
000000b0: 0001 ff02 fe5a ffaa 0f02 d35a fa5a ffaa  .....Z.....Z.Z..
000000c0: ff20 444d 4381 8110 6f00 0000 8e93 3b00  . DMC...o.....;.
```

As far as I can tell, the legacy Compool devices (lx3400 and cp3400 in my example) are the same basic protocol that Easytouch is based on, as documented (https://web.archive.org/web/20220117165056/https://carlstrom.com/pool/compool/protocol.txt).

rstrouse, 2023-07-10:
If you set this up it may identify as a SunTouch controller

jimshew, 2023-07-11:
My system didn't get probed as SunTouch, here's console from starting:

```
root@home:~# docker run --rm -it -u node:dialout -p 4200:4200 --device /dev/ttyUSB0 --name polcon msmi/nodejs-poolcontroller
Init state for Pool Controller
[7/11/2023, 3:01:15 PM] info: The current git branch output is master
[7/11/2023, 3:01:15 PM] info: The current git commit output is 8197e90
[7/11/2023, 3:01:15 PM] info: Starting up SSDP server
[7/11/2023, 3:01:15 PM] info: Server is now listening on 0.0.0.0:4200 - 172.17.0.3:4200
[7/11/2023, 3:01:15 PM] info: Serial port: /dev/ttyUSB0 request to open successful 9600b 8-none-1
[7/11/2023, 3:01:15 PM] info: Checking njsPC versions...
[7/11/2023, 3:01:15 PM] info: Starting Pool System undefined
[7/11/2023, 3:01:15 PM] info: Listening for any installed OCPs
[7/11/2023, 3:01:15 PM] info: Auto-backup initialized Last Backup: 1970-01-01T00:00:00.000+0000
[7/11/2023, 3:01:22 PM] info: New socket client connected N9VB_k-QIX-yRtJXAAAA -- 192.168.21.111
RESETTING DATA -- Controller type changed from undefined to nixie
[7/11/2023, 3:01:22 PM] info: Initializing Nixie Control Panel for Nixie Single Body
[7/11/2023, 3:01:22 PM] info: Initializing Intake/Return valves
[7/11/2023, 3:01:22 PM] info: Initializing Nixie Controller
[7/11/2023, 3:01:22 PM] info: Initializing Nixie body Pool
[7/11/2023, 3:01:23 PM] info: Initializing Filter Filter
[7/11/2023, 3:01:23 PM] info: Initializing Nixie circuit Pool
[7/11/2023, 3:01:23 PM] info: Nixie Controller Initialized
[7/11/2023, 3:01:23 PM] info: Nixie Single Body control board initialized
[7/11/2023, 3:02:01 PM] warn: dt:Tue Jul 11 2023 15:02:01 GMT+0000 (Coordinated Universal Time) lat:undefined lon:undefined Not enough information to calculate Heliotrope.  See https://github.com/tagyoureit/nodejs-poolController/issues/245
[7/11/2023, 3:02:01 PM] warn: dt:Tue Jul 11 2023 15:02:01 GMT+0000 (Coordinated Universal Time) lat:undefined lon:undefined Not enough information to calculate Heliotrope.  See https://github.com/tagyoureit/nodejs-poolController/issues/245
[7/11/2023, 3:02:01 PM] info: Checking njsPC versions...
[7/11/2023, 3:03:01 PM] warn: dt:Tue Jul 11 2023 15:02:01 GMT+0000 (Coordinated Universal Time) lat:undefined lon:undefined Not enough information to calculate Heliotrope.  See https://github.com/tagyoureit/nodejs-poolController/issues/245
[7/11/2023, 3:03:01 PM] warn: dt:Tue Jul 11 2023 15:02:01 GMT+0000 (Coordinated Universal Time) lat:undefined lon:undefined Not enough information to calculate Heliotrope.  See https://github.com/tagyoureit/nodejs-poolController/issues/245
```

And after connecting with dash panel to enable silly logging:

```
[7/11/2023, 3:13:10 PM] verbose: Tossed Inbound Bytes 142,147,50,0,0,112,0,4,16,254,90,255,170,15,27,2,16,0,9,2,8,0,85,0,0 due to an unrecoverable collision.
[7/11/2023, 3:13:13 PM] verbose: Tossed Inbound Bytes 0,142,147,50,0,0,112,0,4,16,254,90,255,170,15,27,2,16,0,9,2,8,0 due to an unrecoverable collision.
[7/11/2023, 3:13:15 PM] verbose: Tossed Inbound Bytes 85,0,0,0,142,147,50,0,0,112,0,4,16,90,90,90,255,170,15,27,2,16,0,9,2,8 due to an unrecoverable collision.
[7/11/2023, 3:13:18 PM] verbose: Tossed Inbound Bytes 0,85,0,0,0,142,147,50,0,0,112,0,4,16,254,90,255,170,15,27,2,16,0 due to an unrecoverable collision.
[7/11/2023, 3:13:21 PM] verbose: Tossed Inbound Bytes 9,2,8,0,85,0,0,0,142,147,50,0,0,112,0,4,16,254,90,255,170,15,27,2,16 due to an unrecoverable collision.
[7/11/2023, 3:13:23 PM] verbose: Tossed Inbound Bytes 0,9,2,8,0,85,0,0,0,142,147,50,0,0,112,0,4,16,254,90,255,170,15,27,2 due to an unrecoverable collision.
[7/11/2023, 3:13:26 PM] verbose: Tossed Inbound Bytes 16,0,9,2,8,0,85,0,0,0,142,147,50,0,0,112,0,4,16,254,90,255,170 due to an unrecoverable collision.
[7/11/2023, 3:13:29 PM] verbose: Tossed Inbound Bytes 15,27,2,16,0,9,2,8,0,85,0,0,0,142,147,50,0,0,112,0,4,16,250,90,255 due to an unrecoverable collision.
[7/11/2023, 3:13:31 PM] verbose: Tossed Inbound Bytes 170,15,27,2,16,0,9,2,8,0,85,0,0,0,142,147,50,0,0,112,0,4,16 due to an unrecoverable collision.
[7/11/2023, 3:13:31 PM] verbose: Tossed Inbound Bytes 254,90,255,170,15,27,2,16,0,9,2,8,0,90,0,0,0,142,147,50,0,0 due to an unrecoverable collision.
[7/11/2023, 3:13:34 PM] verbose: Tossed Inbound Bytes 112,0,4,21,254,90,255,170,15,27,2,16,0,9,2,8,0,90,0,0,0,142,147,50,0,0 due to an unrecoverable collision.
[7/11/2023, 3:13:37 PM] verbose: Tossed Inbound Bytes 112,0,4,21,254,90,255,170,15,27,2,16,0,9,2,8,0,90,0,0,0,142,147,50,0,0 due to an unrecoverable collision.
[7/11/2023, 3:13:39 PM] verbose: Tossed Inbound Bytes 112,0,4,21,254,90,255,170,15,27,2,16,0,9,2,8,0,90,0,0,0,142,147,50,0,0 due to an unrecoverable collision.
[7/11/2023, 3:13:42 PM] verbose: Tossed Inbound Bytes 112,0,4,21,254,90,255,170,15,27,2,16,0,9,2,8,0,90,0,0,0,142,147,50,0,0 due to an unrecoverable collision.
[7/11/2023, 3:13:45 PM] verbose: Tossed Inbound Bytes 112,0,4,21,254,90,255,170,15,27,2,16,0,9,2,8,0,90,0,0,0,142,147,50,0,0 due to an unrecoverable collision.
[7/11/2023, 3:13:47 PM] verbose: Tossed Inbound Bytes 112,0,4,21,254,90,255,170,15,27,2,16,0,9,2,8,0,90,0,0,0,142,147,50,0,0 due to an unrecoverable collision.
[7/11/2023, 3:13:50 PM] verbose: Tossed Inbound Bytes 112,0,4,21,254,90,255,170,15,27,2,16,0,9,2,8,0,90,0,0,0,142,147 due to an unrecoverable collision.
[7/11/2023, 3:13:53 PM] verbose: Tossed Inbound Bytes 50,0,0,112,0,4,21,254,90,255,170,15,27,2,16,0,9,2,8,0,90,0,0,0 due to an unrecoverable collision.
[... cut ...]
[7/11/2023, 3:14:19 PM] verbose: Tossed Inbound Bytes 254,90,255,170,15,27,2,16,1,9,2,8,0,95,0,0,0,142,147,50,0,0 due to an unrecoverable collision.
[7/11/2023, 3:14:22 PM] verbose: Tossed Inbound Bytes 112,0,4,27,254,90,255,170,15,27,2,16,1,9,2,8,0,95,0,0,0,142,147,50,0,0 due to an unrecoverable collision.
```

Is there a static test case configuration file for SunTouch or EasyTouch that I can force the model and skip the probing? I don't see an option defined in the wiki or my running `config.json`.

rstrouse, 2023-07-11:
Flip the A/B data wires on the adapter.  I am not seeing preamble in the output.

jimshew, 2023-07-17:
Apparently I read too many random old Perl docs that had the wires crossed, making it look sane to me. Flipping the wires yields:

```
root@home:~# xxd /dev/ttyUSB0
00000000: 00a5 01ab c2c9 fbdf b7dd ffef ff2f ffff  ............./..
00000010: ffe3 b21a ffff 1fff f757 962d ff00 a500  .........W.-....
00000020: 9578 72bf 77eb f7ef ff2f ffff ffe3 b21a  .xr.w..../......
00000030: ffff 1fff f757 962d ff00 a500 9578 72bf  .....W.-.....xr.
00000040: 77eb f7ef ff2f ffff ffe3 d91a ffff 1fff  w..../..........
00000050: f757 962d ff00 a501 abc2 c9fb dfb7 ddff  .W.-............
00000060: efff 2fff ffff e3d9 1aff ff1f fff7 5796  ../...........W.
00000070: 2dff 00a5 01ab c2c9 fbdf b7dd ffef ff2f  -............../
00000080: ffff ffe3 b21a ffff 1fff f757 962d ff00  ...........W.-..
00000090: a501 abc2 c9fb dfb7 ddff efff 2fff ffff  ............/...
000000a0: e399 1aff ff1f fff7 5796 2dff 00a5 01ab  ........W.-.....
000000b0: c2c9 fbdf b5dd ffef ff2f ffff ffe3 d91a  ........./......
000000c0: ffff 1fff f755 962d ff00 a501 abc2 c9fb  .....U.-........
000000d0: dfb5 ddff efff 2fff ffff e3b2 1aff ff1f  ....../.........
000000e0: fff7 5596 2dff 00a5 0095 7872 bf57 ebf7  ..U.-.....xr.W..
000000f0: efff 2fff ffff e3ba 1aff ff1f fff7 5596  ../...........U.
00000100: 2dff 00a5 01ab c2c9 fbdf b5dd ffef ff2f  -............../
00000110: ffff ffe3 b21a ffff 1fff f755 962d ff00  ...........U.-..
00000120: a501 abc2 c9fb dfb5 ddff efff 2fff ffff  ............/...
00000130: e3b2 1aff ff1f fff7 5596 2dff 00a5 0095  ........U.-.....
00000140: 7872 bf57 ebf7 efff 2fff ffff e3b2 1aff  xr.W..../.......
00000150: ff1f fff7 5596 4b00 a500 9578 72bf 57eb  ....U.K....xr.W.
00000160: f7ef ff2f ffff ffe3 b21a ffff 1fff f755  .../...........U
00000170: 962d ff00 a501 abc2 c9fb dfb5 ddff efff  .-..............
00000180: 2fff ffff e3b2 1aff ff1f fff7 5596 2dff  /...........U.-.
00000190: 00a5 0095 7872 bf57 ebf7 efff 2fff ffff  ....xr.W..../...
000001a0: e3b2 1aff ff1f fff7 5596 2dff 00a5 0095  ........U.-.....
```

After removing the config and restarting the docker container, I still get the same behavior of an undefined model then switching to Nixie single body. I tried using messageManager on dashPanel, but not seeing anything come through. I'd suspect the RS485 wires, USB adapter, or docker config, but since I can see sane bytes coming through raw ... I'm still thin old controller or panel?  Maybe I do have a transmit problem, I never see the transmit LED light up, only receive (using the $16 waveshare USB-RS485 from amazon to the CP3400's second RJ-12 socket). Even if I did have a transmit problem, shouldn't the defaults allow for decoding temperature, auto/service mode, or whatever the CP3400 panel sends/receives to the LX3400 controller? or will it never get to those since it didn't successfully probe?

I switched dashpanel to Nixie shared, just to be able to select Pool circuit, and here's a small sample from the console (now with wires reversed):

```
[7/17/2023, 1:02:56 AM] verbose: Tossed Inbound Bytes 247,67,150,45,255,0,165,0,149,120,114,191,183,182,247,239,255,51,255,255,255,227,178,38,255,255,31,255 due to an unrecoverable collision.
[7/17/2023, 1:02:58 AM] verbose: Tossed Inbound Bytes 247,67,150,45,255,0,165,0,149,120,114,191,183,182,247,239,255,51,255,255,255,227,218,38,255,255,31,255 due to an unrecoverable collision.
[7/17/2023, 1:03:01 AM] info: [1:03:01 AM] 192.168.21.10 PUT /state/circuit/setState {"id":6,"state":true}
[7/17/2023, 1:03:01 AM] verbose: Turning on a body circuit Pool
[7/17/2023, 1:03:01 AM] info: NCP: Setting Circuit Pool to true
[7/17/2023, 1:03:01 AM] verbose: Tossed Inbound Bytes 247,67,150,45,255,0,165,1,171,194,201,251,223,155,219,255,239,255,51,255,255,255,227,178,38,255,255,31,255 due to an unrecoverable collision.
[7/17/2023, 1:03:04 AM] verbose: Tossed Inbound Bytes 247,67,150,45,255,0,165,0,149,188,114,191,183,182,247,239,255,51,255,255,255,227,218,38,255,255,31,255 due to an unrecoverable collision.
[7/17/2023, 1:03:06 AM] verbose: Tossed Inbound Bytes 247,67,150,45,255,0,165,0,149,120,114,191,183,182,247,239,255,51,255,255,255,227,210,38,255,255,31,255 due to an unrecoverable collision.
[7/17/2023, 1:03:09 AM] verbose: Tossed Inbound Bytes 247,67,150,45,255,0,169,128,149,120,114,191,183,182,247,239,255,51,255,255,255,227,178,38,255,255,31,255 due to an unrecoverable collision.
[7/17/2023, 1:03:10 AM] info: [1:03:10 AM] 192.168.21.10 PUT /state/circuit/setState {"id":6,"state":false}
[7/17/2023, 1:03:10 AM] verbose: Turning off a body circuit Pool
[7/17/2023, 1:03:10 AM] info: NCP: Setting Circuit Pool to false
[7/17/2023, 1:03:12 AM] verbose: Tossed Inbound Bytes 247,67,150,45,255,0,165,1,171,194,201,251,223,155,219,255,239,255,51,255,255,255,227,178,38,255,255,31,255 due to an unrecoverable collision.
[7/17/2023, 1:03:14 AM] verbose: Tossed Inbound Bytes 247,67,150,45,255,0,165,1,171,194,201,251,223,155,219,255,239,255,51,255,255,255,227,210,38,255,255,31,255 du
```

rstrouse, 2023-07-17:
I am not seeing a preamble pattern from the controller.  This should be a pattern like

`255,0,255,165, ...`
`ff00, ffaf, ...`

I would have expected to see this even from Compool.  The probing does not matter as the OCP will announce itself then the probing will begin.  It has not even seen a valid preamble at this point.

jimshew, 2023-07-19:
I played around with the wiring as I was getting Err 4 and Err 5 messages on the CP3400. I ended up back in the type of data as  I posted initially. I Swapped the CP3400 out for a CP3800 I got on craigslist in excellent shape, just in case it was getting weird. I played around with decoding the "basic command packet" assuming the `ffaa` was the old preamble/sync bytes (as older compool related projects note such as https://carlstrom.com/pool/compool/protocol.txt.

I threw together a python script to only decode the basic traffic from the lx3400 controller (https://raw.githubusercontent.com/jimshew/gadgetstuff/main/decodecompool.py), it's a decent sanity check though I didn't bother checking the length and therefor last portion of the packet.  Th important thing is the Desired Temp byte does indeed match what the CP3800 says (well, except for lame rounding +/- err in F calculation). With the CP3800, I switched SPA on, changed temp, and it tracked in this output.

```
root@shew:/opt/gadgetstuff# ./decodecompool.py
Synchronization byte found: 0f1b02101316020802670000007693380000700004325ada5affaa
Remaining bytes: 0f1b02101316020802670000007693380000700004325ada5affaa
Byte: 0x0F, Destination Address: 15
Byte: 0x1B, Source Address: 27
Byte: 0x02, OpCode: 2
Byte: 0x10, Info Field Length: 16
Byte: 0x13, Minutes: 19
Byte: 0x16, Hours: 22
Byte: 0x02, Primary Equipment: 2
Byte: 0x08, Secondary Equipment: 8
Byte: 0x02, Delay/Heat Source: 2
Byte: 0x67, Water Temp: 78.35
Byte: 0x00, Solar Temp: 32.0
Byte: 0x00, Spa Water Temp: 32.0
Byte: 0x00, Spa Solar Temp: 32.0
Byte: 0x76, Desired Pool Temp: 85.1
Byte: 0x93, Desired Spa Temp: 98.15
Byte: 0x38, Air Temp: 57.2
Byte: 0x00, UNK 1: 0
Byte: 0x00, UNK 2: 0
Byte: 0x70, Equip Status: 112
Byte: 0x00, Product Type: 0
Byte: 0x04, Hi Byte of Checksum: 4
Byte: 0x32, Lo Byte of Checksum: 50
Synchronization byte found: 0f1b02101316020802670000007693380000700004325ada5affaa
Remaining bytes: 0f1b02101316020802670000007693380000700004325ada5affaa
Byte: 0x0F, Destination Address: 15
Byte: 0x1B, Source Address: 27
Byte: 0x02, OpCode: 2
Byte: 0x10, Info Field Length: 16
Byte: 0x13, Minutes: 19
Byte: 0x16, Hours: 22
Byte: 0x02, Primary Equipment: 2
Byte: 0x08, Secondary Equipment: 8
Byte: 0x02, Delay/Heat Source: 2
Byte: 0x67, Water Temp: 78.35
Byte: 0x00, Solar Temp: 32.0
Byte: 0x00, Spa Water Temp: 32.0
Byte: 0x00, Spa Solar Temp: 32.0
Byte: 0x76, Desired Pool Temp: 85.1
Byte: 0x93, Desired Spa Temp: 98.15
Byte: 0x38, Air Temp: 57.2
Byte: 0x00, UNK 1: 0
Byte: 0x00, UNK 2: 0
Byte: 0x70, Equip Status: 112
Byte: 0x00, Product Type: 0
Byte: 0x04, Hi Byte of Checksum: 4
Byte: 0x32, Lo Byte of Checksum: 50
```

Should I experiment with `ff aa`` as the shorter sequence in testBroadcastHeader? (https://github.com/tagyoureit/nodejs-poolController/blob/master/controller/comms/messages/Messages.ts#L316-L324) I'm willing to experiment and figure it out, but I expect I'll be slowly making a mess.

tagyoureit, 2023-07-19:
It definitely has a shorter preamble:
```
Byte 0    SYNC BYTE 0         0xFFh (always)
Byte 1    SYNC BYTE 1         0xAAh (always)
```

You can start playing around with the testBroadcastHeader, but ultimately we would need to add another board with the specific decoding that is outlined in the docs you provided.

rstrouse, 2023-07-19:
The preamble looks more like the following for a full dword (32bits).  2 bytes of preamble is pretty dicey if there is noise.  It is odd though does compool speak with IntelliFlo?  If that is the case it would need to be looking for a completely different preamble.

```
byte 0 0xfa - 250
byte 1 0x5a - 90
byte 2 0x5a - 255
byte 3 0xff - 170
```

jimshew, 2023-07-20:
It does appear to be the full dword, and that makes sense. 

The only things currently on RS485 is an LX3400 controller and CP3400 (replaced with CP3800 temporarily). I was waiting to add Aquarite SWG to the bus just for remote features. Pumps are just the old WhisperFlo 2 HP and a couple 1HP, just controlled with relays only. I expect the old pumps to die soon, so smarter pumps is a near future need.

Goal is for family to use dashPanel or Home Assistant to adjust temp, activate spa, etc. ISL-4 spa remote is fried, can only control two pumps with an ESP inching sonoff (currently disconnected) so decided a software Control Panel was "the right way". I barely talked myself out of a $300 EasyTouch upgrade on ebay (now it's gone and I slightly regret it). Family would be strictly using mobile devices for control, so I figure I don't need most EasyTouch features if I can use Home Assistant to automate schedules. 

Between reading status (easily from earlier script), and emulating the service panel, I don't need much; I can do scenes and automations externally. If nobody else has need for Compool here, not much reason to add support. I figure since there hasn't been much interest in it, that means it wouldn't be used by anybody but me? I may keep playing with smaller scripts to get features figured out, which might be enough functionality to get by.

rstrouse, 2023-07-20:
Actually your best route may be Nixe.  Simply take the relays from the Compool and throw an 8 relay hat on the pi.  That will control the pumps and any future pumps you might add.  If you have a spa it will also manage the heater, and valves.  It will also self contain the schedules so that manual overrides are honored... etc.

After that you have a nearly unlimited number of circuits and features that you can implement.  Actually, the ids are an int so sooner or later you will run out of ids but I suspect you will run out of money first.  Lots of Nixie folks out there and you can model it all without any hardware to see what you think.

tagyoureit, 2024-04-16:
Closing due to lack of activity.

closed 2024-04-16 by tagyoureit